moveit/moveit #1: Add coveralls testing coverage badge

davetcoleman, 2016-07-29:
The FCL project has this cool badge that indicates what percent of the project has unit tests:

[image]

You can also set a minimum required coverage percent so that new pull requests require unit tests be written for significant pieces of new code. *Turning on this feature is a future topic*

We would like to add coveralls support in MoveIt! - it does not look too complicated. The discussion and an example on how to do this can be found [here](https://github.com/flexible-collision-library/fcl/pull/98#issuecomment-233742606)

130s, 2016-08-24:
@scottpaulin I unassigned myself following your comment https://github.com/ros-planning/moveit/issues/23#issuecomment-241902583. If you can give it a try that will be great.

scottpaulin, 2016-08-24:
Sorry @130s , I should have assigned it. I have also been taking forever to get it done...

Does anyone know if catkin does something special to allow ros:: package::getPath() to work?
`std::string resource_dir = ros::package::getPath("moveit_resources");`

Coveralls automagically generates a make target, which creates a test executable with the package's unit tests. When I run this test executable none of the resources can be found, [resulting in these errors.](https://gist.github.com/scottpaulin/ed50d9da4f4e43fb97a44de79da1362a). Is it possible that the coveralls make target needs to be somehow catkinized?

rhaschke, 2016-08-24:
> Does anyone know if catkin does something special to allow ros:: package::getPath() to work?
std::string resource_dir = ros::package::getPath("moveit_resources");

The Travis script sources the devel/install space to pull in those.

However, I have prepared a fix, which will work without sourcing any workspace. This will go on top of https://github.com/ros-planning/moveit_resources/pull/7 and #83.

scottpaulin, 2016-08-28:
update: currently dependant on fixing failing assertions in unit tests (when debug build is used).

v4hn, 2017-10-08:
@scottpaulin what is the state here? Did you upload your progress somewhere? Is there anything to upload yet?

Does it make sense to label this for this wmd for someone to pick?

scottpaulin, 2017-10-10:
I couldn't quite get it going sorry. It requires the CI to do a Debug build, which resulted in Travis timeouts at the time. Would be awesome if someone could pick it up.

davetcoleman, 2018-05-23:
@rhaschke recently setup ccaching for Travis, and I contacted Travis a while back about expanding our build time. @scottpaulin do you think we could try this again? could you share the branch you began working on this? @bailaC might be interested in picking this up where you left off

scottpaulin, 2018-05-24:
Nice! 

It is under the coveralls [branch](https://github.com/ros-planning/moveit_ci/tree/coveralls). From memory some flags need to be set in cmakelists files, a debug build needs to be performed (possibly with gcov enabled), and then the results need to be uploaded to coveralls. There is a [tool](https://github.com/ros-planning/moveit_ci/tree/coveralls) that may be of use for uploading the results.

Adding the badge afterward is fairly simple.

agutenkunst, 2018-10-25:
We quickly were able to demonstrate that coverage generation using 
https://github.com/mikeferguson/code_coverage works.

We currently use it in https://github.com/PilzDE/pilz_industrial_motion/tree/kinetic-devel/pilz_trajectory_generation

See
https://github.com/agutenkunst/moveit/commit/773ee86c012fceea61aef50fe75cb525f75556af
on how to integrate it. Use this commit with `catkin_make -DENABLE_COVERAGE_TESTING=ON moveit_core_coverage`

Generated Overview
[image]

Here the complete report:
[moveit_core_coverage.zip](https://github.com/ros-planning/moveit/files/2514253/moveit_core_coverage.zip)