moveit/moveit #4: Automate IK tests

davetcoleman, 2016-07-29:
@jrgnicho did a great job of [creating KinematicsBase tests](https://github.com/ros-planning/moveit_kinematics_tests), but they have not been integrated into the main MoveIt! project and they are not run by travis. We would like to pull this whole repo into the main moveit codebase and have them triggered by travis/catkin every time

Things to consider:
- Do we need to import both robots or can we base them on the PR2 or [Fanuc](https://github.com/ros-planning/moveit_ros/pull/729#issuecomment-236131127) robots?
- The URDFs should probably be moved to moveit_resources
- How long do the tests take and how much output?
- What is the overlap between these tests and the IK tests the PR2 already has in moveit?

jrgnicho, 2016-07-29:
@davetcoleman 
 - There's already tests for the PR2 so no changes should be required there.  Adding a new robot into the unit testing would involve creating the ikfast solver for it and a moveit config package.  All of that is certainly doable  
 - In addition to the URDFs we'd also need to move the moveit_config packages into moveit_resources
 - All tests run in about 20s and the output is roughly 70 lines.
 - The tests program was inspired by the [pr2 test](https://github.com/ros-planning/moveit_pr2/blob/indigo-devel/pr2_moveit_tests/kinematics/src/test_kinematics_as_plugin.cpp) so there's definitely an overlap.

davetcoleman, 2016-07-29:
> Adding a new robot into the unit testing would involve creating the ikfast solver for it and a moveit config package

I thought your tests already work, essentially. They just need to be moved...

jrgnicho, 2016-07-29:
@davetcoleman yes, if you were to just move that whole repo it should just work.

davetcoleman, 2016-07-30:
So do you think its a good idea to add this to our standard test coverage?

Post-repo merge I think we should make a pull request to review and add this. The tests themselves should probably go into either moveit_core, moveit_ros, or moveit_ikfast

v4hn, 2017-01-13:
@jrgnicho ping. What's the state here?

jrgnicho, 2017-01-13:
@v4hn sorry this fell off my radar.
I'll create the PR for this.

130s, 2017-01-13:
Coincidentally I've started working on this (since it's so useful. Thanks @jrgnicho!) and opened https://github.com/ros-planning/moveit_kinematics_tests/pull/7 yesterday.

davetcoleman, 2017-02-15:
+1 to actually moving the whole repo here

davetcoleman, 2019-01-12:
Seems @rhaschke is cleaning this up here: https://github.com/ros-planning/moveit/pull/1272

jrgnicho, 2019-01-14:
Yes this can be closed as #1272 replaces it

rhaschke, 2019-01-23:
Closing via #1272.

closed 2019-01-23 by rhaschke