moveit/moveit #5: [ros_robot_interaction] Fix warning

davetcoleman, 2016-07-29:
In the kinetic build there is [currently a warning](https://travis-ci.org/ros-planning/moveit/builds/148209367):

```
Warnings << moveit_ros_robot_interaction:make /root/ros/ws_moveit/logs/moveit_ros_robot_interaction/build.make.000.log
/root/ros/ws_moveit/src/moveit/moveit_ros/robot_interaction/src/interaction_handler.cpp: In member function ‘const kinematics::KinematicsQueryOptions& robot_interaction::InteractionHandler::getKinematicsQueryOptions() const’:
/root/ros/ws_moveit/src/moveit/moveit_ros/robot_interaction/src/interaction_handler.cpp:614:75: warning: returning reference to temporary [-Wreturn-local-addr]
   return kinematic_options_map_->getOptions(KinematicOptionsMap::DEFAULT).options_;
                                                                           ^
cd /root/ros/ws_moveit/build/moveit_ros_robot_interaction; catkin build --get-env moveit_ros_robot_interaction | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

After fixing need to ensure interactive markers still work with MoveIt! in Rviz

andreaskoepf, 2016-08-23:
What are the requirements for a fix? Would it be ok to just remove `const&` from the return type of `getKinematicsQueryOptions()` or should the method signature stay untouched? Inside this repo the method seems to be used only in [motion_planning_display.cpp#L920](https://github.com/ros-planning/moveit/blob/8d2eabfe96bb68ae27d1f4c0255eb4e6092f82f5/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_display.cpp#L920) and same file line 926.

davetcoleman, 2016-08-23:
This would not be ABI compatible but we can just not merge it into indigo. I think removing ``const&`` is fine, but I'm not 100% sure what to do

de-vri-es, 2016-08-23:
It might make more sense to have `KinematicOptionsMap::getOptions()` return the `KinematicOptions` by `const &` to avoid copying the `KinematicOptions` into a temporary. Since the `KinematicOptions` object is then no longer a temporary, returning a `const &`  to one of its members is safe.

On the other hand, if copying a `KinematicsOptions` object is not expensive, it doesn't matter that much and removing `const &` from `getKinematicsQueryOptions()` would also fix the issue. It's more common to return non-scalars by `const &` when possible though. It seems like this issue was introduced because that assumption was made for `KinematicOptionsMap::getOptions()`.

Neither of the options are ABI compatible with the current situation though.

davetcoleman, 2016-08-23:
@de-vri-es its not clear to me what route you are suggesting

andreaskoepf, 2016-08-23:
I opted for copy-by-value, because:
a) the type seemed to be tiny
b) There is a scope-lock in [kinematic_options_map.cpp#L54](because https://github.com/ros-planning/moveit/blob/4ac0c7432d335f57aab6836cbcaaac3fccf4b6f9/moveit_ros/robot_interaction/src/kinematic_options_map.cpp#L54). Probably only to protect the map-state .. but since `robot_interaction::KinematicOptionsMap::getOptions()` returned by value I thought that might have been done deliberately.

de-vri-es, 2016-08-23:
Ah, I missed the locking. It seems reasonable to assume the class does not want to leak references then. So ignore my comment.

davetcoleman, 2016-08-26:
Fixed

closed 2016-08-26 by davetcoleman