moveit/moveit #6: Migrate remaining Pull Requests from old repos

davetcoleman, 2016-07-29:
There are still a lot of pull requests in various old MoveIt repos, especially moveit_ros:

https://github.com/ros-planning/moveit_core/pulls
https://github.com/ros-planning/moveit_ros/pulls
https://github.com/ros-planning/moveit_plugins/pulls
https://github.com/ros-planning/moveit_commander/pulls
https://github.com/ros-planning/moveit_ikfast/pulls

If you are up to it - rebase, cleanup, and reopen those pull requests to the new repo.

Note: do not claim this issue - it is a meta-issue containing a lot of work. Instead claim the particular pull request you want to work on on the corresponding old PR.

felixvd, 2020-04-16:
Can this be closed?

v4hn, 2020-04-16:
https://github.com/ros-planning/moveit_ros/pulls still seems to include interesting patches:

- OcTree-related:
  - [ ] https://github.com/ros-planning/moveit_ros/pull/636
  - [ ] https://github.com/ros-planning/moveit_ros/pull/599
  - [ ] https://github.com/ros-planning/moveit_ros/pull/597
  - [ ] https://github.com/ros-planning/moveit_ros/pull/593
- Padding-related code to access padding-parameters via yaml:
  - [ ] https://github.com/ros-planning/moveit_ros/pull/509
- random bugfix in code barely used:
  - [ ] https://github.com/ros-planning/moveit_ros/pull/578

Maybe @jschleicher or @peci1 are interested in any of them?

Since the repositories are archived, there seems to be no way to close considered requests anymore. So I'm not sure whether any of these have been addressed in the meanwhile one way or another.

peci1, 2020-04-16:
Nothing on my side...