moveit/moveit #8: Build from source instructions do not work

VictorLamoine, 2016-07-30:
Ubuntu 16.04, ROS Kinetic desktop full installed:

```bash
victor@X301A1:~/libraries/catkin_workspace/src$ rosdep install --from-paths . --ignore-src --rosdistro kinetic
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_core: Cannot locate rosdep definition for [fcl]
moveit_planners_ompl: Cannot locate rosdep definition for [ompl]
moveit_ros_manipulation: Cannot locate rosdep definition for [manipulation_msgs]
moveit_commander: No definition of [python-pyassimp] for OS version [xenial]
```

```bash
$ dpkg -l | grep ros-kinetic-desktop
ii  ros-kinetic-desktop                                         1.3.0-0xenial-20160614-202450-0700                          amd64        A metapackage to aggregate several packages.
ii  ros-kinetic-desktop-full                                    1.3.0-0xenial-20160614-203433-0700                          amd64        A metapackage to aggregate several packages.
```

dbolkensteyn, 2016-07-30:
The build indeed is failing, and it looks like there are 2 immediate ways to fix it:

1. build fcl from sources: you need to add a `package.xml` as described here: https://github.com/ros-planning/moveit_core/issues/313
2. install the pre-released fcl 0.5.0 debian package as done here: https://github.com/davetcoleman/moveit_kinetic_cpp11/blob/master/.travis.before_script

The debian package seem to be on their way according to https://github.com/flexible-collision-library/fcl/issues/137 so a little bit of patience might be the easiest way.

VictorLamoine, 2016-07-30:
I installed the pre-compiled debians and there seems to be more missing:

```cmake
__________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << moveit_msgs:cmake /home/victor/libraries/catkin_workspace/logs/moveit_msgs/build.cmake.000.log                                                                                                
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "object_recognition_msgs" with any of the following names:

    object_recognition_msgsConfig.cmake
    object_recognition_msgs-config.cmake

  Add the installation prefix of "object_recognition_msgs" to
  CMAKE_PREFIX_PATH or set "object_recognition_msgs_DIR" to a directory
  containing one of the above files.  If "object_recognition_msgs" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:14 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "object_recognition_msgs" with any of the following names:

    object_recognition_msgsConfig.cmake
    object_recognition_msgs-config.cmake

  Add the installation prefix of "object_recognition_msgs" to
  CMAKE_PREFIX_PATH or set "object_recognition_msgs_DIR" to a directory
  containing one of the above files.  If "object_recognition_msgs" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:14 (find_package)

```

davetcoleman, 2016-07-30:
@dbolkensteyn answers are correct. If you mimmick the way travis is setting up the kinetic moveit code, it will work: https://github.com/davetcoleman/moveit_kinetic_cpp11

> moveit_core: Cannot locate rosdep definition for [fcl]

You can ignore that for now, but see https://github.com/ros-planning/moveit_core/pull/321. This depends on how fcl is added to ROS ecosystem

> moveit_planners_ompl: Cannot locate rosdep definition for [ompl]
> moveit_ros_manipulation: Cannot locate rosdep definition for [manipulation_msgs]
> moveit_commander: No definition of [python-pyassimp] for OS version [xenial]

Either you didn't do shadow-fixed, didn't apt-get update, or didn't rosdep update

davetcoleman, 2016-08-01:
I updated the build instructions on the README.md yesterday, they now use the experimental debians. Let me know if they don't work for you

davetcoleman, 2016-08-07:
New install instructions are now live on the website: http://moveit.ros.org/install/source_install.html

I've tested the Kinetic instructions multiple times on my laptop, via Travis, and via Docker. Closing, reopen if issues persist

scottpaulin, 2016-08-08:
I'm getting (different) build errors following the source instructions. 

Ubuntu 16.04
Kinetic binaries

I had permissions issues with some of the instructions in .travis.before_script. I 'fixed' them by putting sudo in front of all apt-get and dpkg commands. 
[Modified travis before script](https://github.com/ros-planning/moveit/files/406575/travis_before_script.txt)

The before script then has issues locating fcl packages ([full output](https://github.com/ros-planning/moveit/files/406576/travis_before_script_output.txt)):
E: Unable to locate package libfcl0.5_0.5.0-1osrf1~xenial1_amd64.deb
E: Couldn't find any package by glob 'libfcl0.5_0.5.0-1osrf1~xenial1_amd64.deb'
E: Couldn't find any package by regex 'libfcl0.5_0.5.0-1osrf1~xenial1_amd64.deb'
E: Unable to locate package libfcl0.5_0.5.0-1osrf1~xenial1_amd64.deb.1
E: Couldn't find any package by glob 'libfcl0.5_0.5.0-1osrf1~xenial1_amd64.deb.1'
E: Couldn't find any package by regex 'libfcl0.5_0.5.0-1osrf1~xenial1_amd64.deb.1'
E: Unable to locate package libfcl0.5-dbg_0.5.0-1osrf1~xenial1_amd64.deb
E: Couldn't find any package by glob 'libfcl0.5-dbg_0.5.0-1osrf1~xenial1_amd64.deb'
E: Couldn't find any package by regex 'libfcl0.5-dbg_0.5.0-1osrf1~xenial1_amd64.deb'
E: Unable to locate package libfcl0.5-dbg_0.5.0-1osrf1~xenial1_amd64.deb.1
E: Couldn't find any package by glob 'libfcl0.5-dbg_0.5.0-1osrf1~xenial1_amd64.deb.1'
E: Couldn't find any package by regex 'libfcl0.5-dbg_0.5.0-1osrf1~xenial1_amd64.deb.1'
E: Unable to locate package libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb
E: Couldn't find any package by glob 'libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb'
E: Couldn't find any package by regex 'libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb'
E: Unable to locate package libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb.1
E: Couldn't find any package by glob 'libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb.1'
E: Couldn't find any package by regex 'libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb.1'

....

And some errors installing fcl:
dpkg: error processing archive libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb (--install):
 conflicting packages - not installing libfcl-0.5-dev
Errors were encountered while processing:
 libfcl-0.5-dev_0.5.0-1osrf1~xenial1_amd64.deb

After running the wstool commands I removed fcl from moveit_core/package.xml, run the rosdep install command and then put the fcl dependencies back into moveit_core/package.xml. 

catkin build then results in [errors with calls to fcl from moveit_core](https://github.com/ros-planning/moveit/files/406592/build_errors.txt)

I don't know much about dependency resolution, dpkg e.t.c but here is my $0.02:

- I am not sure why my computer should require sudo permissions but your laptop doesn't @davetcoleman  unless we are using different operating systems? Docker doesn't usually seem to require sudo permissions for stuff like apt-get. 

- My best guess is that the build errors are because the required libraries are not found when the before travis script is executed. Why are there three different wget commands?

davetcoleman, 2016-08-08:
Good point about the sudo, mind making a PR for that? Docker is already root so doesn't require it. For my personal laptop I actually installed FCL from source, so I actually didn't fully test the Kinetic instructions for that.

From your logs it looks like ``libfcl-0.5-dev`` is still installed despite the command ``apt-get -qq remove libfcl*``. Manually uninstall it.

There are three wget because there are three debs

Protip: use https://gist.github.com/ when sharing text files, so I don't have to download stuff

scottpaulin, 2016-08-09:
> From your logs it looks like libfcl-0.5-dev is still installed despite the command apt-get -qq remove libfcl*. Manually uninstall it.

Nailed it! Builds fine. I managed to remove libfcl-0.5-dev with 
`sudo apt-get remove libfcl*`
 so I am wondering if the -qq option impeded it somehow?

From the [man-pages](http://linux.die.net/man/8/apt-get) (bolding mine):

> Quiet. Produces output suitable for logging, omitting progress indicators. More q's will produce more quiet up to a maximum of two. You can also use -q=# to set the quiet level, overriding the configuration file. Note that quiet level 2 implies -y, **you should never use -qq without a no-action modifier such as -d, --print-uris or -s as APT may decided to do something you did not expect**. 

Maybe some sort of no-action modifier is required in the command?

As a side note I still get the unable to locate package errors listed above.

davetcoleman, 2016-08-09:
Maybe it needs the ``-y`` or ``-f`` command?

scottpaulin, 2016-08-10:
Yip! Just tested it with -y and it worked. There were also no error messages. I have made a PR.

scottpaulin, 2016-08-12:
I think this can be closed if you are happy @VictorLamoine @davetcoleman

v4hn, 2016-08-12:
Closing. Feel free to comment below if you still experience problems.

closed 2016-08-12 by v4hn