moveit/moveit #10: move_group segfaults after multiple planning requests

dbolkensteyn, 2016-08-01:
I have a [simple URDF for a robotic arm](https://github.com/ros-planning/moveit/files/394210/arm.urdf.txt) for which I generate a MoveIt setup.

Then I run the demo using
```
roslaunch [generated package] demo.launch debug:=true
```

From there, if I run several times [usually less than 20] "Plan & Execute" to a random valid state, I end up seeing the move_group process segfault.

Thanks to the `debug:=true` parameter, I am able to see some details (on Kinetic):
```
[ INFO] [1470066134.494887876]: LBKPIECE1: Created 17 (8 start + 9 goal) states in 13 cells (7 start (7 on boundary) + 6 goal (6 on boundary))
[ INFO] [1470066134.496745447]: LBKPIECE1: Created 45 (19 start + 26 goal) states in 35 cells (18 start (18 on boundary) + 17 goal (17 on boundary))
[ INFO] [1470066134.498151876]: LBKPIECE1: Created 85 (39 start + 46 goal) states in 72 cells (37 start (37 on boundary) + 35 goal (35 on boundary))
[Thread 0x7fffb5d62700 (LWP 13802) exited]
[Thread 0x7fffb5561700 (LWP 13801) exited]
[Thread 0x7fffb4d60700 (LWP 13800) exited]

Thread 41 "move_group" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffa7fff700 (LWP 13799)]
0x00007fffc95a99ad in ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors(std::vector<int, std::allocator<int> >&, std::vector<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >&) const () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
(gdb) 
```

I could reproduce this crash on both Kinetic (built from sources) but also Indigo (using Docker).

I'm open to suggestions on how to investigate this further. I'll try to record the requests, and see if the move_group crash is always reproducible with a particular request. I also need to run with debug on on Indigo to compare the error messages.

Also, perhaps this isn't the best repo for this issue: I'll recreate it where appropriate once the root cause is identified.

davetcoleman, 2016-08-01:
This is the best place for new issues - we should start consolidating now.

Can you update your description above with a gdb backtrace? Just type in "bt" in the prompt after it segfaults

Also - compile your whole workspace in debug mode to get the best feedback

dbolkensteyn, 2016-08-02:
The package I run is the one generated by the MoveIt setup assistant, so it only consists of a bunch of launch and config files.

On Indigo using Docker and so a release build, there is nothing of interest:
```
[ INFO] [1470129271.515227848]: ParallelPlan::solve(): Solution found by one or more threads in 0.033511 seconds
[ INFO] [1470129271.515600765]: LBKPIECE1: Attempting to use default projection.
[ INFO] [1470129271.515733742]: LBKPIECE1: Starting planning with 1 states already in datastructure
[ INFO] [1470129271.515785555]: LBKPIECE1: Attempting to use default projection.
[ INFO] [1470129271.515846783]: LBKPIECE1: Attempting to use default projection.
[ INFO] [1470129271.515931512]: LBKPIECE1: Starting planning with 1 states already in datastructure
[ INFO] [1470129271.515996117]: LBKPIECE1: Attempting to use default projection.
During startup program terminated with signal SIGSEGV, Segmentation fault.
(gdb) bt
No stack.
(gdb) quit
[move_group-4] process has finished cleanly
log file: /home/dinesh/.ros/log/5e7158ec-5891-11e6-94b0-0242ac110002/move_group-4*.log
```

On Kinetic with the backtrace output:
```
[ INFO] [1470133362.311205981]: LBKPIECE1: Starting planning with 1 states already in datastructure
[ INFO] [1470133362.312628623]: LBKPIECE1: Created 47 (20 start + 27 goal) states in 37 cells (19 start (19 on boundary) + 18 goal (18 on boundary))
[Thread 0x7fffba563700 (LWP 2392) exited]

Thread 166 "move_group" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb9d62700 (LWP 2391)]
0x00007fffc9cda9ad in ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors(std::vector<int, std::allocator<int> >&, std::vector<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >&) const () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
(gdb) bt
#0  0x00007fffc9cda9ad in ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors(std::vector<int, std::allocator<int> >&, std::vector<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >&) const () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#1  0x00007fffc9cdada5 in ompl::GridB<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*, ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::OrderCellsByImportance, ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::OrderCellsByImportance>::createCell(std::vector<int, std::allocator<int> > const&, std::vector<ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >*) () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#2  0x00007fffc9cdcedd in ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::addMotion(ompl::geometric::LBKPIECE1::Motion*, std::vector<int, std::allocator<int> > const&, double) () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#3  0x00007fffc9cd6f78 in ompl::geometric::LBKPIECE1::solve(ompl::base::PlannerTerminationCondition const&) () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#4  0x00007fffc9dd3a59 in ompl::tools::ParallelPlan::solveMore(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*) ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#5  0x00007ffff5d48c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff621e6fa in start_thread (arg=0x7fffb9d62700) at pthread_create.c:333
#7  0x00007ffff57b7b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 
```

dbolkensteyn, 2016-08-02:
Looks like it might relate to the `boost::shared_ptr` vs `std::shared_ptr` story discussed in https://github.com/ros-planning/moveit_core/issues/315

If so, it'd mean that the Indigo crash has a different root cause.

dbolkensteyn, 2016-08-02:
Argh ompl looks like a pain to work with :)

An official repo using Mercurial on Bitbucket... a 150MB repo git mirror on Github, a `TODO` file and a documentation website in the repo.

I'm wondering at the moment how I could rebuild the `ros-kinetic-ompl` package in Debug configuration to get more informations.

I had a look at https://github.com/ros-gbp/ompl-release but it's not very clear how that works.

v4hn, 2016-08-02:
On Tue, Aug 02, 2016 at 07:57:00AM -0700, Dinesh Bolkensteyn wrote:
> I'm wondering at the moment how I could rebuild the `ros-kinetic-ompl` package in Debug configuration to get more informations.

Where's the problem? You can download the tarball [here](http://ompl.kavrakilab.org/download.html) and
build the project via cmake with `-DCMAKE_INSTALL_PREFIX=<somewhere safe> -DCMAKE_BUILD_TYPE=Debug`, right?

dbolkensteyn, 2016-08-02:
Indeed @v4hn - I've just built ompl like that and uninstalled `ros-kinetic-ompl` just to be sure the local build would be used.

I've grabbed the latest 1.2.1 release sources. I could build the project, but some 2 tests are failing:
```
89% tests passed, 2 tests failed out of 18

Total Test time (real) = 250.84 sec

The following tests FAILED:
	  6 - test_random (Failed)
	 13 - test_2denvs_geometric (Failed)
Errors while running CTest
```

I happily ignored that, and proceeded with the MoveIt test :) There is now a ton more info in gdb:
```
Thread 85 "move_group" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb0ada700 (LWP 21063)]
0x00007fffbb834dd7 in ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors (this=0x7fffac010aa0, coord=std::vector of length 0, capacity 0, 
    list=std::vector of length 0, capacity 4) at /home/dinesh/Downloads/ompl-1.2.1-Source/src/ompl/datastructures/Grid.h:150
150	                coord[i]--;
(gdb) bt
#0  0x00007fffbb834dd7 in ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors (this=0x7fffac010aa0, 
    coord=std::vector of length 0, capacity 0, list=std::vector of length 0, capacity 4) at /home/dinesh/Downloads/ompl-1.2.1-Source/src/ompl/datastructures/Grid.h:150
#1  0x00007fffbb832723 in ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors (this=0x7fffac010aa0, 
    coord=std::vector of length 0, capacity 0, list=std::vector of length 0, capacity 0) at /home/dinesh/Downloads/ompl-1.2.1-Source/src/ompl/datastructures/GridN.h:155
#2  0x00007fffbb830726 in ompl::GridB<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*, ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::OrderCellsByImportance, ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::OrderCellsByImportance>::createCell (this=0x7fffac010aa0, coord=std::vector of length 0, capacity 2, 
    nbh=0x0) at /home/dinesh/Downloads/ompl-1.2.1-Source/src/ompl/datastructures/GridB.h:177
#3  0x00007fffbb82ead9 in ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::addMotion (this=0x7fffac010aa0, motion=0x7fffa001fd10, 
    coord=std::vector of length 0, capacity 2, dist=0) at /home/dinesh/Downloads/ompl-1.2.1-Source/src/ompl/geometric/planners/kpiece/Discretization.h:213
#4  0x00007fffbb82ce35 in ompl::geometric::LBKPIECE1::solve (this=0x7fffac00f470, ptc=...) at /home/dinesh/Downloads/ompl-1.2.1-Source/src/ompl/geometric/planners/kpiece/src/LBKPIECE1.cpp:137
#5  0x00007fffbb9d86d8 in ompl::tools::ParallelPlan::solveMore (this=0x7fffac004ba8, planner=0x7fffac00f470, minSolCount=1, maxSolCount=10, ptc=0x7fffb8f9f630)
    at /home/dinesh/Downloads/ompl-1.2.1-Source/src/ompl/tools/multiplan/src/ParallelPlan.cpp:159
#6  0x00007fffbb9dc524 in std::_Mem_fn_base<void (ompl::tools::ParallelPlan::*)(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*), true>::operator()<ompl::base::Planner*&, unsigned long&, unsigned long&, ompl::base::PlannerTerminationCondition const*&, void> (this=0x7fffac007728, __object=0x7fffac004ba8)
    at /usr/include/c++/5/functional:600
#7  0x00007fffbb9dc309 in std::_Bind<std::_Mem_fn<void (ompl::tools::ParallelPlan::*)(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> (ompl::tools::ParallelPlan*, ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)>::__call<void, , 0ul, 1ul, 2ul, 3ul, 4ul>(std::tuple<>&&, std::_Index_tuple<0ul, 1ul, 2ul, 3ul, 4ul>) (this=0x7fffac007728, __args=<unknown type in /home/dinesh/Desktop/btb/cmake_repo/lib/libompl.so.12, CU 0x1c46ca0, DIE 0x1c75d2c>)
    at /usr/include/c++/5/functional:1074
#8  0x00007fffbb9dbf0f in std::_Bind<std::_Mem_fn<void (ompl::tools::ParallelPlan::*)(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> (ompl::tools::ParallelPlan*, ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)>::operator()<, void>() (this=0x7fffac007728)
    at /usr/include/c++/5/functional:1133
#9  0x00007fffbb9dbe7c in std::_Bind_simple<std::_Bind<std::_Mem_fn<void (ompl::tools::ParallelPlan::*)(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> (ompl::tools::ParallelPlan*, ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> ()>::_M_invoke<>(std::_Index_tuple<>) (
    this=0x7fffac007728) at /usr/include/c++/5/functional:1531
#10 0x00007fffbb9dbd40 in std::_Bind_simple<std::_Bind<std::_Mem_fn<void (ompl::tools::ParallelPlan::*)(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> (ompl::tools::ParallelPlan*, ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> ()>::operator()() (this=0x7fffac007728)
    at /usr/include/c++/5/functional:1520
#11 0x00007fffbb9dbc4a in std::thread::_Impl<std::_Bind_simple<std::_Bind<std::_Mem_fn<void (ompl::tools::ParallelPlan::*)(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> (ompl::tools::ParallelPlan*, ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*)> ()> >::_M [... cut ...]
152	                iterator pos = hash_.find(&coord);
153	                Cell *cell = (pos != hash_.end()) ? pos->second : nullptr;
154	
(gdb) 
155	                if (cell)
156	                    list.push_back(cell);
157	                coord[i] += 2;
158	
159	                pos = hash_.find(&coord);
160	                cell = (pos != hash_.end()) ? pos->second : nullptr;
161	
162	                if (cell)
163	                    list.push_back(cell);
164	                coord[i]--;
(gdb) 
```

v4hn, 2016-08-02:
Looks like you hit a bug in OMPL's kpiece algorithm.
I do not think this has anything to do with c++11.
@davetcoleman is our expert for this, so I'll leave this to him.

I suppose it would be a good idea to send that backtrace to the ompl devs and ask there for hints.

davetcoleman, 2016-08-02:
Looks like you've entered THE GRID ;-)

Great debugging, thanks. Do you get the same kind of problem if you change solvers to something like RRT? Testing this will help determine if the error is how we are using OMPL or within OMPL itself.

davetcoleman, 2016-08-02:
I created a moveit_config package for your robot and tested it on 16.04/Kinetic... it worked fine for me!

[image]

Perhaps the way you are installing OMPL is wrong - I added some [documentation](https://github.com/ros-planning/moveit_planners/pull/80/files#diff-7de2256022192afba8845b582f3c0323R33) for how to build with catkin

v4hn, 2016-08-02:
@davetcoleman installation of the upstream tarball should work equally well, although in-workspace-installation does have its advantages.
I also experienced move_group segfaults due to ompl before, but they where never reproducible.
If you have a more-or-less deterministic bug @dbolkensteyn it would be great if you could track it down and report to the responsible party :)

davetcoleman, 2016-08-02:
One new way of reporting bugs that I haven't tried yet but would be really cool: create a reproducible setup in a Docker, push the container to your own Dockerhub account, then share with us. You can build the original container off of MoveIt's pre-configured ones...

dbolkensteyn, 2016-08-02:
I'll try to do that tomorrow @davetcoleman 

I've tried various algorithms as you suggested, and I'm only able to reproduce the crash when I leave it on `<unspecified>`:
[image]

In that case, in the console I see messages in this form:
```
[ INFO] [1470170297.188631440]: LBKPIECE1: Created 23 (12 start + 11 goal) states in 21 cells (11 start (11 on boundary) + 10 goal (10 on boundary))
```

If I explicitly select the `LBKPIECEkConfigDefault` algorithm, then the messages are in this form:
```
[ INFO] [1470170278.715947544]: arm[LBKPIECEkConfigDefault]: Created 23 (10 start + 13 goal) states in 17 cells (9 start (9 on boundary) + 8 goal (8 on boundary))
```
and I am then not able to reproduce the issue.

I've also noticed that all I need to reproduce this problem with `<unspecified>` selected is to hit `Plan` many times: There is no need to update the query goal to a new random valid position each time.

v4hn, 2016-08-02:
Now that sounds like we should review the "default planner" implementation,
that should probably be streamlined anyway.
I believe the feature to specify a default planner via yaml was added quite late
and might not even be respected everywhere...

Sounds like you made good progress on isolating the issue!

davetcoleman, 2016-08-03:
Sounds like an issue with the Rviz plugin...

davetcoleman, 2016-09-16:
@dbolkensteyn any progress with this issue?

v4hn, 2017-01-13:
The code for the default planner selection was reworked in the meanwhile.
I will assume this fixed the bug (unless someone comes along and reports it again).

closed 2017-01-13 by v4hn