moveit/moveit #11: Switch Sphynx Documentation Visual Theme

davetcoleman, 2016-08-02:
Current one is visually outdated, would be nice to use something similar to how ReadTheDocs appears. A great starting point would be to copy Fetch's version of the ReadTheDocs theme https://github.com/fetchrobotics/docs/tree/master/source/_themes/sphinx_rtd_theme

The MoveIt! logo uses color #2ab2e6 so would be nice to sprinkle that in.

A huge bonus is that it would give us the "Edit On Github" button for free (top right [here](http://docs.fetchrobotics.com/))

IanTheEngineer, 2016-08-19:
+1 for ReadTheDocs theme. It's modern, clean, and easy to navigate.

davetcoleman, 2016-10-27:
I completed this, see http://docs.ros.org/kinetic/api/moveit_tutorials/html/

closed 2016-10-27 by davetcoleman