moveit/moveit #12: Create an overview page over plugin mechanism

v4hn, 2016-08-03:
MoveIt has quite a number of plugin mechanisms.
They *should* be the primary interface for new work in a number of sub-fields handled by MoveIt,
but we are missing documentation on them.
These are (In no particular order):

- [ ] `move_group::MoveGroupCapability`
- [ ] `occupancy_map_monitor::OccupancyMapUpdater`
- [ ] `planning_request_adapter::PlanningRequestAdapter`
- [ ] `kinematics::KinematicsBase`
- [ ] `planning_interface::Planner`
- [ ] `planning_interface::PlannerManager`
- [ ] `moveit_controller_manager::MoveItControllerManager`
- [ ] `moveit_ros_control_interface::ControllerHandleAllocator`
- [ ] `moveit_sensor_manager::MoveItSensorManager`
- [ ] `constraint_samplers::ConstraintSamplerAllocator`
- [ ] `collision_detection::CollisionPlugin`

and in kinetic-devel also
- [ ] `warehouse_ros::DatabaseConnection`

Some of them are mentioned in passing on http://moveit.ros.org/documentation/concepts/ ,
but most are not explained anywhere except for their class definition in the code.
It would be great to have a web page "Plugin interfaces" in the "Documentation" tab on moveit.ros.org with a short block of documentation on each of them:
- What is it used for?
- What can I achieve by implementing a plugin of that type?
- What does the resp. plugin interface look like (including a link to the base class header)
- Where can an example implementation be found?

This involves quite a bit of code research on the interfaces and is obviously too much for one person to do, so the idea is that you can just add a note here saying which plugin you wish to document and file a pull request against https://github.com/ros-planning/moveit.ros.org 's request-plugins-explained branch explaining the plugin [here](https://github.com/ros-planning/moveit.ros.org/blob/request-plugins-explained/documentation/plugins/index.markdown).

alemme, 2016-08-23:
Hi,

I like to contribute here :)

Cheers,
Andre

JHoef, 2016-08-23:
I join you!

v4hn, 2016-11-22:
This has been created on the website: http://moveit.ros.org/documentation/plugins/

closed 2016-11-22 by v4hn