moveit/moveit #13: make MoveIt's RViz display properly resizable

v4hn, 2016-08-03:
One of the most annoying everyday MoveIt issues many people experience is that the [motion_planning display](https://github.com/ros-planning/moveit_ros/tree/indigo-devel/visualization/motion_planning_rviz_plugin) in RViz has a huge minimal size.
While everyone using RViz in fullscreen mode on a 24'' monitor does not notice this problem,
this can be a serious annoyance when working on a laptop screen.
This can likely be changed by placing either the QTabWidget or all subwidgets in 2D scrollable Qt Widget.

dornhege, 2016-08-04:
Huge +1!

v4hn, 2016-08-11:
while dornhege implemented this in the referenced request. The changes have to be ported to the new repo, tested, a screenshot should be added, and the patch possibly improved (see the comments there).

yoshimalucky, 2019-11-20:
Directly pasting gdb error log which include #XX (XX means number) makes us confused. Only #42 is related with this issue...

As @davetcoleman mentioned, now I'm thinking about how to modify the GUI property for auto scaling, and now I know that the minimum size of the dspinboxes in Scene Objects tab block shrinking the entire display size of motion_planning_display.