moveit/moveit #14: add "<previous>" robot state to RViz motion display

v4hn, 2016-08-03:
In some situations it would be nice to have a "<previous>" entry in the Update Start/Goal State drop-down list, to set the start/goal state of the display to the previous pose the robot was in (before the latest motion). This can be used e.g. to look for alternative paths for the latest robot motion.

This could be implemented by adding a new `robot_state::RobotState previous_state_` member to the `MotionPlanningDisplay` that is updated after each successful execution (as well as plan&execute) and might be used by `MotionPlanningFrame::updateQueryStateHelper` to set a query state.

takarakasai, 2019-11-20:
I am trying for this issue at moveit day 2019.

rhaschke, 2019-11-23:
Addressed in #1742.

PR #1742 (2019-11-23): Add "<previous>" support robot state to RViz motion display #14.
### Description

Add an entry "<previous"> in the Update Start/Goal State drop-down list for referencing a previous pose. 

Fixes #14 

[image]

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [x] Include a screenshot if changing a GUI
closed 2019-11-23 by rhaschke