moveit/moveit #15: Need help porting CHOMP to latest Moveit-ROS.

ksatyaki, 2016-08-04:
I recently started trying to port the CHOMP planner to the latest moveit API. 
I have worked a bit on (https://github.com/jhu-lcsr-forks/moveit_planners). For now, the moveit plugin for planning compiles and runs with moveit. Please find my fork here: https://github.com/ksatyaki/moveit_planners

However, I need some insight to figure out why I get this error.

Could not initialize hybrid collision world from planning scene
Could not initialize optimizer

These are errors generated when creating an optimizer instance.

I tried running it on a jaco2 arm.I didn't configure a kinect / 3d sensor. I first wanted to try it without any obstacles. Am I missing something? 

Please let me know if you need more information. Any help is highly appreciated.

My launch files are at:
https://github.com/ksatyaki/jaco_chomp_launch
https://github.com/ksatyaki/manipulation_launch

I run the manipulation.launch from jaco_chomp_launch which basically loads move_group with a different planning pipeline.

ksatyaki, 2016-08-04:
### Update

I discovered that the "planning_scene->getCollisionWorld().get()" is returning a NULL pointer. I don't understand it completely. But, I could work around the problem by adding a new CollisionDetectorHybrid to the planning scene and calling the setActiveCollisionDetector() to set it to active.

However, now I encounter the following problem:
> No link state found for link jaco_link_finger_1

I am guessing that this has to do something with the urdf or srdf.

Ry0, 2016-08-04:
I'm also facing that problem.
I just tried your `moveit_planner` (https://github.com/ksatyaki/moveit_planners).
The following errors are not happened.

```
Could not initialize hybrid collision world from planning scene
Could not initialize optimizer
```

But, The following description is output.
 
```
No link state found for link hoge
```
After the output, `move_group` process died.

I have a question.
Do you use `ros-planning/moveit_experimental` for using `collision_distance_field` package?
(https://github.com/ros-planning/moveit_experimental)

jrgnicho, 2016-08-04:
@ksatyaki are you implementing a [`planning_interface::PlannerManager`](http://docs.ros.org/indigo/api/moveit_core/html/classplanning__interface_1_1PlannerManager.html) and ```planning_interface::PlanningContext```. These classes define the planning workflow which involves initializing the planning scene before calling the __solve__ method on the planner.  Take a look at the [stomp_planner_manager.h and stomp_planner.h](https://github.com/ros-industrial/industrial_moveit/tree/indigo-devel/stomp_moveit/include/stomp_moveit) if you need an example of this

davetcoleman, 2016-08-04:
@jbohren has worked with CHOMP and moveit_experimental, see https://github.com/davetcoleman/moveit/pull/2

Would be great to get that PR or your work merged into MoveIt soon

@jrgnicho can probably advise on this subject also

ksatyaki, 2016-08-04:
@jrgnicho Thanks so much. I will get back soon after I look at the stomp. :)

@davetcoleman Thanks again!

@Ry0 I am using @jrgnicho's moveit_experimental ... This one is ported to the new moveit API. And to catkin ...

ksatyaki, 2016-08-05:
While setting up a planning scene, is it enough to set it up for the planner like it is done in Stomp? Or should one use the PlanningSceneMonitor read-write class? What is the difference?

v4hn, 2016-08-05:
On Fri, Aug 05, 2016 at 02:49:53AM -0700, Chittaranjan Srinivas Swaminathan wrote:
> While setting up a planning scene, is it enough to set it up for the planner like it is done in Stomp?
> Or should one use the PlanningSceneMonitor read-write class? What is the difference?

See http://moveit.ros.org/documentation/concepts/ on Planning Scene.
If you like to have the scene updated with the robot's joint states, etc. you will need a monitor.

jrgnicho, 2016-08-05:
@ksatyaki There is a [stomp_test_kr_moveit_config](https://github.com/ros-industrial/industrial_moveit/tree/indigo-devel/stomp_test_kr210_moveit_config) package that you can use for your testing.  It's already configured to load the stomp planner instead of ompl.  Essentially, you must create a planning_plugin for your planner using the pluginlib library as shown [here](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/stomp_moveit/src/stomp_planner_manager.cpp#L147) and by setting up a [plugin.xml](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/stomp_moveit/planner_manager_plugins.xml) file with the plugin information.  Then, in your moveit_config package, you define a [your_planner_name]_moveit_planning_pipeline.launch file ([see example](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/stomp_test_kr210_moveit_config/launch/stomp_moveit_planning_pipeline.launch.xml)) and then have the [move_group.launch](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/stomp_test_kr210_moveit_config/launch/move_group.launch#L17) file load it by setting the *pipeline* parameter to your planner's name.  
Since the planner is loaded by the __move_group__ node I don't think there's any need to maintain a planning scene monitor inside your planner,  in fact move_group should ensured that the planning_scene pointer gets passed to the planner before requesting a to solve a motion plan.

ksatyaki, 2016-08-07:
I have tried using the config. I still get a similar error. 

> No link state found for link gripper_body

I have tried commenting the return statement when a link state is not found. It appears that not all links have a corresponding link state. I am not entirely sure why this is the case or how this works. But when I do that, I get the error: 

> No link body decomposition for link workcell_bounds

I am wondering if the code in [collision_robot_distance_field.cpp](https://github.com/jrgnicho/moveit_experimental/blob/indigo-devel/collision_distance_field/src/collision_robot_distance_field.cpp) needs some update.

jrgnicho, 2016-08-08:
@ksatyaki the code in moveit_experimental was copied into [industrial_moveit](https://github.com/jrgnicho/industrial_moveit/tree/indigo-devel/moveit_experimental) where it went through some further development. You should try switching to that repo instead.  It be helpful if you were to provide a more detailed description of the workflow that you follow that exposes these issues.

ksatyaki, 2016-08-08:
@jrgnicho Oh, thanks. It looks like the last commit on the other repo pertains to my error.

I am starting the "demo.launch" for the config you gave me, but with the chomp plugin. Then I issue a simple goal using rviz. Once the planner receives the goal, the error occurs.

jrgnicho, 2016-08-08:
@ksatyaki yes it looks like the commit may address that "No link body decomposition ... " issue you saw.  See if you still get it after switching to the newer `moveit_experimental' code.

ksatyaki, 2016-08-11:
After trying to use the moveit_experimental package from industrial_moveit, I still get the error

> No link body decomposition for link workcell_bounds

ksatyaki, 2016-08-15:
I have an update on where the error happens and why. 

When I add a collision detector as mentioned in the previous post, it simply creates an empty [CollisionRobotDistanceField](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/moveit_experimental/collision_distance_field/src/collision_robot_distance_field.cpp) using the first constructor. So the `link_body_decomposition_vector_ `is not initialized. 

I am wondering how one could initialize this class properly or add a Collision Detector properly.

ksatyaki, 2016-08-16:
It now seems like `link_body_decomposition_vector_` is indeed initialized, but when the method 
```
PosedBodyPointDecompositionPtr
CollisionRobotDistanceField::getPosedLinkBodyPointDecomposition(const moveit::core::LinkModel* ls) const 
``` is called, it is empty! I don't understand what is happening. 
I also tried calling the constructor (which again calls `addLinkBodyDecompositions` method and I can see that the models are indeed being generated and the size of the map is not zero when it completes) from within the `getPosedLinkBodyDecomposition()` method if the map happens to be null. But once it returns the size of the map is still 0. :-(

jrgnicho, 2016-08-17:
Make sure that the code ever reaches [this point](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/moveit_experimental/collision_distance_field/src/collision_robot_distance_field.cpp#L1069) where it creates a _spherical decomposition_ of the link.
If it does then it could be that the link for which you are getting the bd (body decomposition) has no geometry in which case no bd would've been generated.
Can you do any planning?

ksatyaki, 2016-08-17:
I can't do any planning yet. The execution does indeed reach that point. I have tried to print the size of the vector and map when that function `addLinkBodyDecompositions` ends. And it is not zero. It's profoundly confounding!

jrgnicho, 2016-08-17:
Which link is it complaining about? Does it have a geometry or is it an empty link?

ksatyaki, 2016-08-18:
@jrgnicho It complains about the "workcell_bounds" link. I don't know how I could easily verify if it's an empty link.

ksatyaki, 2016-08-18:
Here is a "fork" of the moveit_experimental package from the ros_industrial that I am using now for testing. For simplicity I have only moved the moveit_experimental package and not the entire stack. https://github.com/ksatyaki/moveit_experimental

jrgnicho, 2016-08-18:
You can look at the urdf and see if the link contains a **collision geometry**.  As far as I recall it should have one but it's been a while since I set that up.  Can you paste printout of the console output, it be helpful if you added some ROS_INFO for each body decomposition that it creates.

ksatyaki, 2016-08-22:
@jrgnicho Once again thanks for helping out.
[Here](https://docs.google.com/document/d/1BmUIij9Wp4g6Tz_1ELI0GSBpd7iam2crBpSqU7QTOE0/edit?usp=sharing) is the entire output from the current version of moveit_experimental and moveit_planners from my github. 

If you look at the latest commit on the [moveit_experimental](https://github.com/ksatyaki/moveit_experimental) package, you should see what is printed on the screen using the ROS_INFO calls.

The link _does_ contain collision geometry.

ksatyaki, 2016-08-22:
Just to make things easier if you are trying to run the code on your computer, I set up this repo with the files modified to work with the chomp planner: https://github.com/ksatyaki/chomp_test

davetcoleman, 2016-08-23:
Glad you're making progress on this. I'd suggest you create a preliminary PR so we maintainers can start giving you feedback and understanding where the changes are targeted to be made. It doesn't need to be merged yet if its not ready.

ksatyaki, 2016-08-24:
@davetcoleman I have merge my repos into the moveit repo preserving commit history. However I have ros-indigo. I am not able to get rid of the FCL errors even if I have geometric_shapes source. Uninstalling the deb version removes all the ros-indigo- packages. So I can't do that. Any suggestions?

davetcoleman, 2016-08-25:
> I am not able to get rid of the FCL errors even if I have geometric_shapes source. 

I don't know what errors you are referring to, should work fine in indigo. The trickery is only for kinetic. If your problem is with kinetic, I'd switch to shadow-fixed because geometric_shapes was just released

Travis CI will test if it compiles - I've put a lot of time into making it work well for MoveIt

davetcoleman, 2016-10-27:
This has been completed except documentation, issue: https://github.com/ros-planning/moveit/issues/237

closed 2016-10-27 by davetcoleman

BoltzmannBrain, 2018-01-24:
@ksatyaki I'm hoping you can help me here. I've been trying to extract the signed distance field (SDF) functionality from CHOMP -- I simply want to write a method `PlanningScene::CheckSDF` that gives me a vector of distance values from each robot sphere to its nearest obstacle. With the following code I came across this issue,

```
    const collision_detection::CollisionWorldHybrid* hy_world;
    const collision_detection::CollisionRobotHybrid* hy_robot;

    hy_world =
        dynamic_cast< const collision_detection::CollisionWorldHybrid* >(
            planning_scene_->getCollisionWorld(planning_scene_->getActiveCollisionDetectorName()).get());
    if (!hy_world) {
        std::cerr << "Could not initialize hybrid collision world from planning scene" << std::endl;
        return false;
    }

    hy_robot = dynamic_cast<const collision_detection::CollisionRobotHybrid*>(
        planning_scene_->getCollisionRobot(planning_scene_->getActiveCollisionDetectorName()).get()
    );
    if (!hy_robot) {
        std::cerr << "Could not initialize hybrid collision robot from planning scene" << std::endl;
        return false;
    }

    auto req = collision_detection::CollisionRequest();
    auto res = collision_detection::CollisionResult();
    moveit::core::RobotState state(robot_model_);
    collision_detection::GroupStateRepresentationPtr gsr;
    auto robot = *hy_robot->getCollisionRobotDistanceField().get();

    hy_world->checkCollisionDistanceField(req, res, robot, state, gsr);  // Passthrough to `cworld_distance_->checkCollision(req, res, robot, state, gsr);`

    // Debug lines to see what's going on:
    int num_collision_points_ = 0;
    std::map<std::string, std::string> fixed_link_resolution_map;
    std::cout << "Num grads = "<< gsr->gradients_.size() << std::endl;
    for (auto g = 0; g < gsr->gradients_.size(); g++)
    {
        num_collision_points_ += gsr->gradients_[g].gradients.size();
        collision_detection::GradientInfo& info = gsr->gradients_[g];
        std::vector<double> distances = info.distances;
        std::cout << distances.size() << std::endl;
    }

```

which gives me the `Could not initialize hybrid collision world from planning scene` error message. 

I'm using the collision_distance_field package from moveit_experimental. Any help you can offer me would be greatly appreciated :smile:

ksatyaki, 2018-01-25:
It's been a long time since I worked on this package. So it is possible that I might be of no help at all. I will try, nevertheless :-) 

I am not sure what you mean by "extract the signed distance field (SDF) functionality from CHOMP". Do you simply want to use the SDF Functions? 

If that is the case, take a look at [this repo](https://github.com/OrebroUniversity/grasping_oru/tree/devel). On the devel branch we have a sdf_tracker_grasping, sdf_tracker_msgs and sdf_collision_check. I suggest these, because they are maintained at our University at the moment and we can provide help in using these packages. These packages are not dependent on the rest of the grasping stack, so you should be able to use them separately. 

However, you might find in that repository how we use this SDF tracker along with the collision checker. Maybe the author of sdf_tracker, @dcanelhas, can provide some help too. 

For the issue at hand, I believe that there is something fishy with the moveit configuration files / parameters that leads to the error, "Could not initialize hybrid collision world from planning scene". But I am not sure what. 

Sorry if I haven't been of any help :)

BoltzmannBrain, 2018-01-25:
Thank you for the quick reply @ksatyaki! I'll look through the SDF functions in that repo and follow-up later. 

What I mean by "extract the SDF functionality from CHOMP", I mean that I'd like to use a sphere-robot model and a distance field to check for collisions during motion planning, but I'll be using a different optimization algorithm. Importantly, I need the distance field to provide a vector of distances at a given state, one value for each of the robot's spheres to the nearest obstacle. 

Regarding that error message, I believe it's due to a null pointer from getCollisionWorld, as mentioned earlier in this thread.

BoltzmannBrain, 2018-01-25:
I appreciate you sharing your grasping code @ksatyaki, but I'm constrained to use the PlanningScene class provided by MoveIt. It looks like the sdf_X code is all custom and independent from the MoveIt package. That is, I already have an initialized PlanningScene with pointcloud info, and all I need is a simple function that returns a 3-dimensional std::vector representing a distance field.

dcanelhas, 2018-01-26:
I noticed this issue is closed, but let me know if I can be of assistance. It seems if you are working with point cloud data, you probably don't want the full volumetric SDF representation just to get the distances. You're better off using eg nanoflann to build a kd tree in tiles and doing a nn lookup, and computing the point to plane distance.

ksatyaki, 2018-01-26:
@BoltzmannBrain I don't know why there is a null pointer returned. If I remember correctly, collision distance field worked fine when I tried it.

BoltzmannBrain, 2018-01-26:
@dcanelhas thank you for the tip. Would you be able to point me to some more info on that approach? Is it something included in ROS or MoveIt (I hope!)?

BoltzmannBrain, 2018-01-26:
@ksatyaki do you mean the code I posted, using collision distance field from moveit_experimental, would work fine? Regarding the null pointer, I'm referring to [your earlier comment in this thread](https://github.com/ros-planning/moveit/issues/15#issuecomment-237559048).

dcanelhas, 2018-01-26:
@BoltzmannBrain  have a look at [KdTreeFLANN](http://docs.ros.org/hydro/api/pcl/html/classpcl_1_1KdTreeFLANN.html) Assume you have a query point *q* and a obtain the nearest map point *p* with associated normal *n*, the (signed) point-to-plane distance is simply 
dot( (p-q), n)

khalilibrahim, 2019-04-11:
@ksatyaki I have problem when I install stomp planner (Moveit/Rviz) (indigo) no library loading in my  package.I found this error (process has died) as you see. anyone see this before to help me!!!
[image]