moveit/moveit #16: moveit_setup_assistant should require author details

v4hn, 2016-08-04:
catkin requires a maintainer and mail address to be specified in each package.xml.
Up to now the setup assistant does not ask for that information during setup, but just adds "assistant@moveit.ros.org" there. This is clearly wrong and people forget to change that value later on.
An issue for this was filed [here](https://github.com/ros-planning/moveit_setup_assistant/issues/110).

[This commit](https://github.com/v4hn/moveit_setup_assistant/commit/3181604e8bf203fe49efd5e02f653006b232b841) implements basic support for configuration of the contact details in the GUI, but it is not yet ready for merging.

- [ ] Check for valid mail address
- [ ] Make sure the setup assistant can load the new parameters from a config already generated
- [ ] Prohibit generation of the config when the contact details are not (yet) filled in

jan0e, 2016-08-23:
I started working on this and implemented functionality for bullet point 1 and 3. Will also look into point 2.

davetcoleman, 2016-10-26:
This has been completed

closed 2016-10-26 by davetcoleman