moveit/moveit #22: Jade Release

davetcoleman, 2016-08-06:
@130s I think we need to release the jade packages again to bring in https://github.com/ros-planning/moveit_ros/pull/699, right?

At that point, after the next ROS sync, I think we can consider jade released right? We'll want the new ``ros-jade-moveit`` metapackage. the moveit_pr2 packages will still not be available though https://github.com/ros-planning/moveit_pr2/issues/71

@v4hn had called  https://github.com/ros-planning/moveit_ros/pull/728 a jade blocker but we've since reverted those changes, so I think it should be skipped for this release

I'm adding jade install instructions to the moveit website at the moment, with an experimental warning

v4hn, 2016-08-06:
On Fri, Aug 05, 2016 at 05:33:02PM -0700, Dave Coleman wrote:
> the moveit_pr2 packages will still not be available though https://github.com/ros-planning/moveit_pr2/issues/71

I talked to @dornhege about the respective pr2 packages and we might have a way to resolve these problems soon.
(It's on my todo list...)

> @v4hn had called  https://github.com/ros-planning/moveit_ros/pull/728 a jade blocker but we've since reverted those changes, so I think it should be skipped for this release

+1 Nevertheless we need fixes for the safety issues soon.

130s, 2016-08-06:
Sorry for being very slow (family issue). 

>  @v4hn had called ros-planning/moveit_ros#728 a jade blocker but we've since reverted those changes, so I think it should be skipped for this release

Ok. 

Just want to get a confirmation. We want to make a new Jade release from https://github.com/ros-planning/moveit, correct?

davetcoleman, 2016-08-06:
Sorry for the family issue :-/

> We want to make a new Jade release from https://github.com/ros-planning/moveit, correct?

Correct. 

fyi the new jade installation instructions: https://github.com/ros-planning/moveit.ros.org/pull/31/files#diff-19947c649c1c205bf153f135fc8cd954R58 (and feel free to review and merge)

dornhege, 2016-08-08:
@v4hn : Do I remember it correctly that the shorthand solution is to somehow get pr2_kinematic released?

v4hn, 2016-08-08:
Yes. https://github.com/PR2/pr2_kinematics/issues/6

v4hn, 2016-08-09:
I talked to @TheDash and he just released the module for jade https://github.com/ros/rosdistro/pull/12303 .
Thus, `moveit_pr2` and `moveit_full_pr2` can be released into jade as far as I can see.

davetcoleman, 2016-08-09:
awesome!

130s, 2016-08-11:
Sorry for the delay. I still feel like asking for confirmation for a few things. @davetcoleman @v4hn

- Do we want to make Jade release only for now? Or IJ at the same time?
- Assuming K is not ready yet for FCL issue. Correct?
- From https://github.com/ros-planning/moveit, we want to release the following packages, under a new `moveit` metapackage?

  ```
 	moveit_commander
	moveit_core 	
	moveit_experimental
	moveit_ikfast 	
	moveit_planners 
	moveit_plugins 	
	moveit_ros 	
	moveit_setup_assistant
  ```
- Can someone create `moveit-release` release repository, maybe something like `https://github.com/ros-gbp/moveit-release.git`? And give a write access of it to @130s?

I'm following `Releasing Debians` section in [davetcoleman/moveit_merge](https://github.com/davetcoleman/moveit_merge/blob/5133432b4fc64bfbf35db1fd0185f528102e0e9e/README.md).

130s, 2016-08-11:
And just for the record, more or less for myself, Jade can be released AFAIK.
- Blocker https://github.com/ros-planning/moveit_ros/pull/728, which was discussed at [the recent maintainer's mtg](http://discourse.ros.org/t/maintainer-meeting-recap-july-26th/366), is [dismissed](https://github.com/ros-planning/moveit_ros/pull/728#issuecomment-237686531).
- [warehouse_ros 0.9.0 was released](https://github.com/ros/rosdistro/pull/11897), which should fix the `warehouse`-related issue so that the [moveit_ros_warehouse regression](https://github.com/ros-planning/moveit/issues/39) should be fixed once `moveit` metapackage is released.

v4hn, 2016-08-11:
+1 from my side.
Please release the moveit repository in indigo and jade.

We planned to fix the safety issues before the release, but didn't get them fixed until now and everything related to them has been reverted.
It makes sense to have a release.

davetcoleman, 2016-08-11:
> Do we want to make Jade release only for now? Or IJ at the same time?

At the same time makes sense to me, unless there is a good reason not to

> Assuming K is not ready yet for FCL issue. Correct?

Correct

> From https://github.com/ros-planning/moveit, we want to release the following packages, under a new moveit metapackage?

- AFAIK moveit_experimental has never been released, and it not even catkinized, so should not be on that list.
- Just to note: moveit_ikfast is new to the metapackage, but I think it should be included since its in the merged repo. It doesn't bring in any dependencies anyway
- A metapackage needs to be defined in a package somewhere. Currently it is [here](https://github.com/ros-planning/moveit_metapackages/tree/master/moveit_full). I think for the new metapackage is should just in the main repo under ``moveit/moveit/package.xml``

> Can someone create moveit-release release repository, maybe something like https://github.com/ros-gbp/moveit-release.git? And give a write access of it to @130s?

Done. I also added @IanTheEngineer and @mikeferguson since they have released moveit in the past

130s, 2016-08-14:
- [x] Add meta package moveit #41
- [ ] Temporarily move `moveit_ikfast` https://github.com/ros-planning/moveit/pull/44

v4hn, 2016-08-15:
I'm working on a fix for https://github.com/ros-planning/moveit/pull/43 .
This is a medium regression and should be part of the release.

v4hn, 2016-08-15:
#43 is resolved

davetcoleman, 2016-08-18:
ping @130s - we need to release moveit-jade so ROS can sync

130s, 2016-08-18:
Sorry I'm on this now

130s, 2016-08-19:
Taking longer due to the [changelog automation command (catkin_generate_changelog) not working](http://answers.ros.org/question/242004/catkin_generate_changelog-issue-with-a-repo-that-consolidated-multiple-repos/). Gave up automation for this time of release.

davetcoleman, 2016-08-19:
That makes sense but sounds like a pain, thanks for the progress

130s, 2016-08-19:
https://github.com/ros-planning/moveit/pull/44#issuecomment-239902177

> I don't see a reason to add CATKIN_IGNORE at moveit_ikfast; aside from release context, I don't see any harm for having a problematic version number (assuming moveit_ikfast stays OUT of moveit metapkg).

I almost had to take this back (but I wouldn't) -- `catkin_prepare_release` is a convenient tool to `tag` and handles incrementation, but it seems that the tool wants all packages in the repository (maybe only those at top-level directory) to be the same version.

Rather than finding a permanent solution I'm manually tagging and incrementing for now.

130s, 2016-08-20:
Now I'm stuck. Looks like version number of packages in a repository need to be the same.

```
$ bloom-release --rosdistro jade --track jade moveit
:
RuntimeError: Two packages have different version numbers (3.2.0 != 0.8.3):
- /tmp/tmpOYe1R2/upstream/moveit_ikfast/package.xml
- /tmp/tmpOYe1R2/upstream/moveit_plugins/moveit_controller_manager_example/package.xml
```

`bloom` tries to grab all packages in the repo, not those listed in metapkg (looks like I've already faced this issue [3 years ago](http://answers.ros.org/question/60138/bloom-runtimeerror-two-packages-have-different-version-numbers/). Sorry I forgot that).

@wjwwood Some of packages' version in this repo are outof sync (we'll like to discuss more about those versions). Is there a way to work this around?

v4hn, 2016-08-20:
Does `CATKIN_IGNORE` work? If yes, go forward with that solution.
Otherwise, feel free to remove the package from the repository for now.
https://github.com/ros-planning/moveit_ikfast still builds fine at the moment and there haven't been any changes yet that are not in there.

https://github.com/ros-planning/moveit/issues/67 will remove the problem hopefully next Tuesday.
(Or after further review...) Afterwards we will have all packages in the repo at the same version.

I believe it would be a good idea to have the release *before* WMD.

130s, 2016-08-20:
I opened https://github.com/ros-planning/moveit/issues/96 to remove `moveit_ikfast` for now as a "safer" workaround (if that doesn't work that would be bad). I'm not entirely sure if `bloom` takes `CATKIN_IGNORE` into account.

130s, 2016-08-21:
With #96 `bloom` seems to be able to make a release. I still [had to remove existing entries from rosdistro](https://github.com/ros/rosdistro/pull/12431) before opening a PR for the new release.

wjwwood, 2016-08-21:
Sorry, I've been on a trip but you can use an ignore file in bloom to skip certain packages, for example: https://github.com/ros-gbp/ros_comm-release/blob/master/jade.ignored

130s, 2016-08-21:
Release request made, finally. https://github.com/ros/rosdistro/pull/12432

@wjwwood thanks, that seems useful. We can still used that technique for releasing other distros.

closed 2016-08-21 by 130s

davetcoleman, 2016-08-22:
:sparkles: Thanks for all the work @130s !

Its so green now:
[image]