moveit/moveit #23: Proposal: Adding a CMakeModules directory

scottpaulin, 2016-08-07:
There doesn't seem to be a common CMakeModules directory for this repo. This would be useful for storing CMake files that are required for multiple packages e.g. the [Coveralls cmake modules](https://github.com/JoakimSoderberg/coveralls-cmake) that will be required for issue #1. One place to put this would be moveit/CMakeModules, that is at a higher level than all of the packages in this repo. 

Having a common CMakeModules could also be useful for reducing code repetition across packages. For example, packages in this repo generally have some code in their CMakeLists.txt that sets the default build type to release. It typically looks like this (taken from [moveit_core](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/CMakeLists.txt)):
```

if(NOT CMAKE_CONFIGURATION_TYPES AND NOT CMAKE_BUILD_TYPE)
  message("${PROJECT_NAME}: You did not request a specific build type: Choosing 'Release' for maximum performance")
  set(CMAKE_BUILD_TYPE Release)
endif()
```

v4hn, 2016-08-07:
I'm not sure it is good practice to have a non-ros-package folder in the repo if it only contains folders with ros packages up to now. Do you know of any other ros-project that does it that way?

coverall actually sounds like it might make sense to add it to [cmake_modules(https://github.com/ros/cmake_modules) instead?

davetcoleman, 2016-08-07:
I was going to start a similar discussion for where to put moveit's docker files. They belong in this repo so that they will be auto-triggered to build when there is a change. So I was thinking of either adding a ``/docker`` folder or perhaps ``/util/docker`` folder. A ``/util`` folder (or similar name) could also be used for common scripts and tools for moveit, such as auto code formatting, travis scripts, etc. Perhaps the ``cmake_modules`` folder could go in there. Not sure what a good standard name for this folder would be.

> if it only contains folders with ros packages up to now.

Up to now was just a few days ago :-) But if we think this is important we could keep all non-code common assets in a new ``moveit`` metapackage folder.

I think adding coveralls to cmake_modules is a good idea (@wjwwood?) but we probably want to test it out in a ROS environment within the moveit project before pushing it to a common ROS package.

@scottpaulin thanks for starting on the coveralls proposal!

v4hn, 2016-08-08:
If we go for adding new top-level folders, maybe it would make sense to have hidden folders? E.g. ".docker" and ".cmake"? Not sure.

mikeferguson, 2016-08-08:
@davetcoleman -- I can see having a docker folder (as it is something we would not release into Ubuntu/Fedora/etc), however, the cmake folder is totally different.

If we depend on some cmake in a top level folder (or anything outside of the ROS packages), we will have to release it as a ROS package as well. I'm not opposed to this, just pointing out that we would need to release it, maybe as "moveit_cmake_modules"?

v4hn, 2016-08-08:
On Mon, Aug 08, 2016 at 09:25:40AM -0700, Michael Ferguson wrote:
> If we depend on some cmake in a top level folder (or anything outside of the ROS packages), we will have to release it as a ROS package as well.

I just noticed that otherwise we would break builds that use the release repositories (this is the default with rosinstall_generator).
Thus full :-1: on a top-level cmake folder that is not a ROS-package.

I'm not sure the overlap between current moveit modules justifies such a separate module though.
So I would propose to check if it works/is a good idea to add the coverall code to `cmake_modules` instead.

davetcoleman, 2016-08-08:
> moveit_cmake_modules

I don't like the idea of having to maintain yet another moveit package. Instead we could for the short term keep it in moveit_core (all moveit packages depend on it) and hopefully eventually move it to cmake_modules.

>  I can see having a docker folder 

Do you prefer ``/docker`` or ``/util/docker``?

v4hn, 2016-08-08:
`/.docker` ?

scottpaulin, 2016-08-11:
Ill try adding it to [ros/cmake_modules](https://github.com/ros/cmake_modules). Thanks for the heads up @v4hn, I didn't know that repo existed.

closed 2016-08-11 by scottpaulin

wjwwood, 2016-08-11:
Does the Coveralls cmake module need to get installed? I agree that cmake modules you need when releasing should not exist outside of the package directories, but if it is only used on travis for doing coverage testing then that's probably fine.

I don't think the Coverall's cmake module makes sense to go into ros/cmake_modules, but if there are other cmake modules that you guys commonly use and that are needed to release the code then that's ok.

You can just do `find_package(coveralls)` and the `if (coveralls_FOUND)`, so that if the coveralls cmake module is not around, then you just don't do coverage. For the travis run you can add the coveralls cmake module to the cmake module path by passing an argument to cmake via the catkin command.

v4hn, 2016-08-12:
> I don't think the Coverall's cmake module makes sense to go into ros/cmake_modules

Why?

davetcoleman, 2016-08-12:
> if it is only used on travis for doing coverage testing then that's probably fine.

this is a good point

wjwwood, 2016-08-15:
> > I don't think the Coverall's cmake module makes sense to go into ros/cmake_modules
> Why?

Because, as far as I understand, it should only be used in CI, specifically on Travis. I try to minimize the things in that repository because it should only be a temporary solution until those modules get put somewhere else (in the dependency's binary package or upstream with cmake). I don't think this module would ever get removed from `cmake_modules`

wjwwood, 2016-08-15:
Sorry, I hit `<tab>` + `<space>` on accident before I was done with my previous post, so I edited it.

v4hn, 2016-08-15:
>  it should only be a temporary solution until those modules get put somewhere else (in the dependency's binary package or upstream with cmake)

I didn't think of the package that way. That policy does make sense though.

I would prefer not to depend on some magical external cmake module for running all tests (including coverage), so +1 for a solution that stores the file in the repo.

wjwwood, 2016-08-15:
It should be possible to make the dependency on coveralls optional, so that it's only needed when running in travis. So it wouldn't be a prerequisite for running the tests. Most packages do not offer an option to run tests with coverage (often because of the impact on performance which can cause some tests to fail), so I don't think it's unreasonable to not offer coverage when running locally without some extra options set.

davetcoleman, 2016-08-16:
+1 to keeping the Coveralls CMake module local. I propose a ``.coveralls`` folder.

v4hn, 2016-08-16:
>  I propose a .coveralls folder.

if this is only relevant for coveralls +1.

davetcoleman, 2016-08-23:
@scottpaulin what is the status on adding coveralls?

scottpaulin, 2016-08-23:
I got up at 9am pacific (4am local) time to do it in the spirit of World Moveit day but saw that @130s had self-assigned it.

scottpaulin, 2016-08-23:
Sorry just saw the last comment in #127 delegating it back to me. I have a job interview now so Ill do it around 6pm Pacific if it still needs attention.

davetcoleman, 2016-08-23:
Yes I think it would be best if you worked on it since you've already started.

Good luck on the job interview, but make sure they use MoveIt! before accepting ;-)