moveit/moveit #24: Improve MoveIt! Quick Start guide

davetcoleman, 2016-08-07:
Would like to expand upon our current Rviz Motion Planning plugin tutorial with the PR2 to make it streamlined for new users:

- Can we consolidate the tutorials so they have a central navigation pane on the left?
- Should new users first learn the [PR2 with Setup Assistant](http://docs.ros.org/indigo/api/moveit_setup_assistant/html/doc/tutorial.html) or do we want them to be able to immediatly plan in Rviz with the PR2?
- There are some broken links
- Currently points at an indigo page

There is now a link at the button of the [install](http://moveit.ros.org/install/) page for the Quick Start

dornhege, 2016-08-08:
Definitely +1 for expanding the Rviz motion plugin tutorial. I was quite confused, when I used that the first time, but it all makes sense!

I'd remove the setup assistant from the basic tutorials. It's not necessary for using MoveIt! and ideally that's already done for the robots of new users. New MoveIt! users with their own robot can still refer to that tutorial, but also learn, how to use MoveIt! with a properly set up robot.

v4hn, 2016-08-08:
I somewhat agree with @dornhege. The assistant shouldn't be the first thing to learn about.
Still, it *is* important and "ideally that's already done" is simply not the usual use-case of MoveIt...

davetcoleman, 2016-08-08:
Another benefit of the SA is that, even if you already have a config package for your robot, it teaches you about some of the components and how they are configured and it gives you a sense of ownership over the setup.

dornhege, 2016-08-08:
I agree that the SA is relevant, but I don't think that this is something that beginners should face first.

davetcoleman, 2019-06-24:
This has been addressed in this code sprint last year:
https://moveit.ros.org/moveit!/ros/2018/02/26/tutorials-documentation-codesprint.html

closed 2019-06-24 by davetcoleman