moveit/moveit #25: Benchmarking: Finish switching moveit to PlannerArena

davetcoleman, 2016-08-07:
A pull request was started to deprecate moveit's old benchmarking capability with a new web-based system used by OMPL: https://github.com/ros-planning/moveit_ros/pull/616

It needs to be rebased, tested at runtime, and have the [documentation](https://github.com/ros-planning/moveit_ros/blob/be704ab36edec0c586a11e3f623c7887b690ffc5/benchmarks/README.md) checked

Also closes https://github.com/ros-planning/moveit_ros/pull/650 and addresses https://groups.google.com/forum/#!searchin/moveit-users/benchmark|sort:date/moveit-users/_JdZdMlAMfw/RvYCP2sABQAJ

@mamoll is the expert on this new benchmarking tool

toliver, 2016-08-22:
We can deal with this task for the WMD.

kirstyellis, 2016-08-23:
working on this issue

kirstyellis, 2016-08-23:
@davetcoleman and @v4hn, I am having trouble applying the commits from this PR (https://github.com/ros-planning/moveit_ros/pull/616) to the new consolidated moveit repository. I rebased to get the commits added to the most recent changes in kinetic-devel, but when I try to apply the commits as patches to the new moveit repo, I get errors related to the files which the commit is deleting as "does not exist in index". I have also tried to rebase the two branches from the separate repos, this also did not work, with the most recent change in the logs being in February.

mamoll, 2016-08-23:
@kirstyellis message me directly for any high-level questions. A little bit of general background. The MoveIt! benchmarking rewrite aims to produce benchmark log files in OMPL's [benchmark format](http://ompl.kavrakilab.org/benchmark.html#benchmark_logfile_format). I don't think the rewrite is using OMPL's Benchmark class directly. The script `moveit_benchmark_statistics.py` should be identical to the script `ompl_benchmark_statistics,py` since this is what produces the SQLite3 database that users can upload to http://plannerarena.org (or upload to their local instance of Planner Arena). The `moveit_benchmark_statistics.py` could potentially be simplified by stripping out the matplotlib-related code, although it might still be useful for some to see how they can create custom plots.

davetcoleman, 2016-08-23:
@kirstyellis when we migrated to the consolidated repo, we changed all the commit hashes in order to [preserve the file history after the location move](https://github.com/davetcoleman/moveit_merge#merge-method).  thus, to moved that PR you basically need to checkout that branch on the old repo, then manually copy it on the filesystem into the new repo.

kirstyellis, 2016-08-23:
hi @davetcoleman, I have managed to get my patches to work by running a git filter-branch on all of the directories as was done in the consolidation of the repos. The patches were applied with no errors so hopefully it will be OK!

kirstyellis, 2016-08-23:
I have to leave the office now but I can work on this again in the morning

davetcoleman, 2016-08-23:
thanks for the status update@

davetcoleman, 2016-10-27:
Benchmarking has been completed, closing issue

closed 2016-10-27 by davetcoleman