moveit/moveit #26: Integrate Fermi GUIs into moveit

davetcoleman, 2016-08-07:
Original topic: https://github.com/ros-industrial-consortium/fermi/issues/26

@rkojcev has done a great job of adding a new method for using MoveIt! via a GUI with interactive markers. Perhaps we want to merge this functionality into moveit

[image]

- Have outside developer check that Fermi is documented enough to be robot agnostic
- Open PR against main moveit repo so we can start review

rkojcev, 2016-08-22:
> Have outside developer check that Fermi is documented enough to be robot agnostic
@davetcoleman any progress on this one?

davetcoleman, 2016-08-22:
@rkojcev this is a suggested project for World MoveIt! Day tomorrow, so I was hoping someone else might want to take this issue up and look into it. If you have the cycles to do it, feel free to go ahead and open said PR with good documentation