moveit/moveit #27: Execution of Cartesian Path Fails Due to Differing Current and Start State

CMobley7, 2016-08-08:
To Whom It May Concern,

I'm using the code below to demonstrate sealing a window using Moveit's python interface. When I test it on both Clearpath Robotics' Husky with UR5 and Fetch Robotics' Fetch, I get either an error or erratic behavior that appears to be due to current pose and the start_pose not matching. However, they are set one after each other.

When the code is run on the UR5, the UR5 driver, Thomas Tim's ur_modern_driver ( https://github.com/ThomasTimm/ur_modern_driver), produces the following error.

` [ERROR] [1470606513.263462897]: Goal start doesn't match current pose`

However, this appears to occur every other trajectory. The Moveit! terminal output is included below.

```
[ INFO] [1470606478.642762190]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1470606478.642964878]: Planning attempt 1 of at most 5
[ INFO] [1470606478.651951898]: No planner specified. Using default.
[ INFO] [1470606478.655162522]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1470606478.672401766]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1470606478.672509679]: Solution found in 0.017943 seconds
[ INFO] [1470606478.684172456]: SimpleSetup: Path simplification took 0.011350 seconds and changed from 4 to 2 states
[ INFO] [1470606502.709724240]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1470606502.710689320]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1470606502.780919741]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1470606502.780984621]: Solution found in 0.070456 seconds
[ INFO] [1470606502.824677454]: SimpleSetup: Path simplification took 0.000493 seconds and changed from 3 to 2 states
[ INFO] [1470606502.829340760]: Received new trajectory execution service request...
[ INFO] [1470606504.651201818]: Execution completed: SUCCEEDED
[ INFO] [1470606505.912313522]: Received request to compute Cartesian path
[ INFO] [1470606505.912874758]: Attempting to follow 2 waypoints for link 'ur5_arm_marker_link' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1470606508.899455833]: Computed Cartesian path with 58 points (followed 100.000000% of requested trajectory)
[ INFO] [1470606508.906403401]: Received new trajectory execution service request...
[ INFO] [1470606511.885918977]: Execution completed: SUCCEEDED
**[ INFO] [1470606511.890627476]: Received request to compute Cartesian path
[ INFO] [1470606511.890953046]: Attempting to follow 2 waypoints for link 'ur5_arm_marker_link' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1470606513.256998251]: Computed Cartesian path with 27 points (followed 100.000000% of requested trajectory)
[ INFO] [1470606513.262543511]: Received new trajectory execution service request...
[ WARN] [1470606513.263840859]: Controller arm_controller failed with error code INVALID_GOAL
[ WARN] [1470606513.263936079]: Controller handle arm_controller reports status FAILED
[ INFO] [1470606513.264018256]: Execution completed: FAILED**
[ INFO] [1470606513.265924877]: Received request to compute Cartesian path
[ INFO] [1470606513.266154401]: Attempting to follow 2 waypoints for link 'ur5_arm_marker_link' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1470606516.585900623]: Computed Cartesian path with 64 points (followed 100.000000% of requested trajectory)
[ INFO] [1470606516.591274222]: Received new trajectory execution service request...
[ INFO] [1470606520.105039153]: Execution completed: SUCCEEDED
**[ INFO] [1470606520.107687355]: Received request to compute Cartesian path
[ INFO] [1470606520.107977450]: Attempting to follow 2 waypoints for link 'ur5_arm_marker_link' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1470606521.445493228]: Computed Cartesian path with 27 points (followed 100.000000% of requested trajectory)
[ INFO] [1470606521.453115526]: Received new trajectory execution service request...
[ WARN] [1470606521.454731340]: Controller arm_controller failed with error code INVALID_GOAL
[ WARN] [1470606521.454904598]: Controller handle arm_controller reports status FAILED
[ INFO] [1470606521.455072531]: Execution completed: FAILED**
[ INFO] [1470606521.457603549]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1470606521.457719245]: Planning attempt 1 of at most 5
[ INFO] [1470606521.459060714]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1470606521.472863524]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1470606521.480973790]: Solution found in 0.022110 seconds
[ INFO] [1470606521.483070776]: SimpleSetup: Path simplification took 0.001920 seconds and changed from 3 to 2 states
[ INFO] [1470606527.690005836]: Received event 'stop'
[ INFO] [1470606527.690275769]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1470606527.690327849]: Planning attempt 1 of at most 5
[ INFO] [1470606527.690454600]: Starting state is just outside bounds (joint 'ur5_arm_wrist_1_joint'). Assuming within bounds.
[ INFO] [1470606527.691189388]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1470606527.715571919]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1470606527.716842772]: Solution found in 0.025781 seconds
[ INFO] [1470606527.716900227]: SimpleSetup: Path simplification took 0.000013 seconds and changed from 3 to 2 states
[ INFO] [1470606527.717317337]: Planning adapters have added states at index positions: [ 0 ]
```

While the Fetch executes every trajectory, at the end of some of the trajectories (sometimes all of them, except the last), the arm will move back to a position along the Cartesian path 

and repeat much of the last trajectory before beginning the next. The Moveit! terminal output is included below. In addition, I have linked to a video due to GitHub not letting me add it as a .zip. (https://drive.google.com/file/d/0B8QbGCFcn-dwWTNjOVVZcmlPRmM/view?usp=sharing)

[... 212 lines cut ...]
        # Finish the ur5_arm in the stow pose stored in the SRDF file
        self.arm.set_named_target("stow")
        self.arm.go()
        rospy.sleep(2)
        
    def cleanup(self):
        rospy.loginfo("Stopping the robot")
        
        # Stop any current arm movement
        self.arm.stop()
        
        # Move the arm to the stow position
        self.arm.set_named_target("stow")
        self.arm.go()
        
        #Shut down MoveIt! cleanly
        rospy.loginfo("Shutting down Moveit!")
        moveit_commander.roscpp_shutdown()
        moveit_commander.os._exit(0)

if __name__ == "__main__":
  try:
    CornerPrediction()
  except KeyboardInterrupt:
      print "Shutting down CornerPrediction node."

v4hn, 2016-08-08:
Hey @CMobley7 ,

it looks like you stumbled over a well-known bug in MoveIt.
It has been around for years and we recently tried to fix it in
https://github.com/ros-planning/moveit_ros/pull/728

However, as you can see, we had to revert the proposed fixes because they broke the PlanningSceneMonitor.
We are currently trying to fix it.

The most simple workaround is to add sleeps between the different motions to wait for updates to the RobotState before setting the new start state to what the monitor believes to be the current state.

CMobley7, 2016-08-08:
@v4hn and @davetcoleman,

Thank you for your help. I really appreciate it. After posting the issue last night, I reread through my code and noticed that I hadn't added rospy.sleep() commands as I have typically seen done and thought that might have been the problem. Thanks for validating this. I was able to get everything working this morning.  Should I close this ticket since this is a known bug in Moveit! or keep it open?

Thanks again. Have a great rest of the week. God bless.

Very Respectfully,
CMobley7

v4hn, 2016-08-08:
closed as "resolved"

closed 2016-08-08 by v4hn