moveit/moveit #28: Apply clang-tidy code linter

davetcoleman, 2016-08-08:
OMPL has been upgrading its code to be faster, follow best practices, and use C++11 by using the automated tool [clang-format](http://clang.llvm.org/extra/clang-tidy/). Would be great to use this linter tool here too. 

It consists of lots of individual checks that will propose or even alter the code. I think each check should be run separate with individual pull requests to the codebase.

davetcoleman, 2017-10-17:
I think this would be a great moveit day project to start. Just found an example config [from a similar project](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/.clang-tidy)

v4hn, 2017-10-17:
:+1: We could definitely use such a cleanup and these patches are very much welcome.

jconley88, 2018-10-25:
Is there additional work to be done for this?

It looks like a [clang PR](https://github.com/ros-planning/moveit/pull/880/) was already merged in.

BryceStevenWilley, 2018-10-25:
I've been working on applying clang-tidy to the code base. A current roadmap is something like this:
* [ ] decide on name formatting style/merge #917 
* [ ] `clang-tidy` uses auto in loop, we have to hand change those to be the full types (PR ready, I plan on opening it after #917 is merged)
* [ ] apply `clang-tidy` to the entirety of MoveIt (it's been restricted to `moveit_core` while we iron out the exact style guidelines).
* [ ] put the `clang-tidy` check in the travis checks somehow.  

I have no idea how to to the final task of putting this in travis, and I think that would be the best aspect for additional work.  Applying to all of MoveIt will be a little cumbersome and will take a while.

davetcoleman, 2018-10-26:
@BryceStevenWilley awesome. Note that all these kinds of changes should now target Melodic, of course

BryceStevenWilley, 2019-01-05:
Some more elaboration on these next steps. Personally can't spare any cycles for this soon, so anyone can feel free to work on this!
1. Apply clang-tidy to the entire codebase. Mostly, that would be going through all of the MoveIt! package's CMakeLists.txt files and adding  `set(CMAKE_EXPORT_COMPILE_COMMANDS ON)` (as seen [here](https://github.com/ros-planning/moveit/pull/693/files#diff-a1e6489257911e9623a4b64c91b728c3R6)). I'm honestly not sure about the performance effects of setting this project wide, most likely it should become a flag/setting used when first setting up MoveIt! (see [the docs](http://moveit.ros.org//documentation/contributing/code/#clang-tidy-linting)).  Once that's figured out, simply build all of MoveIt! and run `run-clang-tidy-3.9.py -fix -p=$HOME/moveit_ws/build/moveit_core/  <package_path>` for each package (as specified in [the docs](http://moveit.ros.org//documentation/contributing/code/#clang-tidy-linting)). Then try to merge the very-large PR.
2. Put clang-tidy in the travis checks. It should look similar to the [check clang format script](https://github.com/ros-planning/moveit_ci/blob/9a2b23cb927196b776e27a12012f514654911ccd/check_clang_format.sh). However, it should mostly likely run after `catkin build` is run in travis, probably at [the end of the file](https://github.com/ros-planning/moveit_ci/blob/master/travis.sh#L201).  `clang-tidy` should run without the `-fix` flag, just like we do with `clang-format`. Should be aware of how long it takes to run: I know the travis time budget is already restrictive for us, I'm not sure if it will finish the checks in time.
3. Add the last two requested checks: `readability-inconsistent-declaration-parameter-name` and `modernize-loop-convert`. Most of the discussion is in [this PR](https://github.com/ros-planning/moveit/pull/880#pullrequestreview-117404830), but in short, we like converting loops, but don't like `auto` (not as readable), and if we're changing parameter names, we might as well make them more readable. These changes will have to be manual: we can turn them on, run clang-tidy, and then go through the codebase and change back all of the `auto`s that it adds to the actual types, etc.  The good news is that once everything is changed, clang-tidy won't complain (i.e. it doesn't replace the actual typenames with `auto` again). I think it's more important to get clang-tidy running in the build before all these manual changes occur though.

BryceStevenWilley, 2019-04-24:
With https://github.com/ros-planning/moveit/pull/1394 merged, I think this issue can be closed. @davetcoleman? 

We could also do clang-tidy on the other upstream ros-planning repos. If we want to, I can make issues in those repos.

davetcoleman, 2019-04-30:
Yes I think you and everyone else has done a great job with this!

closed 2019-04-30 by davetcoleman