moveit/moveit #29: Add continous FCL collision checking for speedup

davetcoleman, 2016-08-08:
MoveIt! currently only collision checks motions/segments/edges by discretizing the segment at some resolution into individual states that are then sent one-by-one to FCL. This is also how MoveIt! plans with OMPL. Switching to FCL's powerful continuous collision checking will likely result in an important speedup (CC is the slowest part of planning).

Currently there are [unimplemented functions](https://github.com/ros-planning/moveit_core/blob/kinetic-devel/collision_detection_fcl/src/collision_robot_fcl.cpp#L178) for this in MoveIt! 

We probably want to use FCL's  [BroadPhaseContinuousCollisionManager](https://github.com/flexible-collision-library/fcl/blob/master/include/fcl/broadphase/broadphase.h#L158)

An example implementation is already [in OMPLApp](https://github.com/ompl/omplapp/blob/d6bfff6da08e65790261185cf7238534f48fab89/src/omplapp/geometry/detail/FCLContinuousMotionValidator.h)

Would be great to find someone to work on this

@Levi-Armstrong, @mamoll, @panjia1983, @jslee02

mamoll, 2016-08-08:
In OMPL.app, discrete collision checking is actually significantly faster than continuous collision checking (see [CollisionCheckers.cpp](https://github.com/ompl/omplapp/blob/master/demos/SE3RigidBodyPlanning/CollisionCheckers.cpp) for a comparison program. Of course, this depends on the collision checking resolution and most likely depends on the type of robot (OMPL.app only does rigid body motion planning). Finally, OMPL.app deals only with triangle soup meshes internally. It could be that *in general* cont. collision detection is actually faster.

davetcoleman, 2016-08-08:
MoveIt! only does rigid body motion planning, but they key difference (I hope) is that we are doing large URDF robot models rather than simpler OMPL mesh models

v4hn, 2016-08-08:
is 50-60 links "large" in terms of ompl?

mamoll, 2016-08-08:
@davetcoleman, by rigid body motion planning I meant planning in SE(2) and SE(3) and not articulated mechanisms.

@v4hn, yes, but not outrageous :-). We have used OMPL to [model the flexibility of large macromolecules](http://www.kavrakilab.org/publications/gipson.moll2013sims-hybrid-method.html) which have thousands of degrees of freedom. Typically, when the number of DOFs gets large, the number of constraints also increases. You probably need a special sampler and/or projection of the c-space to effectively explore a 50-60 dof configuration space.

Levi-Armstrong, 2016-08-10:
@davetcoleman; I started implementing this into MoveIt and I don't think it would take much to finish but I would like to discuss one aspect of the implementation. For the FCL CCD you need to define the motion of the object between two transforms and since it is a robot this is not straight forward because one link motion can be affected by another links motion. FCL provides four types of [motion](https://github.com/flexible-collision-library/fcl/blob/master/include/fcl/ccd/motion.h) (SplineMotion, ScrewMotion, InterpMotion, TranslationMotion). I think there is three option.

Option 1: Assume that the motion between two states is small and use the InterpMotion. This would be the simplest.

Option 2: Assume that the motion can be approximated by a 4 point spline and use the SplineMotion.  The two intermediate points would be determined from joint interpolation between the two states. This would be a little more work than option 1. 

Option 3: The path between two state is broken up into multiple intermediate states based on some criteria and use option 1 or 2. This would be similar to OMPL's longest_valid_segment parameter.

davetcoleman, 2016-08-10:
I think we should use the option that most closely mimics how interpolation is already being done inside OMPL with MoveIt!. For joint-based planning, currently every edge is defined as the [straight line interpolation of each joint between the two states](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/robot_model/src/revolute_joint_model.cpp#L127). It might be more complicated for Cartesian planning, which is used in OMPL when you specify certain types of constraints, but I'm not sure off the top of my head.

At least for joint-based planning, this sounds like Option 1

mamoll, 2016-08-10:
@davetcoleman, I am not sure how FCL is integrated in MoveIt!, but I imagine it performs CCD between every rigid body of the robot (i.e., every link) and the rest of the robot and environment. The SE(3) motion of each link when performing straight line interpolation in joint space will not be a straight line in SE(3) (nor a ScrewMotion).

All of the options still introduce some approximation error. Part of the appeal of CCD is hard guarantees that no collision is missed. With the approximations that guarantee is out the window. The other benefit could be a performance increase over discrete collision checking, but I am actually skeptical that this will be the case.

arocchi, 2019-02-07:
Now that https://github.com/ros-planning/moveit/pull/1156 is merged it looks like this is now closer than ever to happening. @Levi-Armstrong do you still have this on your radar?

davetcoleman, 2019-02-21:
I believe @Levi-Armstrong has been focusing https://github.com/ros-industrial-consortium/tesseract, but the MoveIt! maintainer team would really appreciate help bringing those changes back into moveit

jcvtieck, 2019-11-21:
Hi,
what is the current status of the integration of FCL continuous collision checking?
is it still in development?
what is missing?

davetcoleman, 2019-11-21:
@j-petit @mamoll ?

j-petit, 2019-11-25:
There has no progress been so far on integrating CCD with FCL, however I integrated Bullet with CCD this summer which currently still lives on a `feature-bullet-trajopt` branch but should be merged into master soon.

savolla, 2022-02-28:
Hi all!
I know this is a really old issue but I'd like to know if CCD is done or not. I'm coming from [MoveIt! OMPL Docs](https://ros-planning.github.io/moveit_tutorials/doc/ompl_interface/ompl_interface_tutorial.html?highlight=longest%20valid%20segment) which I think is not regularly updated. Should we wait for CCD in MoveIt! or do you guys suggest other planners for this?

v4hn, 2022-03-01:
There is a GSoC project proposal on Bullet-based CCD and a draft patch with quite some discussion [here](https://github.com/ros-planning/moveit/pull/2897).

Levi-Armstrong, 2022-03-03:
@savolla If you are needing faster collision checking, I recommend checking out Tesseract Motion Planning framework and bench marking data can be found [here](https://tesseract-docs.readthedocs.io/en/latest/_source/core/overview/index.html#tesseract-collision).

[image]

[image]

[image]

AndyZe, 2022-03-03:
@Levi-Armstrong fyi it would help to have axis labels. I guess "More is better" but it's hard to say for sure

Levi-Armstrong, 2022-03-03:
The y-axis is the number of collision checks perform per second.

Levi-Armstrong, 2022-03-03:
This is using the move collision checking benchmark environment which randomly places something like 50 collision objects of different sizes and then randomly generates robot states and checks collision. I set up the bench mark so it randomly generate the collision objects and robot states upfront so each collision checker gets the same environment and checks the same states.

Levi-Armstrong, 2022-03-03:
I try to run this frequently, but these results are from 4 months ago. Though this explains why I saw a jump in the MoveIt data.

simonschmeisser, 2022-03-03:
So the code can be found here https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_collision/fcl/src/fcl_discrete_managers.cpp if anybody else is wondering ... maybe we can take some inspiration :wink:

simonschmeisser, 2022-03-03:
@Levi-Armstrong is this testing the same scene with the same robot state repeatedly?

simonschmeisser, 2022-03-04:
@Levi-Armstrong could you add some more details on how to run your benchmark? I have some ideas I would like to test but convenient benchmarking and fancy graphs would help of course

Levi-Armstrong, 2022-03-04:
Unfortunately, I do not have access to the code used to run the benchmark anymore.

Levi-Armstrong, 2022-03-04:
It was based on the one @j-petit put together.