moveit/moveit #34: `catkin_lint` packages

v4hn, 2016-08-10:
In an initial attempt (e.g. https://github.com/ros-planning/moveit_core/pull/305) I ran [`catkin_lint`](https://github.com/fkie/catkin_lint) on `moveit_core`, `geometric_shapes` and some other repos and cleaned up many issues spotted by the linter.

Many of MoveIt's ROS-packages still produce a number of errors.
This is your chance to change that and prevent future problems with broken dependencies, broken CMakeLists constructs, issues to port packages from OSRF's /opt/ folder to /usr, etc. :-)

A few issues reported by catkin_lint are false-positives or explicit decisions by the maintainers.
The most famous one is:
> moveit_core: CMakeLists.txt(6): error: variable CMAKE_BUILD_TYPE is modified

It was decided that MoveIt should default to "Release" build type for maximum performance, so this is no error.
Bonus points if you find a way to make catkin_lint stop complaining about it.

As this touches a number of different packages, please do not assign to this ticket.
Instead, find yourself a moveit package `catkin_lint` complains about, add a note saying you are looking into this package and submit one pull-request for each package you clean.

davetcoleman, 2016-08-22:
linting the entire code base would allow us to then create a continuous integration test to capture bugs like: https://github.com/ros-planning/moveit_ci/issues/11

corot, 2017-01-16:
:+1:  I'm also interested on making catkin_lint stop complaining about overwritten CMAKE_BUILD_TYPE

mintar, 2017-01-16:
@corot : `catkin_lint --ignore immutable_var` suppresses the warning about `CMAKE_BUILD_TYPE` (but perhaps also others?).

If you call `catkin_lint --explain`, it will give a short explanation of each error / warning and tell you how to ignore it.

corot, 2017-01-16:
Very useful; viel danke!  :clap:

roehling, 2017-04-18:
> Bonus points if you find a way to make catkin_lint stop complaining about it.
I changed the `catkin_lint` logic (as of version 1.4.14) so it won't complain any more if the `set()` command occurs inside an `if()` that checks for `NOT CMAKE_BUILD_TYPE`, so I'm helping myself to a few bonus points: .....

v4hn, 2017-04-18:
Nice, one more step towards a catkin_lint-compliant MoveIt!
@roehling Give me a call when you want to cash your points ;)

davetcoleman, 2019-06-24:
It appears some of these have already been done. Given our effort to port to moveit2, I will close this and we can focus on cleaning up ament instead.

closed 2019-06-24 by davetcoleman