moveit/moveit #35: Integrating STOMP

jrgnicho, 2016-08-10:
Given the recent [interest(https://github.com/davetcoleman/moveit/pull/2) in STOMP and [CHOMP](https://github.com/ros-planning/moveit/issues/15#issuecomment-238108132) planners  I'll be working on adding [stomp](https://github.com/ros-industrial/industrial_moveit) from industrial_moveit as another MoveIt planner.  The current plan is to copy all relevant packages from **industrial_moveit** into the **moveit_planners** directory however there are some challenges related to that
- The (industrial_collision_detection)[https://github.com/ros-industrial/industrial_moveit/tree/indigo-devel/industrial_collision_detection] package is a dependency so should that go in **moveit_planners** too.
- The stomp_test_support and stomp_test_kr210_moveit_config packages contain files and resources (meshes, urdfs, launch, yamls) that we use for our testing the *stomp planner*, should those be included too?
- How to preserve as much of the commit history as possible? Should we?

davetcoleman, 2016-08-10:
> The industrial_collision_detection package is a dependency so should that go in moveit_planners too.

Would be ideal if the collision detection could be adapted for moveit's collision plugin interface - [example of how fcl extends it](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_robot_fcl.h#L45). @mikeferguson can advice on how to do this

> The stomp_test_support and stomp_test_kr210_moveit_config packages contain files and resources (meshes, urdfs, launch, yamls) that we use for our testing the stomp planner, should those be included too?

Testing resources should live in [moveit_resources](https://github.com/ros-planning/moveit_resources) so as to not bloat the file size of the main repo. Currently the PR2 tests live there but [others are on their way](https://github.com/ros-planning/moveit/issues/17)

> How to preserve as much of the commit history as possible? Should we?

I personally don't think its that important as long as the original repo stays around for the foreseeable future for reference. However if the consensus is we should migrate it, @wjwwood and I worked on [two different methods](https://github.com/davetcoleman/moveit_merge) for merging repos that can be adapted for your purposes.

jrgnicho, 2016-08-10:
@davetcoleman 

- The [CollisionRobotIndustrial](https://github.com/ros-industrial/industrial_moveit/blob/indigo-devel/industrial_collision_detection/include/industrial_collision_detection/collision_robot_industrial.h) already implements the *CollisionRobot* interface.  What it incorporates in addition to CollisionRobotFCL is some speed improvements and some distance calculation functionality.

- I can put all the resource related packages in *moveit_resources*

- I'll look into those merging methods but I think that omitting  the commit history will help keep things cleaner.

v4hn, 2017-10-11:
We decided that stomp will stay with ROS-I for the time being.
If they ever wish not to maintain it any longer, we can include it in moveit directly.

closed 2017-10-11 by v4hn