moveit/moveit #36: Consolidate moveit tutorials and other sphynx documentation

davetcoleman, 2016-08-10:
Currently our non-moveit website documentation lacks a common navigation side bar or central location for finding it. This makes understanding how documentation works, where stuff belongs, etc difficult.

I've consolidated all of the *.rst sprinkled across the moveit repo and moveit_pr2. Note I have not spent the time fixing inline code or images yet because this is just a proposal (I did have to change some of the navigation though) and there are a few other small bugs:

- [Temporary host location](http://dav.ee/dave/moveit/html/index.html)
- [Temporary github repo](https://github.com/davetcoleman/moveit_docs)

I've put it in the already existing repo moveit_docs. I've also switched the page theme to mimick readthedocs.org, similar to how @wjwwood setup [catkin_tools](https://catkin-tools.readthedocs.io/en/latest/) and how @mikeferguson setup the [Fetch documentation](docs.fetchrobotics.com). This issue is also explained here: https://github.com/ros-planning/moveit/issues/11

Advantages:
- Easier to navigate between pages for users
- Easier to add to and update for contributors
- Centralized theme that looks good
- Edit on Github button
- Can provide more streamlined experience than having docs in separate pkgs

Disadvantages
- Docs are not inline with each package

I'm hoping everyone agrees with the advantages here. The one question I'm trying to decide on is where the new ``moveit_docs`` folder should go. I think within our new ``moveit`` repo is best, but keeping it in a separate repo has advantages:

Advantages of adding it to ``moveit`` repo:
- Will basically still be inline with code, so API changes can be propagated easier with grep commands, etc
- Can request all new features to include in the same PR an update to the docs
- Can easily track changes in API across different ROS versions with our pre-existing ``-devel`` branches
- This is how the [ompl project does it](https://github.com/ompl/ompl/tree/master/doc) and many others

Disadvantages of adding it to the ``moveit`` repo:
- We might not want to include images in the main repo for file size reasons. If this is the case, we could keep them in moveit_resources as assets but takes a little more work. We don't have that many GUIs/images though, so this is likely minimum effort
- This is how [fetch does it](https://github.com/fetchrobotics/docs)

This consolidation would greatly ease user's abilities to improve documentation on World MoveIt! Day, so that is my target date to finish

v4hn, 2016-08-11:
On Wed, Aug 10, 2016 at 04:14:08PM -0700, Dave Coleman wrote:
> Currently our non-moveit website documentation lacks a common navigation side bar or central location for finding it. This makes understanding how documentation works, where stuff belongs, etc difficult.
> 
> I've consolidated all of the *.rst sprinkled across the moveit repo and moveit_pr2. Note I have not spent the time fixing inline code or images yet because this is just a proposal (I did have to change some of the navigation though) and there are a few other small bugs:
> 
> - [Temporary host location](http://dav.ee/dave/moveit/html/index.html)
> - [Temporary github repo](https://github.com/davetcoleman/moveit_docs)
> 
> I've also switched the page theme to mimick readthedocs.org

Personally, I very much prefer the current layout as it looks more simple.
But if you find someone who agrees with you here, I don't mind.

> Disadvantages
> - Docs are not inline with each package

This would be a severe regression! Some pages basically become empty!
Compare http://docs.ros.org/indigo/api/pr2_moveit_tutorials/html/planning/src/doc/move_group_interface_tutorial.html to http://dav.ee/dave/moveit/html/move_group_interface_tutorial.html#compiling-the-code .
:-1: until the code examples are displayed *on the tutorial page*. This is where they belong. Sorry.

> This consolidation would greatly ease user's abilities to improve documentation [...]

I somewhat agree, but see above.

davetcoleman, 2016-08-11:
> until the code examples are displayed *on the tutorial page*. This is where they belong. Sorry.

I am certainly going to fix that, that was what I meant by "Note I have not spent the time fixing inline code or images yet because this is just a proposal"

v4hn, 2016-08-11:
> ApplyPlanningScene is a MoveGroupCapability

Ah, ok.

How do you plan to handle the PR2 dependency of the tutorials?
I don't think it is an option to have (runnable) tutorials in the moveit repo if that makes it depend on the pr2-packages.

I'm not entirely sure I understand your approach here..

davetcoleman, 2016-08-11:
> ApplyPlanningScene is a MoveGroupCapability

Did you mess up copy/paste? This seems like a random quote.

> How do you plan to handle the PR2 dependency of the tutorials?

No dependency will be added. I was planning on having a copy of the necessary include code files in the documentation folder. I would not add any package.xml dependency but, as the tutorials already say, you would just run ``apt-get install ros-indigo-moveit-pr2`` and you would have all your deps. 

This copy would be exactly the same as the master version that remains in the moveit_pr2 repo and the tutorial's links would point to there. While semantically not ideal I think this is ok because:

- The moveit_pr2 tutorials rarely change and haven't been updated in 3 years
- Its very common to have a separate copy of code in documentation from the source
- Its very easy to sync any changes... just copy the file exactly.

Another thing to keep in mind is that the pr2 tutorials are documenting the many classes and functionality in the ``moveit`` repo - so if we moved the docs there we'd actually be keeping the documentation closer to the source code we care about. The focus of the MoveIt! tutorials is not to learn how to use the PR2 - its to learn how to use MoveIt!. For example - with your proposed rename of move_group to move_group_interface, that change could be applied to the tutorials in the same PR.

v4hn, 2016-08-11:
> Did you mess up copy/paste? This seems like a random quote.

Yes, I fixed it above, sorry. :)

> The moveit_pr2 tutorials rarely change and haven't been updated in 3 years

Wrong, I changed something in there a short time ago to explain ApplyPlanningScene. :)

I do see the benefits of actually having the documentation around in the repo.
This doesn't solve the two-pull-requests for one change issue though, because the `moveit_pr2` repo will still have to be updated (in order to check the correctness of any change) presumably *before* the tutorial change goes upstream, but *after* the actual API change/extension becomes available. This sounds to me like it would be three pull-requests (instead of two) or, alternatively, more work for the maintainers to review, if they should review the tutorial changes together with the actual change.

At the moment I think it would be better to have the docs in a separate repository.
If you wish to merge all of the docs into one single repository, go ahead, that sounds like a good idea.
But instead of copying sources, I would propose to merge `pr2_moveit_tutorials` and any other tutorial packages we have docs in in there with their full list of dependencies.

mikeferguson, 2016-08-12:
I'm a big fan of docs in a separate repository -- while I haven't actually checked it out yet, I presume the post-merge moveit repo has a heavy history, and documentation *should* include lots of visuals, which makes the repository size bloat quickly. Also, we will inevitably still end up with moveit-related packages that are not part of the mainline -- ROS-industrial has been creating new things, some of which will remain independent, vendors who support robots on MoveIt will keep their robot-specific configurations external to the main repo, etc. I would still advocate that we make it easy to document these things within the main wiki (the ROS wiki is a great example of this, huge amount of docs are part of this single wiki which spans hundreds or thousands of repositories, but everyone knows where the jumping off point is).

One thing to also keep in mind is that the tutorials/detailed-user-documentation in this repo is not a replacement for developer-level stuff that is documented in doxygen. If properly indexed, the ROS website will automatically build and host those doxygen-generated docs, and so we should be able to easily link to them from our tutorials/detailed-user-documentation. With this sort of split, the lower-level code-API docs tend to stay with the code, wherever it might be, and the higher-level explanations and user documentation can be stored in a single location.

davetcoleman, 2016-08-16:
I think the discussion in this thread is getting murky because we are not being precise in separate concerns of **documentation** and **tutorials**. Currently MoveIt! has almost no **documentation** - just a few short .rst files sprinkled in different packages throughout the repos, and a single [concepts](http://moveit.ros.org/documentation/concepts/) page that lacks on outline or navigation pane. We do have fairly good tutorials thanks to SRI/@sachinchitta, @hersh, and acorn but they too are split up between pr2_moveit_tutorials, moveit_setup_assistant, and other packages. All these tutorials are tied together but a page on the [moveit website](http://moveit.ros.org/documentation/tutorials/) but I think we can do better, including a navigation pane.

>  documentation should include lots of visuals, which makes the repository size bloat quickly

I had proposed we keep visuals in moveit_resources, separate from the main repo, but that does get complicated in keeping things in sync and having to do two PRs.

> This doesn't solve the two-pull-requests for one change issue though, because the moveit_pr2 repo will still have to be updated (in order to check the correctness of any change) presumably before the tutorial change goes upstream, but after the actual API change/extension becomes available.

You are assuming we have to keep a working version of the code examples separate from the documentation files. While there are benefits to doing this, I think most documentation I've seen does not go to this length of completeness, instead just keeping the code examples in the documentation format only. 

So it seems we are balancing two objectives: keeping tutorials and documentation in one unified place for easy improvement and maintenance vs. keeping them in the semantically correct place. Note that the maintenance is considerable - you have to have a rosdoc.yaml file, a conf.py, a theme folder, and the proper bloom configuration. Currently much of our tutorials still points to hydro links. 

**My conclusion**

- Consolidate all tutorials to ``pr2_moveit_tutorials`` including ones that are currently not there like the setup assistant. Apply the new theme and unified navigation bar proposed in this thread
- Consolidate all other documentation onto the moveit website, switching to markdown instead of rst. Note there is very little in this category.

v4hn, 2016-08-16:
> Consolidate all tutorials to `pr2_moveit_tutorials` including ones that are currently not there like the setup assistant. Apply the new theme and unified navigation bar proposed in this thread

I know this time my opinion is the wrong way around, but I would prefer a new packages `moveit_tutorials` instead. Nevertheless this could use `pr2_moveit_tutorials` as a basis. The rest sounds good to me.

> Consolidate all other documentation onto the moveit website, switching to markdown instead of rst. Note there is very little in this category.

I'm not sure which rst files you are talking about exactly.

davetcoleman, 2016-08-16:
In ``moveit`` with ``find . ! -name CHANGELOG.rst -name *.rst``:

```
./moveit_core/doc/custom_constraint_samplers.rst
./moveit_ikfast/doc/ikfast_tutorial.rst
./moveit_ikfast/index.rst
./moveit_plugins/moveit_fake_controller_manager/doc/tutorial.rst
./moveit_planners/ompl/doc/tutorial.rst
./moveit_setup_assistant/doc/tutorial.rst
./moveit_ros/visualization/doc/tutorial.rst
./moveit_ros/visualization/doc/joystick.rst
```

We remove all of these and have no sphynx documentation in the code base, centralizing it all in either the tutorials package or the website (which is git-based so we can still modify it in our editors)

>  I would prefer a new packages moveit_tutorials instead

I like that. We would move and rename ``pr2_moveit_tutorials`` and have the new package depend on ``moveit_pr2``.

v4hn, 2016-08-16:
On Tue, Aug 16, 2016 at 12:09:57PM -0700, Dave Coleman wrote:
> We remove all of these and have no sphynx documentation in the code base,
> centralizing it all in either the tutorials package or the website

As I have never seen some of these files +1

> I like that. We would move and rename ``pr2_moveit_tutorials`` and have the new package depend on ``moveit_pr2``. 

+1 Go ahead if you like.

davetcoleman, 2016-08-18:
I attempted to make the new tutorials package on my Ubuntu 16.04 / Kinetic laptop but I cannot build moveit_pr2 because it is not released. Rethinking things - I do not think we should have MoveIt!'s tutorials depend on a robot's extensive software working or not. I contacted Clearpath recently and they confirmed:

> It is correct that we're ending support this year for PR2; we explored a number of ways of funding support for the platform and could not find a solution.  There is no planned support for PR2 in Kinectic or onward, although PR2 is supported in Jade along with the Indigo release.  

I know you said PR2 support will continue in the open source community, but I imagine it will be a long while before PR2 is released into Kinetic. Following this argument, I still think we should eventually switch demo robots.

I think my original solution of having a copy of the code in moveit_docs would be best for the time being, so that it works on current ROS versions regardless of the PR2's release state.

We have many examples of MoveIt tutorials for robots that are currently being sold and that do not require any dependencies to build the docs:
UR5: https://www.clearpathrobotics.com/assets/guides/husky/HuskyManip.html
Fetch: http://docs.fetchrobotics.com/manipulation.html
Baxter: http://sdk.rethinkrobotics.com/wiki/MoveIt_Tutorial
ShadowHand: https://github.com/guihomework/dexterous-manipulation-tutorial/wiki/tuto-using-part1

v4hn, 2016-08-18:
As @dornhege and I will probably gain access to the repositories/packages in https://github.com/PR2 in a short while according to my latest mail conversation with clearpath, we can hopefully release the  ROS packages required by `moveit_pr2` rather soon.

Moving the whole PR2 to kinetic will take more time, but that is a different story.

If you want to wait for the *four* pr2 packages required by the moveit tutorials,
that takes away the burden to rearrange the tutorials.
I see your point though.

davetcoleman, 2016-08-18:
I'm worried it will be difficult for people to find and edit documentation next Tuesday - my [proposed restructure](http://dav.ee/dave/moveit/html/index.html) makes it super easy to see and edit the flow that people learn MoveIt! with (even has next buttons on the bottom of the page).

v4hn, 2016-08-19:
Does the package build when you add `rosinstall_generator --rosdistro jade pr2_description pr2_controllers_msgs pr2_arm_kinematics pr2_mechanism_msgs` to the workspace?

Is it a problem to have a documentation package that does not build due to dependencies at the moment?
You planned to have code that does not have to build at all in the docs, so from my perspective having a package that does not build out-of-the-box is still an improvement. :)

IanTheEngineer, 2016-08-19:
Sorry to jump in here late. Big +1 for the proposed documentation restructuring. I've always admired the ReadTheDocs themes for being clean and approachable, and I agree that Fetch and catkin_tools make great use of it. Looking at the old documentation, it feel static and scattered in its organization.

I care less about where the documentation is stored, whether separate repo or part of the larger MoveIt repo. Though all other things equal, the easier to maintain, the better.

If we do go down this route, Dave, what would it take to put these docs in place for Tuesday's WMD?

davetcoleman, 2016-08-22:
> Does the package build when you add `rosinstall_generator --rosdistro jade pr2_description pr2_controllers_msgs pr2_arm_kinematics pr2_mechanism_msgs` to the workspace?

I have not tried this and assume there would be things that do not work in kinetic yet

> Is it a problem to have a documentation package that does not build due to dependencies at the moment?

It somewhat is...  I just realized ``rosdoc_lite .`` still works without any of the pr2 dependencies working but I'm not sure if the ROS buildfarm can generate the documentation if the dependencies don't work. But since it works in indigo still, we can just keep generating it there. You bring up a good point I hadn't thought of.

@IanTheEngineer I could clean it up tomorrow morning and I think it would be ready

v4hn, 2016-08-22:
On Sun, Aug 21, 2016 at 08:55:27PM -0700, Dave Coleman wrote:
> It somewhat is...  I just realized ``rosdoc_lite .`` still works without any of the pr2 dependencies working but I'm not sure if the ROS buildfarm can generate the documentation if the dependencies [...]

+1 to do it that way and provide a central tutorials package tomorrow!

davetcoleman, 2016-08-23:
The new tutorials repo is ready: https://github.com/ros-planning/moveit_tutorials

Note: I'm on an airplane right now so haven't been able to upload the demo html website

It is similar and replaces [pr2_moveit_tutorials](https://github.com/ros-planning/moveit_pr2/tree/indigo-devel/pr2_moveit_tutorials) and it has the same linking of source code. I haven't tried building it because I'm on Kinetic, but should be trivial if it doesn't already build for indigo.

I've added a new theme, improved navigation, and moved all *.rst files from across moveit into it so that tutorials are consolidated.

To get this ready for tomorrow's World MoveIt! Day I need to do the following:

- [ ] Release to bloom so that the ROS buildfarm can build the html for us
- [ ] Ask Tully to add a rule that makes it rebuild more often than every 24 hours
- [ ] Update the links on moveit.ros.org
- [ ] Remove the duplicate documentation in the moveit repo
- [ ] Remove the duplicate docs in moveit_pr2

mikeferguson, 2016-08-23:
LGTM +1

davetcoleman, 2016-08-24:
Closing this issue because it is complete, though https://github.com/ros-planning/moveit_pr2/pull/80 should eventually be merged

closed 2016-08-24 by davetcoleman