moveit/moveit #37: rename planning_interface::MoveGroup to planning_interface::MoveGroupInterface

v4hn, 2016-08-11:
An often observed source of confusion in MoveIt is the class ["MoveGroup"](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group.h#L69).

While everyone instantly believes to recognize this as the node that encapsulates most of MoveIt's functionality in ROS, this class actually lives in "planning_interface" and only provides *a client interface* to interact with the node via ROS-interfaces. It is often believed that most of the class methods call library functions and do not wrap ROS interfaces.

As such it should be renamed "MoveGroupInterface" in kinetic. This includes a number of steps (in order)

- [ ] move [`move_group.h`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group.h) to `move_group_interface.h` and rename the class
- [ ] provide a new `move_group.h` that provides a typedef named `MoveGroup` to be downward compatible
- [ ] rename the class throughout MoveIt's own code base
- [ ] adjust the tutorials to the new name
- [ ] add a deprecation notice to `move_group.h`

davetcoleman, 2016-08-11:
I want to clarify what you are saying because it was hard to follow. The problem is that there are two classes with the name ``move_group.cpp`` and one is confusing.

The node ``moveit_ros/move_group/move_group.cpp`` **is** the node that encapsulates most of MoveIt's functionality in ROS. When it is launched it creates a bunch of ROS-msg/service/action interfaces that allow you to do planning, etc. It is the central library in MoveIt!.

``planning_interface/move_group_interface/src/move_group.cpp`` is the **interface** for interacting with MoveIt! via ROS comm (as opposed to directly as a library). It would be better if it was renamed to match the folder it lives in - ``move_group_interface.cpp``

If I understand @v4hn correctly, then +1

v4hn, 2016-08-11:
> I want to clarify what you are saying because it was hard to follow.

Thanks, I'm a bit under the weather today. I see that my explanation was hard to follow...

You got it right. Two minor corrections:

- There is only *one* class `MoveGroup`. It's just that the loose term "move group" is generally associated with the node and therefore the class should actually be called `MoveGroupInterface`.
(`moveit_ros/move_group/move_group.cpp` actually uses the name `MoveGroupExe` internally)

- `move_group` (the node) is *not* a library. It's a ROS node. It's as simple as that. :-)

davetcoleman, 2016-08-11:
Good points. I've updated my previous comment to be a bit more semantically correct.

v4hn, 2016-09-26:
#251 implements the proposal. Discussion should be continued there.

davetcoleman, 2016-09-26:
I like your checklist, above. It seems this issue should not actually be closed yet

v4hn, 2016-09-26:
you're right, I missed the tutorial updates.
Everything else is addressed in the pull-request

davetcoleman, 2016-10-26:
tutorials updated: https://github.com/ros-planning/moveit_tutorials/pull/38

closed 2016-10-26 by davetcoleman