moveit/moveit #38: make PlanningSceneInterface add/remove CollisionObjects synchronously

v4hn, 2016-08-11:
I would very much like to adjust [`PlanningSceneInterface::{add/remove}CollisionObjects`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning_interface/planning_scene_interface/include/moveit/planning_scene_interface/planning_scene_interface.h#L86)
to use the new `ApplyPlanningScene` service such that it performs synchronous updates to the planning scene.
This is what most users expect and it is not even documented in the [tutorials](http://docs.ros.org/indigo/api/pr2_moveit_tutorials/html/planning/src/doc/move_group_interface_tutorial.html#adding-removing-objects-and-attaching-detaching-objects) that this is not the case.

At the moment these methods wrap simple `publish` calls whereas the other methods in this class query the `move_group` via services.

Thus, the current behavior is (1) inconsistent, (2) not documented, and (3) makes people add arbitrary long `sleep` calls after the function call.

**However**, ApplyPlanningScene is a MoveGroupCapability [that has to be added manually in each projects `move_group.launch`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/move_group.launch#L59) and the PlanningSceneInterface has been around quite long.
Thus, simply changing the implementation would break many use-cases.
As a downward-compatible workaround I propose to check whether the service exists and either (1) call the detected service or (2) print an appropriate warning telling the user to enable ApplyPlanningScene in their move_group and afterward continue with the current publish.
With ROS-m,n, or o we could eventually remove the publish then.

andreaskoepf, 2016-08-20:
+1, we at Xamla had trouble with the asynchronous (unreliable) nature of this too. @v4hn, @rhaschke  Does someone of you want to work on this during WMD? Otherwise I could review the code and the proposed solution over the weekend  to fix this on Tuesday.

v4hn, 2016-08-20:
On Sat, Aug 20, 2016 at 03:36:22AM -0700, Andreas Köpf wrote:
> +1, we at Xamla had trouble with the asynchronous (unreliable) nature of this too.
> @v4hn, @rhaschke  Does someone of you want to work on this during WMD?
> Otherwise I could review the code and the proposed solution over the weekend to fix this on Tuesday.

I will be with you on WMD, so we can discuss things there and I can try to advise you.

As for all `simple enhancement` tickets, just add a note saying you will work on it.

v4hn, 2016-08-22:
@andreaskoepf do you claim this issue for tomorrow?

andreaskoepf, 2016-08-23:
@v4hn Yes, let's discuss this later. Would be happy to work on this.

v4hn, 2017-01-31:
This has been merged upstream in the meanwhile.

closed 2017-01-31 by v4hn