moveit/moveit #39: Regressions on Jade

wjwwood, 2016-08-11:
I'd like to sync Jade, but moveit has regressions. It looks like Trusty has a regression for ``:

wjwwood, 2016-08-11:
Sorry, accidentally hit `<tab>` then `<space>`...

---

I'd like to sync Jade, but moveit has regressions. It looks like Trusty has a regression for moveit_ros_warehouse:

http://build.ros.org/view/Jbin_uT64/job/Jbin_uT64__moveit_ros_warehouse__ubuntu_trusty_amd64__binary/72/console

If someone could look into that, I'd appreciate it. I think all of the moveit related regressions stem from this failure, but there may be others.

v4hn, 2016-08-11:
The problematic code has more or less recently been replaced by a plugin mechanism after long discussion:
https://github.com/ros-planning/moveit_ros/pull/699
A release is more or less on its way: https://github.com/ros-planning/moveit/issues/22

@130s, Sorry for the ping.. What's the state of this? We should probably aim to increase our bus-factor for releases.

130s, 2016-08-11:
Yeah, as I [left a comment](https://github.com/ros-planning/moveit/issues/22#issuecomment-239268837), once we make a new release of `moveit` (that contains the current `moveit_ros`), this particular regression should be fixed.

davetcoleman, 2016-08-11:
I think we should move fast on releasing I/J so we don't hinder the sync

davetcoleman, 2016-08-11:
Side note: there's a pretty table of our Jade regressions in our README now: https://github.com/ros-planning/moveit

v4hn, 2016-08-22:
Fixed. See https://github.com/ros-planning/moveit/issues/22

closed 2016-08-22 by v4hn

130s, 2016-08-23:
Jade DEBs are built and available from [shadow](http://wiki.ros.org/ShadowRepository) repository. 

Now Jade regression is almost gone http://repositories.ros.org/status_page/ros_jade_default.html?q=REGRESSION.

davetcoleman, 2016-08-23:
what is not gone yet? ompl?

130s, 2016-08-23:
It is `ompl` for the entire Jade world for today. 
[I poked them](https://bitbucket.org/ompl/ompl/issues/266/ros-jade-new-version-is-older-than-the-one). Doesn't seem to be critical.

@wjwwood Sorry it took forever to get MoveIt! out for Jade. I assume we want some soak time (a week? 2 weeks? Anyone?) because the new release introduces the tremendous amount of changes.

wjwwood, 2016-08-23:
It would be prudent to have some time for people to test the new moveit in Jade. We can plan to do a sync in a week or so.

The ompl thing isn't a regression. It is them fixing their version numbers. They may need to make sure that future releases can be considered newer than what's already there. Since nothing has changed between the two numbers, that I'm aware of, it should be fine for now.

davetcoleman, 2016-08-23:
we've never officially released jade because it only worked on one platform, so i don't think we need much soak time

wjwwood, 2016-08-23:
That's also ok for me, since we've waited quite a while for Jade to sync already.