moveit/moveit #48: C++11ifying the public API for the kinetic release

de-vri-es, 2016-08-16:
So, there has been some discussion about C++11ifying the moveit API. It would make sense to do that for the Kinetic release, or we're stuck with the old API for another 4 years until the next long-term support platform.

As far as I can see, C++11ifying the API basically comes down mainly to replacing `boost` types with the equivalent `std` types from C++11. From the top of my head the most likely changes here would be:

* Switch from boost to C++11 smart pointer.
* Switch from `boost::function` to `std::function`
* Switch from `boost::chrono` types to `std::chrono` types.
* Switch from boost synchronisation types to C++11 equivalents (`std::mutex` and friends)
* Possibly use `std::tuple` instead of boost equivalents (probably not needed)
* Use `enum class` instead of old enums.

There are some more changes possible which are unlikely to affect the public API:
* Switch from boost to `std::bind`
* Switch from boost to `std::thread`
* Use the new random number generator facilities
* Simplify things internally using C++11 features (lambdas, range based for loops, etc...)

Some shared_ptr types will be forced by the ROS API, but those are generally used through typedefs like `sensor_msgs::JointStatePtr`. Those would be left alone of course.

@davetcoleman  also suggested seeing what clang-tidy comes up with, which seems like a good idea.

So, I would like to undertake a project like this on the short term (on time for Kinetic which shouldn't be delayed too long I think). But before doing anything like that of course I'd like to know what other people think about C++11ifying the API.

de-vri-es, 2016-08-16:
Note that this is a more general continuation of ros-planning/moveit_core#319

v4hn, 2016-08-16:
I'm unsure about whether we should move the API to `std`'s types.
This implies a lot of breakage that people would have to fix by converting their projects (especially for smart-pointers).
Maybe it makes more sense to try to encapsulate all usage of pointers into typedefs in kinetic and switch in lunar?
Are std::function and boost::function compatible? As long as one can pass in a boost type and we work with std types, I'm fine with it...

I'm all in for simplifying code via C++11 features on the other hand.
So if you like to refactor some part of the code via C++11, feel free to go forward and file requests.

davetcoleman, 2016-08-16:
+1 to switching our public headers to ``std`` types. According to the author of MoveIt! @isucan: "i would not code without c++11 now". 

I think trying to do a phased conversion will only continue to give us headaches, as demonstrated by ROS Kinetic's "minimum requirements: C++11" but reliance on boost for public headers in ros-desktop.

We already have a nice encapsulation method for shared pointers used extensively (but not exclusively) in moveit. Its the macro in [class_forward.h](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/macros/include/moveit/macros/class_forward.h) and for most of our shared_ptrs all we have to do is change two lines in that file and our API is changed. The following classes currently use it:

RobotModel
RobotState
DynamicsSolver
Transforms
PlanningContext
PlannerManager
PlanningScene
PlanningSceneMonitor
PlanningPipeline
PlanExecution
PlanWithSensing
TrajectoryExecutionManager
MoveGroup
RenderShapes
TrajectoryVisualization
ActionBasedControllerHandleBase
ConstraintsLibrary
ConstraintSamplerManager
ConstraintSampler
KinematicsMetrics

I think we should switch more of our shared_ptrs to using that macro so that cherry-picking between branches remains super easy.

de-vri-es, 2016-08-16:
> Are std::function and boost::function compatible? As long as one can pass in a boost type and we work with std types, I'm fine with it...

`boost::function` and `std::function` are rarely constructed explicitly. Normally, you just pass a functor which is  converted implicitly. Considering that both `boost::function` and `std::function` are functors, they should be implicitly convertible to each other.

> I think we should switch more of our shared_ptrs to using that macro so that cherry-picking between branches remains super easy.

For `shared_ptrs` I agree that using typedefs is the best way forward. The first step can be to eliminate the explicit mention of a specific `shared_ptr` type, using the macro mentioned by @davetcoleman where possible. I don't think anyone will have objections to that, so I'll put it on my todo list.

> Maybe it makes more sense to try to encapsulate all usage of pointers into typedefs in kinetic and switch in lunar?

@v4hn: One reason why I do like the idea of updating the API in Kinetic is that if we don't, it will be four years before the change will really have any effect because most people will stick with LTS versions.

v4hn, 2016-08-16:
On Tue, Aug 16, 2016 at 09:07:50AM -0700, Maarten de Vries wrote:
> Considering that both `boost::function` and `std::function` are functors, they should be implicitly convertible to each other.

That was my idea too. My point was: if the API practically does not change because the user does not have to touch their code: go forward.
If the change would entail that a whole project has to switch to `std::shared_ptr`, that's something where we should be a bit more cautious..

> For `shared_ptrs` I agree that using typedefs is the best way forward.
> The first step can be to eliminate the explicit mention of a specific `shared_ptr` type, using the macro mentioned by [...]

+2 This is really important because we will have to support cherry-picks to boost-pointers for a while.

The reason for my proposal to change the `shared_ptr` only in lunar was to give people time to move to the typedefs,
because, as dave pointed out, there are a few rough edges in MoveIt where there are *no* typedefs at the moment.
So we can't blame it on the users for not using the appropriate typedef if their code is broken because they explicitly used boost pointers.

davetcoleman, 2016-08-16:
Just a note: its trivial to ask users to replace the word boost with std, and the C++ community has been doing it since 2011.

v4hn, 2016-08-16:
On Tue, Aug 16, 2016 at 10:37:41AM -0700, Dave Coleman wrote:
> Just a note: its trivial to ask users to replace the word boost with std, and the C++ community has been doing it since 2011.

And they still haven't won. So it can't be that trivial :-)

de-vri-es, 2016-08-17:
> For shared_ptrs I agree that using typedefs is the best way forward. The first step can be to eliminate the explicit mention of a specific shared_ptr type, using the macro mentioned by @davetcoleman where possible. I don't think anyone will have objections to that, so I'll put it on my todo list.

Work-in-progress available for (p)review at #70

de-vri-es, 2016-08-21:
Another one at #97, this time for `moveit_planners/*` and `moveit-experimental/*`.

de-vri-es, 2016-08-31:
If #193 is merged, that should be all explicit `shared_ptrs` regarding moveit types (barring `moveit_experimental`, chomp and sbpl).

However, there are 4 uses of `boost::weak_ptr` for moveit types in `moveit_core`. These are not exposed in the API, but they will break the build when switching to `std::shared_ptr`. Additionally, there are two classes inheriting from `boost::enable_shared_from_this`.

For the effort of switching to `std::shared_ptr`, these are minor issues that can be solved by hand when the time comes. However, users of moveit currently have no clean way to work with `weak_ptrs` without explicitly using `boost::weak_ptr`.

The easiest method to address that is by also defining `${Class}WeakPtr` and `${Class}ConstWeakPtr`. But that does introduce a huge number of new type definitions. I don't expect name clashes though.

Thoughts?

v4hn, 2016-09-01:
On Wed, Aug 31, 2016 at 02:48:47PM -0700, Maarten de Vries wrote:
> However, users of moveit currently have no clean way to work with `weak_ptrs` without explicitly using `boost::weak_ptr`.
>
> The easiest method to address that is by also defining `${Class}WeakPtr` [...]

You're right, `weak_ptr`s are a problem if we tell people to *always* use the `${Class}Ptr` typedefs.
On the other hand I'm not sure how useful they are w.r.t. MoveIt class instances. I never used them with MoveIt datatypes up to now.

I'd propose to add a pull request for the additional typedefs you proposed and switch MoveIt's internal uses to them.
If someone else considers this useful, I would support to merge it.

davetcoleman, 2016-09-08:
> I'd propose to add a pull request for the additional typedefs you proposed and switch MoveIt's internal uses to them.

+1

de-vri-es, 2016-09-08:
To get a feel of what would be required to switch to `std::shared_ptr` for the API, I did so on this branch (currently based on #206): 
https://github.com/delftrobotics-forks/moveit/tree/std-shared-ptr-api

I would say it went pretty smooth. Besides switching the typedefs, assuming `weak_ptr` is taken care of, I also had to:
* Switch to `std::enable_shared_from_this`
* Switch to `std::make_shared`
* Use `PluginLoaded::createUnmanagedInstace` and wrap in the correct `shared_ptr` manually.

The last point can even be split off and put in a PR now. since it is compatible with `boost::shared_ptr`.

davetcoleman, 2016-09-08:
Should we make a new version interface for the ``pluginlib::ClassLoader`` in kinetic that optionally returns an std::shared_ptr? Seems like [fixing it](https://github.com/ros/pluginlib/blob/indigo-devel/include/pluginlib/class_loader.h#L92) upstream would be the better route

v4hn, 2016-09-09:
On Thu, Sep 08, 2016 at 03:21:26PM -0700, Dave Coleman wrote:
> Seems like fixing it upstream would be the better route

+1

de-vri-es, 2016-09-09:
> Should we make a new version interface for the pluginlib::ClassLoader in kinetic that optionally returns an std::shared_ptr? Seems like fixing it upstream would be the better route

I do have in idea that would be compatible with both `boost::` and `std::shared_ptr`. I'll see what the pluginlib people think of it. But if they want differently named functions for `boost::shared_ptr` and `std::shared_ptr, the raw pointer way is the only one that would work for both types.

davetcoleman, 2016-09-09:
Please open an issue/pull request with them. In the meantime, I am concerned about further delays in getting MoveIt! released into kinetic so perhaps your suggested solution would be quicker at least in the shortrun

de-vri-es, 2016-09-09:
I just realized I forgot about the deleters. I'll wrap it up quickly with a different solution for `weak_ptr` and class loaders for now.

de-vri-es, 2016-09-11:
Hmm, it is slightly more complicated. Creating unmanaged instances with `ClassLoader` changes the behaviour of all `ClassLoaders` in the process (it disables on-demand unloading of libraries). I'll see if that can be avoided somehow. In the mean time I filed ros/class_loader#38 (will need to do pluginlib after that).

davetcoleman, 2016-09-27:
@de-vri-es can this issue be closed now?

de-vri-es, 2016-09-27:
Yeah. There is one more change I'd like to make, which is replacing `boost::function` with `std::function`. But I don't think we need this issue to track that. It's a much smaller change and API compatible so I'll just file a PR when it is done.

closed 2016-09-27 by de-vri-es