moveit/moveit #52: Migrate docker to main repo

davetcoleman, 2016-08-16:
Steps to complete migration:

- [ ] merge first PR
- [ ] Setup on dockerhub and verify it works
- [ ] cherry-pick this to I/J with the keyword "kinetic" replaced.
- [ ] Add documentation on moveit website
- [ ] Switch moveit_ci to use this docker image
- [ ] Document that moveit_docker is deprecated

davetcoleman, 2016-08-16:
@v4hn if you or any other maintainers are interested, I would like to share the ``moveit`` account on dockerhub with others besides myself, so that I'm not the only one with access. To do this you'll need to create an account on dockerhub.com and tell me your username (hopefully the same as github)

v4hn, 2016-08-16:
On Tue, Aug 16, 2016 at 12:13:17PM -0700, Dave Coleman wrote:
> @v4hn if you or any other maintainers are interested, I would like to share the ``moveit`` account on dockerhub with others besides myself, so that I'm not the only one with access. To do this you'll need to create an account on dockerhub.com and tell me your username (hopefully the same as github)

I'm not interested as I don't use docker. I feel it brings about bad habits.
But yes, a bus factor >1 should be ensured, so it would be great if someone else could step up.

davetcoleman, 2016-08-16:
@130s ?

davetcoleman, 2016-08-16:
@rhaschke actually gave the the idea to use docker for testing and maintaining moveit - I know he uses them

rhaschke, 2016-08-19:
> @rhaschke actually gave the the idea to use docker for testing and maintaining moveit - I know he uses them

That was a misunderstanding, actually ;-)
However, I like the idea of docker in principle. But I'm not really acquainted to it yet.
I just created a dockerhub account, named `rhaschke`.

davetcoleman, 2016-08-22:
@rhaschke has been added. I also forgot but @ruffsl is an owner of the dockerhub team as well, since he made the OSRF images that moveit's builds on top of.

davetcoleman, 2016-08-24:
I just finished fully migrating our docker containers to the consolidated moveit repo

closed 2016-08-24 by davetcoleman