moveit/moveit #61: Params for trajectory execution are not correctly set currently?

wkentaro, 2016-08-17:
Using `moveit_setup_assistant` templates, parameters `allowed_execution_duration_scaling` and `allowed_goal_duration_margin` is set for `move_group` in `trajectory_execution.launch.xml`:
https://github.com/ros-planning/moveit/blob/7a3057a68927dfcc9bf2336c794597917cb9d2a6/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/trajectory_execution.launch.xml#L9-L12

It is included in `move_group.launch` with namespace `move_group`:
https://github.com/ros-planning/moveit/blob/7a3057a68927dfcc9bf2336c794597917cb9d2a6/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/move_group.launch#L28-L37

However, in rqt_reconfigure, the parameters seems being not set correctly as below and correct namespace looks `/move_group/trajectory` not `/move_group`:
[image]
As the figure shows, the parameters `<param name="allowed_execution_duration_scaling" value="1.2"/>` and `<param name="allowed_goal_duration_margin" value="0.5"/>` is not set.

I provided a patch for this: https://github.com/ros-industrial/fanuc_experimental/pull/24/files, but not sure if this is the correct solution.
Any ideas?

gavanderhoorn, 2016-08-17:
More users seem to have encountered this, see [How do I disable execution_duration_monitoring ?](http://answers.ros.org/question/196586) on ROS Answers fi.

davetcoleman, 2016-08-17:
+1 to fixing this in the templates of the setup assistant

rhaschke, 2016-08-18:
It's not yet solved completely. See https://github.com/ros-planning/moveit/pull/68#issuecomment-240586343.

wkentaro, 2016-08-18:
Fixed via https://github.com/ros-planning/moveit/pull/72 https://github.com/ros-planning/moveit/pull/73

commit de886871 (2016-08-17): msa: push traj exec monitor params down into proper ns. Fix #61. (#68)

The dynamic_reconfigure parameters this launch file is providing values for
already has them in the 'trajectory_execution' namespace.
commit 8fbf49f6 (2016-08-17): msa: push traj exec monitor params down into proper ns. Fix #61. (#68)

The dynamic_reconfigure parameters this launch file is providing values for
already has them in the 'trajectory_execution' namespace.
commit ce306632 (2016-08-17): msa: push traj exec monitor params down into proper ns. Fix #61. (#68)

The dynamic_reconfigure parameters this launch file is providing values for
already has them in the 'trajectory_execution' namespace.
closed 2016-08-18 by wkentaro