moveit/moveit #64: Cartesian Coordinate Planning

mat-l, 2016-08-17:
Hello,
I want to use an kuka kr6 robot for a pick and place task. The coordinates come in Cartesian coordinates. All I want to do is write a little application that connects theses coordinates via paths that I want to create with MoveIt and sends that path to the real robot afterwards. Therefore I want to use the Move Group Interface with Python. I tried to use the pr_2 moveit documentation (http://docs.ros.org/hydro/api/pr2_moveit_tutorials/html/planning/scripts/doc/move_group_python_interface_tutorial.html) and make it work with the kuka instead.
My first attempt  is just to have a working Python script that I can change afterwards.

[cube_move_launch.txt](https://github.com/ros-planning/moveit/files/422775/cube_move_launch.txt)
[cube_move_py.txt](https://github.com/ros-planning/moveit/files/422774/cube_move_py.txt)

I put the launch file into the launch folder within the kr6 moveit config folder from the github kuka_experimental package and the python script is the moveit config folder (with config, launch etc.)
When I try to run the script I get : 
InvalidPackage: Invalid package manifest "/home/zema-4by3/catkin_ws_kr6/src/kuka_experimental-kr6r900sixx_moveit_rsi_convenience/kuka_resources/package.xml": The manifest contains invalid XML:
maximum recursion depth exceeded while calling a Python object.
My first issue here is that the package.xml found is different from the one that should be used. Is there a way to force catkin to search a defined path for the package.xml?
My second issue will be that I don't know how to send the path information created from RViz to the actual robot. Do I have to subscribe to a topic to accomplish that?
Thank you for your help.

gavanderhoorn, 2016-08-22:
Triggered by (as the maintainer of `kuka_experimental`):

```
InvalidPackage: Invalid package manifest "/home/zema-4by3/catkin_ws_kr6/src/kuka_experimental-kr6r900sixx_moveit_rsi_convenience/kuka_resources/package.xml": The manifest contains invalid XML:
maximum recursion depth exceeded while calling a Python object.
```

Did you just `git clone ..` those files? Have you made any changes?

Because I haven't had any reports about the manifest of `kuka_resources` being invalid, and I cannot reproduce this.

It would seem your manifest issue is unrelated to your MoveIt question.

mat-l, 2016-08-26:
Sorry for the late response.
I used git clone to get the kuka_experimental package and it worked fine. I still don't know why exactly I get this error but I guess you are right and it is something in my system. 
Maybe I messed up the workspace overlay.
Thank you anyway. I close the issue then.

closed 2016-08-26 by mat-l

gavanderhoorn, 2016-08-29:
I didn't imply that you should close your issue, just that the part about the errors with the package manifest of `kuka_resources` should probably be reported somewhere else (ie: at `ros-industrial/kuka_experimental`).

If after fixing that issue you can then successfully execute your scripts, it would make sense to close this issue.

If you are still having issues with the `package.xml` from `kuka_resources` I'd be glad to help you over at [ros-industrial/kuka_experimental](https://github.com/ros-industrial/kuka_experimental).

mat-l, 2016-09-02:
I appreciate the offer!
I was not able to execute the script after all but I found another potential way to complete the task by using the Cartesian Path Planner Plugin for MoveIt!.
I am pretty sure that something on my PC causes the problems and not your package.xml . 
That is why I closed the issue. If I get the same error again after cleaning up my workspace I will make sure to file a new issue and search for a real solution.