moveit/moveit #67: Consolidate kinematics plugins into one folder, including ikfast

davetcoleman, 2016-08-17:
Per @v4hn [recently](https://github.com/ros-planning/moveit/pull/44#issuecomment-239885875):

> Personally, I don't understand why the other kinematics plugins are in there and wouldn't ``moveit_ikfast`` in there either...
In my opinion it would make more sense to create a new top-level-folder/package ``moveit_kinematics_plugins`` in the repo and make it part of the meta-package. This should even be possible without breaking anything, because the plugin-descriptions do not reference the package name and the only way the plugins are used is through their plugin descriptions.

Moving ikfast into a new package would also fix a versioning problem we're having ([discussion](https://github.com/davetcoleman/moveit_merge/issues/9#issuecomment-239514240)) because currently its too high (v3 instead of ~v0.7.x)

I'm proposing someone create a ``moveit_kinematics`` package in this repo that removes the solvers from ``moveit_ros/planners``

andreaskoepf, 2016-08-20:
@davetcoleman I would like to take this task for the WMD. Since I am no collaborator or member of the MoveIt organization I think I cannot assign myself directly in the task metadata. I am not sure if you can assign external people ... otherwise I suggest that we just state that we want to work on an issue in a comment. What do you think?

v4hn, 2016-08-20:
On Sat, Aug 20, 2016 at 03:12:30AM -0700, Andreas Köpf wrote:
> @davetcoleman I would like to take this task for the WMD.
> Since I am no collaborator or member of the MoveIt organization I think I cannot assign myself directly in the task metadata.

Yes, Dave probably missed that only maintainers can be assigned issues.

Hereby I proclaim this to be your issue for WMD :)

> I am not sure if you can assign external people ... otherwise I suggest that we just state that we want to work on an issue in a comment. What do you think?

In general, if you're first to comment "I'll do this", then it's yours.

davetcoleman, 2016-08-22:
> Yes, Dave probably missed that only maintainers can be assigned issues.

Yep, I didn't think to check that but its a long standing [feature request](https://github.com/isaacs/github/issues/100). I agree with @v4hn - you just need to say "I'll do this". I also just added an [assigned](https://github.com/ros-planning/moveit/labels) label that a maintainer can manually/optionally add to an issue once someone claims it.

Thanks @andreaskoepf

davetcoleman, 2016-08-22:
I've updated the procedures for WMD: https://github.com/ros-planning/moveit.ros.org/pull/39

gavanderhoorn, 2016-08-22:
A bit late here perhaps, but imho it might not be a good idea to merge `moveit_ikfast` into a `moveit_kinematics` package.

In contrast to 'all' the other MoveIt kinematics plugins (KDL and the recently added LMA one), `moveit_ikfast` is not a plugin in itself, but only a set of template files and a Python script that generates C++ code *for* something that is a MoveIt compatible kinematics plugin. Building `moveit_ikfast` does not even compile any (C++) sources, as there is nothing to build.

Each and every IKFast plugin generated is unique (sort of) to a specific URDF / Collada model, but those sources will never be integrated into MoveIt: that wouldn't make much sense. This is completely different from how the KDL and LMA plugins work: they are generic, the solvers are configured dynamically at runtime and the compiled binaries are installed as part of a binary MoveIt installation.

Conceptually it would seem odd to me to place what is essentially a source code generator -- that generates code tightly coupled to a specific robot model -- in the same package as a set of static sources for libraries that are generically reusable for just about any robot model that can be modelled with a URDF.

Personally, I consider `moveit_ikfast` a package similar to the ROS message generators: stand-alone packages that happen to generate code that is/should be compatible with some other -- independently managed -- infrastructure.

Would it not be easier (looking at #96) and more logical to leave `moveit_ikfast` in its own repository?

@jrgnicho: thoughts?

v4hn, 2016-08-22:
On Mon, Aug 22, 2016 at 09:52:48AM -0700, G.A. vd. Hoorn wrote:
> A bit late here perhaps, but imho it might not be a good idea to merge `moveit_ikfast` into a `moveit_kinematics` package.

Your timing is perfectly fine. :)

> In contrast to 'all' the other MoveIt kinematics plugins (KDL and the recently added LMA one), `moveit_ikfast` is not a plugin in itself

Yes. We were available of that when this was proposed.

> Would it not be easier (looking at #96) and more logical to leave `moveit_ikfast` in its own repository?

@davetcoleman merged it into the moveit repository and nobody complained in the rather longish process. Sounds like you should have. :)
I believe as ikfast is quite independent of the rest of the code, the only reason why it was added is "because it's moveit".

As it stands we have two options:

- make it a part of the new `moveit_kinetics_plugins` package

- revert the deprecation of https://github.com/ros-planning/moveit_ikfast and remove the package from `indigo-devel` and `kinetic-devel` here, too.

It's probably best if you or @jrgnicho decide which option we take.

davetcoleman, 2016-08-22:
These are good points though indeed we have thought of them. 

ikfast is a popular IK solution for MoveIt! and, if it works for your robot, arguably the best. I don't see the harm in grouping it with the other ik solvers (just two at this point) in MoveIt!, even if they only generate code. 

The great benefit of having it together in this repo is that when we make API changes to KinematicsBase, such as the [getPositionIK returning multiple solutions](https://github.com/ros-planning/moveit_ikfast/pull/57) and the upcoming [dependency on RobotModel within KinematicsBase](https://github.com/ros-planning/moveit_ros/issues/731), we can fix all the code at once in one PR, without having to track and time multiple PRs.

In addition, I've been planning to suggest someone work on refactoring ikfast to actually *have* its generated plugins depend on the moveit_ikfast package. A few years ago I actually created this change, but it has fallen behind. Instead of the script generator copying the plugin code for every robot it is generated for, instead it pulls [an include file](https://github.com/davetcoleman/moveit_ikfast/blob/indigo-all/include/moveit_ikfast/ikfast61_moveit_plugin_template.cpp) that is common to all IKFast-running robots. This include file is pull in from a [very simple file](https://github.com/davetcoleman/moveit_ikfast/blob/indigo-all/templates/ikfast61_moveit_plugin_template.cpp#L54) that is specific to each robot. This way whenever we make a breaking change to how kinematics works in MoveIt!, or an upgrade to the ikfast interface, it is reflected out to all robots using it automatically. Thoughts?

andreaskoepf, 2016-08-23:
Ok, I think this needs more discussion? I will then move no to another task.

v4hn, 2016-08-23:
No matter whether `moveit_ikfast` is merged in the new package or not, we should have the package for the other (two) plugins.
Thus, this issue is still valid for today.

@andreaskoepf I removed the assigned again, as you said you'll work on a different task

davetcoleman, 2016-09-23:
I have proposed the move: https://github.com/ros-planning/moveit/pull/247

gavanderhoorn, 2016-09-24:
@davetcoleman wrote:
> In addition, I've been planning to suggest someone work on refactoring ikfast to actually *have* its generated plugins depend on the moveit_ikfast package. A few years ago I actually created this [...]

I can see why you'd like this, but I believe there are benefits to having IKFast plugins being stand-alone, and not automatically being upgraded whenever upstream decides they should be. If anything, it provides a way for a critical part in your planning infrastructure to remain stable. This is going to be the old bw-compatibility / stability vs maintenance overhead / progress debate again, but it's just important to realise that there are those who think that is important.

Nevertheless, I understand why you feel the current setup is problematic, and I don't think I can come up with an alternative in the near future.

davetcoleman, 2016-09-24:
... if you are against things being automatically upgraded, just build moveit from source and never do a pull again

Benefits to my suggestion
- I'm working on a two arm robot and need to improve the IKFast plugin for both arms at the same time. 
- Someone fixes a long existing bug or makes a speedup in the ikfast plugin and it actually gets shared with everyone
- When upgrading to the newest version of ROS or MoveIt!, the plugin is automatically upgraded for you instead of having to fix code rot for every plugin ever made

davetcoleman, 2016-09-24:
Note I already have the next PR, that implements including a shared IKFast plugin code file, implemented [in this branch](https://github.com/davetcoleman/moveit/blob/kinetic-kinematic-ikfast/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp)

gavanderhoorn, 2016-09-24:
As I said, I don't have (the time for) a better alternative right now, so go ahead. I'm not against anything, I just thought it important to write down the sentiment somewhere once.

davetcoleman, 2016-09-25:
Thanks for your sentiments @gavanderhoorn, sorry if I was too direct. Further discussion is of course welcomed.

davetcoleman, 2016-10-18:
This has been completed https://github.com/ros-planning/moveit/pull/247

closed 2016-10-18 by davetcoleman