moveit/moveit #75: Can't launch demo.launch with db:=true

ugocupcic, 2016-08-18:
I'm trying to connect to the warehouse, and in jade and kinetic I'm not able to do so. warehouse_ros does not seem to contain the `mongo_wrapper_ros.py` which is listed in the launch file [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/warehouse.launch). 

Is anyone else using the warehouse?

v4hn, 2016-08-18:
Heya,

the mongodb interface has been changed to a more generic database plugin interface in jade and kinetic.
However, at the moment we don't have a working mongodb plugin (or any other system for that matter) in the upstream repositories.
The work of @TheBrewCrew for this plugin can be found here, I believe:
https://github.com/TheBrewCrew/warehouse_ros_mongo/tree/jade-devel

This will need further cleanup and testing, before we can provide this officially..

ugocupcic, 2016-08-18:
thanks @v4hn I managed to get warehouse_ros_mongo to at least compile using these [instructions](https://github.com/mongodb/mongo-cxx-driver/wiki/Download-and-Compile-the-Legacy-Driver) - branch 26compat. Still have issues actually connecting to it but as you said it's not ready yet so I can move that part of the discussion over there.

davetcoleman, 2016-08-18:
@ugocupcic would be great if you could take the lead on getting warehouse_ros_mongo cleanedup and merged back into MoveIt!. we've discussed it and none of the maintainers use that functionality in moveit and thus are not motivated to work on it.

ugocupcic, 2016-08-19:
I managed to get the connection to the database working with warehouse_ros_mongo + a few changes in the default launch files from moveit:

in `warehouse_settings.launch.xml`, I added:

```
   <param name="warehouse_plugin" value="warehouse_ros_mongo::MongoDatabaseConnection" />
```

and I replaced `warehouse_ros` by `warehouse_ros_mongo` in `warehouse.launch` to be able to spawn the mongo_wrapper_ros.py. 

The db connection is working nicely. 

Regarding cleaning it up and merged back into moveit, I'm not sure how much work is involved for this. Currently building the warehouse_ros_mongo package depends on a source install of the mongo cpp driver.... 

We're talking about a pr that adds the 2 packages (warehouse_ros / warehouse_ros_mongo) to this new moveit repository right? 

I'll see what we can do at Shadow.

v4hn, 2016-08-19:
On Fri, Aug 19, 2016 at 12:46:07AM -0700, Ugo Cupcic wrote:
> I managed to get the connection to the database working with warehouse_ros_mongo + a few changes in the default launch files from moveit:
> in `warehouse_settings.launch.xml`, I added:
> ```
>    <param [...]

Great!

Could you provide a pull-request for this for the `moveit_setup_assistent` templates?
Required-to-build-from-source is better than having no support at all for the database in jade and kinetic!

> Regarding cleaning it up and merged back into moveit, I'm not sure how much work is involved for this.
> Currently building the warehouse_ros_mongo package depends on a source install of the mongo cpp [...]

@130s @mikeferguson do we want to have `warehouse_ros` in moveit again?
As it provides a plugin interface, does not add database dependencies anymore, and is coupled with MoveIt, I think that would make sense.

`warehouse_ros_mongo` should definitely **not** be merged into the moveit repo,
but it might make sense to host it in the ros-planning organization with the MoveIt project having control over it.

The long-term goal should be to have the plugin running with one of the MongoDB client libraries released in ubuntu.
I have no idea how much work that is though.

> I'll see what we can do at Shadow.

Thanks for your support!

ugocupcic, 2016-08-19:
> warehouse_ros_mongo + a few changes in the default launch files from moveit:
> />
> `warehouse.launch` to be able to spawn the mongo_wrapper_ros.py.
>
> Great!
>
> Could you provide a pull-request for this [...]
I'll submit the pr in a bit.

> much work is involved for this.
> install of the mongo cpp driver....
> warehouse_ros_mongo) to this new moveit repository right?
>
> @130s @mikeferguson do we want to have `warehouse_ros` in moveit again? [...]

+1 on all those.

In the meantime, we could may be modify the kinetic source install in
docker to have a workable environment for the db?

>
> > I'll see what we can do at Shadow.
>
> Thanks for your support!

-- 
Ugo Cupcic*Chief Technical Architect*+44 20 7700 2487

ugocupcic, 2016-08-19:
Also added a few instructions on how to install the mongo drivers here https://github.com/TheBrewCrew/warehouse_ros_mongo/pull/1

v4hn, 2016-08-22:
Addressed by #81 , thanks for resolving this!

closed 2016-08-22 by v4hn