moveit/moveit #80: database_grasp_planning service with moveit_msgs/Grasp.msg

v4hn, 2016-08-19:
MoveIt's PickPlace pipeline [supports using external ros-services for Grasp-Planning](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp#L332).
This feature can be used to pick objects from client code via [`move_group_interface->pick("objectname")`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group.h#L659).
In this case the Pick action server invokes [the external service `database_grasp_planning`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp#L71) (that the user has to implement and start beforehand) to compute possible grasps.

The problem is that this service adheres to the specification [`manipulation_msgs/GraspPlanning.srv`](https://github.com/ros-interactive-manipulation/manipulation_msgs/blob/hydro-devel/srv/GraspPlanning.srv). This specification uses the package's version of the `Grasp.msg` which has been reworked since and is now provided in [`moveit_msgs/Grasp.msg`](https://github.com/ros-planning/moveit_msgs/blob/indigo-devel/msg/Grasp.msg). The `manipulation_msgs` package [is deprecated](https://github.com/ros-interactive-manipulation/manipulation_msgs/issues/9).
As a result, all grasps returned by the service [have to be copied field-by-field](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp#L463) and some fields are even missing.
This makes it impossible to [provide time information for the gripper motions](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/manipulation/pick_place/src/plan_stage.cpp#L89) during the pick and it makes it impossible [to specify the `post_grasp_retreat` direction](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp#L489) of the grasp when using the service interface.

To fix this, the service specification should be copied to `moveit_msgs` and adapted to use `moveit_msgs/Grasp`. This will probably add a number of other messages too.
A service client for the new ROS-service should be added [next to the existing one](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp#L71). It should connect to a service named `plan_grasps` (to avoid name collisions).

In `indigo-devel` and `jade-devel` the old service should still be called if the new service [does not exist](https://github.com/ros/ros_comm/blob/indigo-devel/clients/roscpp/include/ros/service_client.h#L156), in `kinetic-devel`, the old service can be removed entirely to simplify the manipulation code.

We might want to discuss the exact shape of `moveit_msgs/GraspPlanning.srv` before merging this upstream, but a full implementation that only has to be refined afterwards is definitely possible in one day of work.

toliver, 2016-08-22:
We can deal with this task for the WMD.

dg-shadow, 2016-08-23:
Hi. I'm about to start work on this. My plan is to fix it in Kinetic first then add the changes back into Jade and Indigo. Is there any reason not to do it that way? Thanks :)

v4hn, 2016-08-23:
On Tue, Aug 23, 2016 at 03:06:09AM -0700, dg-shadow wrote:
> Hi. I'm about to start work on this. My plan is to fix it in Kinetic first then add the changes back into Jade and Indigo. Is there any reason not to do it that way? Thanks :)

Sounds goot to me :)

dg-shadow, 2016-08-23:
Today has been one of those days full of distractions and set backs, and I never really got started with this :( I am conceding defeat, so it's free to be assigned to someone else.

My apologies. I hope the next candidate does better ;)

v4hn, 2016-08-23:
too bad, thanks anyway :)

v4hn, 2016-08-23:
It's definitely not that easy a task, I wasn't sure if I should put it up for WMD.
So I'm the only one to blame if there is no pull-request up for this by tomorrow. :)

v4hn, 2016-10-12:
This has basically been implemented in https://github.com/ros-planning/moveit/pull/264 and https://github.com/ros-planning/moveit_msgs/pull/32 , even though the actual `database_grasp_planning` interface was not touched.
Closing.

closed 2016-10-12 by v4hn