moveit/moveit #86: moveit_commander fails in Kinetic due to pyassimp regression

rhaschke, 2016-08-19:
In Xenial with pyassimp 3.2, importing moveit_commander fails with:
```
>>> import moveit_commander
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/vol/sandbox/ros/install/lib/python2.7/dist-packages/moveit_commander/__init__.py", line 3, in <module>
    from planning_scene_interface import *
  File "/vol/sandbox/ros/install/lib/python2.7/dist-packages/moveit_commander/planning_scene_interface.py", line 48, in <module>
    import pyassimp
  File "/usr/lib/python2.7/dist-packages/pyassimp/__init__.py", line 1, in <module>
    from .core import *
  File "/usr/lib/python2.7/dist-packages/pyassimp/core.py", line 23, in <module>
    from . import structs
ImportError: cannot import name structs
```
@davetcoleman Did you ever tried to run moveit from python in kinect?

rhaschke, 2016-08-19:
By the way, this was revealed by the new [integration_tests](https://github.com/ros-planning/moveit/pull/83) ;-)

rhaschke, 2016-08-19:
The bug is already filed on [launchpad](https://bugs.launchpad.net/ubuntu/+source/assimp/+bug/1589949) and a patch exists.

davetcoleman, 2016-08-19:
No, I don't use commander or any Python interfaces

davetcoleman, 2016-08-19:
Nice job with the integration tests!!

v4hn, 2016-08-22:
@rhaschke can we close this? It's a bug in the ubuntu package that should be fixed in a while (I presume).

davetcoleman, 2016-08-22:
Seems like this issue should be left open until fixed for other user's reference, or we should document it somewhere else like the moveit_command tutorial

v4hn, 2016-08-22:
On Mon, Aug 22, 2016 at 11:21:52AM -0700, Dave Coleman wrote:
> Seems like this issue should be left open until fixed for other user's reference

Ok, let's leave it open then.

davetcoleman, 2016-11-03:
I just tried running the python move_group_interface tutorial and ran into this bug again. I just checked the launchpad link and even though they claim it has been released, Ubuntu 16.04 still lacks the fixed version. I asked a question on that forum, not sure how else to proceed.

MarkusEich, 2016-11-25:
same problem here. Moveit commander is not usable due to ImportError: cannot import name structs. Is there a way to use moveit from python without using the commander? I basically need only a move_to for the endeffector (of Baxter).

davetcoleman, 2016-11-25:
You can try https://github.com/mikeferguson/moveit_python - I'd like to eventually integrate that into our main release

MarkusEich, 2016-11-30:
Was able to fix it. Pull the latest version from assimp. https://github.com/assimp/assimp.git and build it. make install will copy it to /usr/local/lib. Then go to /port/PyAssimp in the source folder and run python setup.py install. This should create the necessary files in /usr/local/lib/python2.7/dist-packages/pyassimp.

move_group_python_interface.py and using the commander works now flawlessly (tested with 16.04 and kinetic)

alexpad05, 2016-11-30:
> Was able to fix it. Pull the latest version from assimp. https://github.com/assimp/assimp.git and build it. make install will copy it to /usr/local/lib. Then go to /port/PyAssimp in the source [...]

I can confirm this fixes the problem

130s, 2017-01-11:
I suppose this can be closed now that Kinetic release https://github.com/ros-planning/moveit/issues/18 is already done. Please report/reopen if the issue persists.

RunOrVeith, 2017-01-16:
I still have this issue, with ros-kinetic-moveit installed and fully updated Ubuntu 14.04 system.

130s, 2017-01-17:
I have a chance to run this on Kinetic and confirmed that it still occurs.

```
$ ipython
Python 2.7.12 (default, Nov 19 2016, 06:48:10)
Type "copyright", "credits" or "license" for more information.

IPython 2.4.1 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import moveit_commander
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-adfba809480d> in <module>()
----> 1 import moveit_commander

/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/__init__.py in <module>()
      1 from exception import *
      2 from roscpp_initializer import *
----> 3 from planning_scene_interface import *
      4 from move_group import *
      5 from robot import *

/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/planning_scene_interface.py in <module>()
     46 except:
     47     # support pyassimp > 3.0
---> 48     import pyassimp
     49
     50 # This is going to have more functionality; (feel free to add some!)

/usr/lib/python2.7/dist-packages/pyassimp/__init__.py in <module>()
----> 1 from .core import *

/usr/lib/python2.7/dist-packages/pyassimp/core.py in <module>()
     21 logger.addHandler(logging.NullHandler())
     22
---> 23 from . import structs
     24 from . import helper
     25 from . import postprocess

ImportError: cannot import name structs

In [2]:
Do you really want to exit ([y]/n)?
$ rosversion moveit_commander
0.9.3
$ apt-cache policy ros-kinetic-moveit-commander |grep Inst
   Installed: 0.9.3-0xenial-20170113-175334-0800
$ apt-cache policy python-pyassimp |grep Inst
   Installed: 3.2~dfsg-3
```

I've reported to the upstream and ask if they can make a new release into Xenia. https://github.com/assimp/assimp/issues/1142

v4hn, 2017-01-17:
Upstream developers are usually neither interested in nor in charge of distribution packages.
*Upstream* has no problem - Let's see how/if they respond.

The right place to complain should be https://launchpad.net/ubuntu/+source/assimp
This is the resource of the ubuntu package of assimp.

130s, 2017-01-18:
I see, opened [a ticket on launchpad](https://bugs.launchpad.net/ubuntu/+source/assimp/+bug/1657374).

lifangda01, 2017-02-08:
@alexpad05 I had the same issue. The simple fix was to change the following lines in `planning_scene_interface.py` from
```Python
if len(face.indices) == 3:
    triangle.vertex_indices = [face.indices[0], face.indices[1], face.indices[2]]
```
to
```Python
if len(face) == 3:
    triangle.vertex_indices = [face[0], face[1], face[2]]
```
After this my mesh imports worked.

dbolkensteyn, 2017-02-20:
@lifangda01@alexpad05 Isn't that a separate issue for which a PR should be opened on this repo?

davetcoleman, 2017-02-24:
I believe #450 fixes this issue

closed 2017-02-24 by davetcoleman

jlack1987, 2017-03-17:
What was the fix for the `pyassimp` regression? Did you somehow remove the dependency on `pyassimp` completely considering simply importing it throws an exception?

130s, 2017-03-17:
@jlack1987 https://github.com/ros-planning/moveit/issues/86#issuecomment-282336998

iglstone, 2017-03-30:
sudo pip uninstall pyassimp
and then,
sudo pip install pyassimp ,
which will install pyassimp 3.3,  solved the problems..may help.

guihomework, 2017-05-13:
For those who cannot upgrade to 3.3, which indeed fixed the problem on our side, if you patch 3.2 be careful about this patch https://launchpadlibrarian.net/263969718/patch.txt mentioned earlier. This solves the import problem but fails on execution of this line https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_commander/src/moveit_commander/planning_scene_interface.py#L95 

```
AttributeError: 'AssimpLib' object has no attribute 'load'
```
This sounds logical since this function was removed by the patch.  But in fact the patch is wrong, and should remove load_mem (because what is returned by the helper lib is a _loadfromfile_, not a _loadfrommem_)

I then did a different patch
```
-    load, load_mem, release, dll = helper.search_library()
+   load, release, dll = helper.search_library()
```
which solved both problems (until now). But in upstream the load and load_mem still exists, only the ubuntu released version ```helper.py``` function returns a single _load_ function. 
I reported this in the launchpad bug report, hoping the maintainers would fix their own patch, which might have more chance to occur than a backport of 3.3 to xenial.

jlack1987, 2017-05-13:
Yeah this issue has been a real pain. I have raised an issue to the developers [here](https://github.com/assimp/assimp/issues/1209)

guihomework, 2017-05-15:
@jlack1987 yes, but I think this is not at the developer's side that it can be fixed, because in the github repository, the load_mem function is not removed, this is a problem in the maintained debian package as far as I could see. Hence my comments on 2 launchpad bug reports when I discovered that.

jlack1987, 2017-05-15:
Installing via `pip` didn't help here. Still can't even import `pyassimp` in a terminal. Tried both python3 and python2 versions. Can verify that installing pyassimp gives version `pyassimp-3.3`

jlack1987, 2017-06-07:
I believe this should be reopened. There is a way to fix this in an automated fashion that doesn't require users to mess with source code of dependencies. The problem is that the debian version of `pyassimp` is broken in Ubuntu 16.04, but the `pip` version works; however, if you install the `pip` version you have to make sure to install the `libassimp-dev` debian as well. 

Simply adding the necessary rosdep keys [here](https://github.com/ros/rosdistro/tree/master/rosdep) and updating the way the dependencies are called out in the package.xml to the new keys would solve this problem so the necessary dependencies could be installed via `rosdep install`. This would save users of moveit a lot of headache I think.

v4hn, 2017-06-08:
@jlack1987 feel free to file a pull-request with your proposal and we can discuss what else is required for it there.

It doesn't look like https://bugs.launchpad.net/ubuntu/+source/assimp/+bug/1657374 will be resolved unless some affected ROS user follows through with https://wiki.ubuntu.com/StableReleaseUpdates .

I think the next-best solution is to have our own pyassimp package (the pip version) for ubuntu 16.04 kinetic.
@tfoote could we do that? Without this, Python MoveIt support is broken at runtime on ubuntu 16.04.

srsidd, 2017-07-07:
I looked into modifying the `rosdep` keys so we can install `pyassimp` from `pip2` and use `apt` to install `libassimp-dev`. However, `rosdep` is not configured to accept a particular version of pip.

@jlack1987 has opened an issue here, [(rosdep/issues/519)](https://github.com/ros-infrastructure/rosdep/issues/519), which is technically a blocker for this to get done.

jlack1987, 2017-07-08:
I just tested in a pristine docker installing pyassimp via `pip install pyassimp` and then `apt install libassimp-dev` and it worked. That should be all that's needed for it to work with `rosdep` right? I didn't have to specify `pip2` or `pip3`

0ldham, 2017-08-01:
I still receive the initial structs error, even after following the advice of @MarkusEich and of @iglstone, and after updating moveit and pyassimp.

k-okada, 2017-08-09:
as @0ldham reported, we still have this problem, on proposal patch is https://github.com/ros-planning/moveit/pull/581

jacknlliu, 2017-11-06:
We still have this problem.
```
Ubuntu: 16.04
ROS: kinetic
moviet: 0.9.9-Alpha
```

v4hn, 2017-11-17:
> We still have this problem.

The problem is worked around in f00addf and things work if you build moveit/the `moveit_commander` package yourself.
However it is still not released in kinetic. We are on our way to release (hopefully) next week (see #686).

naveedhd, 2017-12-04:
Following the moveit_commander and felt into this. [Xenial, Kinetic]. Workaround:

1. removed installed `python-assimp (<=3.2)` by `sudo dpkg --remove --force-depends python-pyassimp` (as apt will remove dependencies also)
2. installed by pip `pyassimp` (3.3) using `sudo -H` flag (to replace the /usr one)

raequin, 2018-05-24:
@naveedhd prescribed

1. `sudo dpkg --remove --force-depends python-pyassimp`
2. sudo pip install pyassimp

That works for importing a mesh but now `sudo apt upgrade` yields

> The following packages have unmet dependencies: ros-kinetic-moveit-commander : Depends: python-pyassimp but it is not installed

> E: Unmet dependencies. Try using -f.

Please tell me what I should do now.

Aparun8, 2019-03-24:
**Segmentation fault(Core dumped)** while adding mesh files to moveit. I updated the pyassimp version to 4.1.3 after some advice, but still I'm getting the same error. Importing moveit_commander works. Adding default shapes to moveit works, but adding meshfiles doesn't works. And the error comes when I add mesh files. I tried with both absolute and relative paths. Any help would be appreciated.

liamstal, 2019-04-02:
pip**3** install --user pyassimp
worked for me.

tandukion, 2019-04-08:
Had this problem before:
**Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)**
And it seems it was segmentation fault like @Aparun8 said that is caused by pyassimp error.
Found a solution [here](https://answers.ros.org/question/316515/cannot-add-mesh-files-to-moveit/). 
Only version 4.1.3 or 3.3 works.

dhiegomaga, 2020-08-07:
> sudo pip uninstall pyassimp
> and then,
> sudo pip install pyassimp ,
> which will install pyassimp 3.3, solved the problems..may help.

Nope, it only worked for you because the current library version of `pyassimp` on pip when you installed it _HAPPENED_ to be 3.3. Today the current one is 4.1.4, and I get the same error using your same commands. This is not a solution, it's mere luck. 

The way to do this is:
```bash
pip install 'pyassimp>=3.3.0,<3.4.0' --force-reinstall
```

If you also have problems with multiple python versions installed, this is an even more robust solution: 

```bash
sudo python -m pip install 'pyassimp>=3.3.0,<3.4.0' --force-reinstall
```