moveit/moveit #92: [kinetic] Missing dependency to moveit_msgs from move_group

jettan, 2016-08-20:
I'm getting errors trying to compile move_group in the kinetic branch because of some missing dependencies to moveit_msgs in the CMakeLists.txt.

Error:
```In file included from /home/jettan/workspace/ros/src/overlay/moveit/moveit_ros/move_group/src/default_capabilities/apply_planning_scene_service_capability.cpp:37:0:
/home/jettan/workspace/ros/src/overlay/moveit/moveit_ros/move_group/src/default_capabilities/apply_planning_scene_service_capability.h:41:44: fatal error: moveit_msgs/ApplyPlanningScene.h: No such file or directory
 #include <moveit_msgs/ApplyPlanningScene.h>
                                            ^
compilation terminated.
In file included from /home/jettan/workspace/ros/src/overlay/moveit/moveit_ros/move_group/src/default_capabilities/query_planners_service_capability.cpp:37:0:
/home/jettan/workspace/ros/src/overlay/moveit/moveit_ros/move_group/src/default_capabilities/query_planners_service_capability.h:42:42: fatal error: moveit_msgs/GetPlannerParams.h: No such file or directory
 #include <moveit_msgs/GetPlannerParams.h>
```

And there is a missing service in general too:

```In file included from /home/jettan/workspace/ros/src/overlay/moveit/moveit_ros/move_group/src/default_capabilities/query_planners_service_capability.cpp:37:0:
/home/jettan/workspace/ros/src/overlay/moveit/moveit_ros/move_group/src/default_capabilities/query_planners_service_capability.h:42:42: fatal error: moveit_msgs/GetPlannerParams.h: No such file or directory
 #include <moveit_msgs/GetPlannerParams.h>
                                          ^
```
The GetPlannerParams.srv is included in the CMakeLists of moveit_msgs, but the file itself is missing.

de-vri-es, 2016-08-20:
`move_group` depends on `moveit_core` which pulls in `moveit_msgs` [1]. Maybe there's a missing `add_dependencies` on the generated files?

`GetPlannerParams.srv` seems to be available since this commit [2]. 

Note that I did encounter similar issues, but they seemed related to catkin. After cleaning my workspace, it compiled without problems. Odd, of course, and possibly worth investigating. Weird thing is I was using `catkin_make_isolated`, meaning a possibly missing `add_dependencies` shouldn't have caused problems.

[1] https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/CMakeLists.txt#L109
[2] https://github.com/ros-planning/moveit_msgs/commit/16d59880ee1d9559f938d8db8004f8f10253ac30

jettan, 2016-08-20:
It worked! I did add the add_dependencies thing while also cleaning my workspace so idk what exactly fixed it but it worked.

closed 2016-08-20 by jettan