moveit/moveit #95: cleanup move_group namespace

rhaschke, 2016-08-20:
The `move` action server's (providing planning and execution capabilities) and the `dynamic reconfigure`'s topics of the `move_group` node, reside in the same namespace, making it hard to distinguish them (see list below). I suggest to rename the `move` action from `move_group` to `move` or `plan`.
Furthermore, I think it might be useful to group all `move_group`-provided stuff into its own namespace. However, this would break existing systems...

```
/move_group/cancel
/move_group/display_contacts
/move_group/display_cost_sources
/move_group/display_grasp_markers
/move_group/display_planned_path
/move_group/fake_controller_joint_states
/move_group/feedback
/move_group/goal
/move_group/monitored_planning_scene
/move_group/motion_plan_request
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
```

v4hn, 2016-08-20:
On Sat, Aug 20, 2016 at 09:29:16AM -0700, Robert Haschke wrote:
> The `move` action server's (providing planning and execution capabilities) and the `dynamic reconfigure`'s topics of the `move_group` node, reside in the same namespace, making it hard to distinguish them (see list below). I suggest to rename the `move` action from `move_group` to `move` or `plan`.

+1 The current situation is totally confusing. Renaming the node (and thus the private topics) is out of question in my opinion, thus the only option is to rename the action.
However, this would likely break **a lot**, so if we are going to do this, a majority of maintainers should agree...
The next question is whether  we want this in kinetic or only starting with lunar.

> Furthermore, I think it might be useful to group all `move_group`-provided stuff into its own namespace. However, this would break existing systems...

Afaics, This is more or less the case right now (or should be anyway, there might be bugs around...):
It's all in the same scope. If the user wishes to have it in a namespace, ti can add a `<group ns="mynamespace">` to the launch file. :)

JHoef, 2016-08-22:
Hi,

I plan to take care of this issue tomorrow.

v4hn, 2016-08-22:
@JHoef This is not actually a "simple improvement" tagged issue,
because we are not yet sure about the "when" and "how".

Still, if you like, you can provide a patch against kinetic-devel that
renames the action throughout moveit to something like `request_motion_plan`.
I prefer this generic name to "move" or "plan" because the action *can* but does not *have to* "move" the robot.

A working patch would demonstrate the scope of the change and is definitely welcome.

JHoef, 2016-08-23:
Okay I see that I have underestimated the issue. Maybe it would be better if I try to contribute to another issue.
Thanks.