moveit/moveit #98: [kinetic] Undefined reference to setCollisionObjectUpdateCallback

jettan, 2016-08-21:
I'm getting another nasty undefined reference error in moveit kinetic while trying to compile something that used to work in indigo using moveit.

```
/home/jettan/workspace/ros/devel/lib/libmoveit_planning_scene_monitor.so: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (boost::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
```

I already tried to rebuild moveit_ros_planning_interface over and over again.

de-vri-es, 2016-08-21:
What exactly are you building? I happen to know my system closely matches yours and I can compile the entire `moveit` repository with `catkin_make_isolated` without problem (or at least those parts that are picked up by `catkin_make_isolated`).

jettan, 2016-08-21:
It's not the moveit repository that crashes, but building something that is linked against moveit.
Do you think gcc-6.1 is causing this problem? Or perhaps newer boost libraries? 

The weird thing is that the moveit repository builds completely fine.

v4hn, 2016-08-22:
On Sun, Aug 21, 2016 at 11:55:35AM -0700, Jethro Tan wrote:
> It's not the moveit repository that crashes, but building something that is linked against moveit.

Most importantly: Does the same code still compile with current `indigo-devel`?
This might be a regression introduced in https://github.com/ros-planning/moveit/pull/70 .
At the moment I don't know if this is intentional, but as we also merged this into indigo,
we might have the problem there too. And we *shouldn't* have it in indigo.

@de-vri-es could you have a look at whether this is related?

jettan, 2016-08-22:
I can't test with current indigo-devel atm since I don't have it setup.

v4hn, 2016-08-22:
On Mon, Aug 22, 2016 at 08:09:32AM -0700, Jethro Tan wrote:
> I can't test with current indigo-devel atm since I don't have it setup.

Could you try to give us a minimal code-example where this happens?
I could test it then.

jettan, 2016-08-22:
It's taking me quite some time to trigger this bug in a minimal code example. I built the examples of moveit (the ones with planning scene and planning ros api tutorial) with success. So I'm trying to figure out what exactly triggers this behaviour.

jettan, 2016-08-22:
But it definitely has something to do with the changes introduced in #70 as you mentioned and as @de-vri-es mentioned.

https://github.com/ros-planning/moveit/pull/70/files#r75216429

de-vri-es, 2016-08-22:
The only way I can see that causing problems is if you're compiling with the new header and linking with an object compiled before the change. Do you have pre-compiled moveit packages installed?

The declaration and definition of 'planning_scene::PlanningScene::setCollisionObjectUpdateCallback' are consistent at any rate.

de-vri-es, 2016-08-22:
If you don't have precompiled moveit packages installed or if removing them (and cleaning your workspace for good measure) does not fix the issue, can you show the full linker command and output?

v4hn, 2016-08-22:
On Mon, Aug 22, 2016 at 11:12:29AM -0700, Maarten de Vries wrote:
> Do you have pre-compiled moveit packages installed?

Might also be other packages that were/are linked to moveit, but to an older version.

> The only way I can see that causing problems is if you're compiling with the new header and linking with an object compiled before the change.

This is called ABI breakage. Do we have this same problem in our indigo-devel branch, when compared to the latest release?
If we have, we should revert the problematic part (at least for now).

> The declaration and definition of 'planning_scene::PlanningScene::setCollisionObjectUpdateCallback' are consistent at any rate.

For stable releases, it is not about *consistency*, but about *change*.
We shouldn't change things (ABI,API) in indigo, even though we can always *extend* them.

de-vri-es, 2016-08-22:
> Might also be other packages that were/are linked to moveit, but to an older version.

Could be, although in that case removing the binary package for moveit-* should also remove those (assuming they were installed with a package manager). 

>This is called ABI breakage. Do we have this same problem in our indigo-devel branch, when compared to the latest release? If we have, we should revert the problematic part (at least for now).

Well, the correction of `collision_detection::ObjectConstPtr` definitely changed the API at a number of places. So for indigo that should be reverted. I'll make a PR for that. What about Jade?

> For stable releases, it is not about *consistency*, but about *change*.
> We shouldn't change things (ABI,API) in indigo, even though we can always *extend* them.

The point was, if they are inconsistent then that could explain the linking problems on even on a fresh compilation. Though in the case of member functions the compiler should already catch the mismatch.

jettan, 2016-08-22:
Got it compiling! It is indeed the ABI breakage. After uninstalling the prepackaged ros-kinetic-moveit_core and ros-kinetic-moveit_msgs and recompiling my moveit  overlay repository everything compiles as it should.

de-vri-es, 2016-08-22:
Quick note: those packages are coming from a personal Arch Linux repository I'm maintaining, in case anyone is wondering how there are binary packages before the actual release =]

v4hn, 2016-08-22:
On Mon, Aug 22, 2016 at 11:42:39AM -0700, Maarten de Vries wrote:
> > Might also be other packages that were/are linked to moveit, but to an older version.
> 
> Could be, although in that case removing the binary package for moveit-* should also remove those (assuming they were installed with a package manager). 

Yes.

> Well, the correction of `collision_detection::ObjectConstPtr` definitely changed the API at a number of places. So for indigo that should be reverted. I'll make a PR for that.

I wasn't sure back then whether *adding* const-ness is an ABI problem.
Turns out it is... I should have known better.
Could you please provide a minimal-invasive patch to revert to the old A{P/B}I in indigo, but keep the other cleanup changes you did?

> What about Jade?

During one of the maintainer meetings, we more or less agreed that jade is testing ground and does not have to be kept ABI-stable.
The changes there are already released, so there is no sense in reverting them.

> The point was, if they are inconsistent then that could explain the linking problems on even on a fresh compilation.

Agreed.

v4hn, 2016-08-22:
On Mon, Aug 22, 2016 at 11:52:07AM -0700, Maarten de Vries wrote:
> Quick note: those packages are coming from a personal Arch Linux repository I'm maintaining, in case anyone is wondering how there are binary packages before the actual release =]

Awsome! Great to hear something like that exists!
Be aware though that quite a number of things might change before the release...

jettan, 2016-08-22:
No worries, it's on a private buildbot.
And I was just getting ready to test things so I could one day mirror it.

de-vri-es, 2016-08-22:
> I wasn't sure back then whether *adding* const-ness is an ABI problem. Turns out it is... I should have known better. Could you please provide a minimal-invasive patch to revert to the old A{P/B}I in indigo, but keep the other cleanup changes you did?

I was actually wondering the same thing. Turns out that name mangling makes sure it is a problem :) Just as well I suppose, as otherwise you might end up with a program calling non-const members on what you thought was a const object.

The PR contains only a pretty minimal patch, see #104.

> Awsome! Great to hear something like that exists! Be aware though that quite a number of things might change before the release...

Aye, but I'm running Arch, so I'm used to having to recompile half the repo because of ABI breakage (yay boost updates) ;)

The binary repository is mainly meant to avoid long compilation times when installing new systems, we're not expecting it to remain stable for now.

v4hn, 2016-08-22:
> Just as well I suppose, as otherwise you might end up with a program calling non-const members on what you thought was a const object.

The standard allows for that (hello `const_cast`), so why should the compiler care. ;)
But yeah, it's good that it does...

> Aye, but I'm running Arch, so I'm used to having to recompile half the repo because of ABI breakage (yay boost updates) ;)

Tell me about it: https://github.com/lunar-linux 
One of my favorites til today are gcc ABI changes: https://github.com/lunar-linux/moonbase-core/blob/master/compilers/gcc/POST_INSTALL#L10

This issue is resolved, so I'll close it.
Thanks for your support!

closed 2016-08-22 by v4hn