moveit/moveit #101: Revive Search Based Planning Library

davetcoleman, 2016-08-22:
MoveIt! used to support the SBPL library but [that plugin](https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_planners/sbpl) was never catkinized. We would like to cleanup and get this plugin working again.

@aurone, @bcohen, and others at the Search-Based Planning Lab at CMU currently have had a working plugin, but it needs to be brought back into MoveIt!:

- https://github.com/aurone/sbpl_manipulation
- https://github.com/aurone/moveit_planners_sbpl

They are also working on a "plugin alternative to FCL that uses the bounding spheres representation that we use in much of our arm planning work" 

Some difficulties we might face in fixing the SBPL library up, according to @aurone:

> At the high level, there would be some difficulty in supporting the more general class of possible planning joint groups. A lot of that difficulty lies in creating infrastructure to specify the state/action space of the robot for any joint type; the current specification really only makes sense if all the planning joints are single-dof joints. I think in OMPL terms this is equivalent to having the different state spaces and sampling routines that make sense for each space.
>
> At the mid level, there are a reasonable amount of robot-specific features that greatly improve the efficiency of the sbpl planners. For instance, we often specifically configure the state lattice to have special actions, such as actions based on IK variants. I think all of these can be disabled through config, but often the planner just won't work well if the given planning joint group doesn't support them. I don't expect there to be special support at the MoveIt! level, so maybe it's just a matter of documentation, configuration, and some glue code between the MoveIt! robot model and the corresponding API in the core planning stack. 
>
> At the lower level, there's a few features that I just haven't had the motivation/time to implement, like supporting joint space goals and path constraints.

davetcoleman, 2016-11-21:
@aurone do you have any cycles right now to help with reviving SBPL from your repos?

aurone, 2016-11-22:
In general, what's the plan going forward with this? At the time, I can probably support minor changes and bug fixes as they arise, in my current repository, but I don't have much time for big integration/release endeavors.

dvanthienen, 2017-04-27:
Any news on this issue?

davetcoleman, 2017-10-18:
I recently talked with @aurone about reviving SBPL and have submitted some codesprint proposals for this, but they are still in the works.

bmagyar, 2017-10-18:
Happy to review!