moveit/moveit #102: ROS Lint Packages

davetcoleman, 2016-08-22:
Cleanup the code base by using [roslint](http://wiki.ros.org/roslint):

> CMake lint commands for ROS packages. The lint commands perform static checking of Python or C++ source code for errors and standards compliance.

Start with just one package (moveit_core?) so that you can get feedback from the maintainers and try to keep each PR small.

v4hn, 2016-08-22:
duplicate of #34

davetcoleman, 2016-08-22:
``catkin lint`` and ``ros lint`` are two separate things! AFAIK the former only helps cleanup your CMake files, while the latter cleans up your code

gavanderhoorn, 2016-08-22:
@davetcoleman: yep. `catkin_lint` 'only' checks `package.xml` and `CMakeLists.txt`.

v4hn, 2016-08-22:
Ah, right, sorry I missed that. Haven't used roslint in quite a while, so I was primed to `catkin_lint`..

Sounds like a lot of work, that would improve MoveIt!

davetcoleman, 2019-06-24:
Closing this in light of moveit2 development.

closed 2019-06-24 by davetcoleman

v4hn, 2019-06-25:
I don't feel strongly about these issues, but I don't see the connection to MoveIt ROS2 development.

davetcoleman, 2019-06-26:
The more optional "cleanup" type tasks we do in moveit1, the harder it will be to sync forward to moveit2.

v4hn, 2019-06-27:
I do see your line of argument, but I am not convinced this is sufficient
to argue against cleanup patches.

Let's briefly address this in the maintainer meeting later today, if time allows.