moveit/moveit #99: Integrate moveit_python into main MoveIt! project

davetcoleman, 2016-08-22:
I just discovered "Pure Python Bindings to ROS MoveIt!" and am interested in at minimum adding documentation for this package of @mikeferguson or perhaps even adding it to the main repo after further discussion. 

@mikeferguson how much better is this than what we currently have?

v4hn, 2016-08-22:
(1.) What is it?

(2.) Where is it?

davetcoleman, 2016-08-22:
Ooops, forgot to add the link - fixed.

> Is the documentation package you promised for tomorrow ready yet? :-)

There have been a ton of Github emails I've been responding to this morning and last night, thank you!

v4hn, 2016-08-22:
On Mon, Aug 22, 2016 at 11:20:04AM -0700, Dave Coleman wrote:
> There have been a ton of Github emails I've been responding to this morning and last night, thank you!

I know, it's quite a ridiculous amout today, right? :-)
Don't blame me, I'm just responding too...
Thanks for the link. Definitely looks interesting!

andreaskoepf, 2017-10-17:
The Xamla team would like to work on this during WMD.

v4hn, 2017-10-17:
Great, I'm looking forward into see progress here!

@davetcoleman @mikeferguson @andreaskoepf : Should we add these packages (if useful) as part of `moveit_commander` or do you consider it justified to create a new top-level folder here?

Personally, I would leave it to Xamla, as long as there is a pull-request to merge it into the MoveIt repo by the end of the day. :-)

Other things to do w.r.t. this package

- It looks like Mike implemented only what he needed at that time, but I believe many functions available through `moveit_commander`, the `MoveGroupInterface`, and the `PlanningSceneInterface` class should also be supported here.
It would be nice if you could look through the available interfaces and come up with a list of things that are missing. (and maybe implement some of them)

- It looks like https://github.com/mikeferguson/moveit_python/blob/master/src/moveit_python/planning_scene_interface.py was implemented before the `apply_planning_scene` service became available. I believe most of the code there should be rewritten to make use of it. This should simplify the logic in there significantly.

mikeferguson, 2017-10-17:
@davetcoleman -- I'm not sure that it's necessarily "better" but it is "different". For me, the pure python mattered -- I wanted to avoid the wonkiness of moveit_commander having roscpp underpinnings. Obviously it doesn't expose every interface, so it's not as full featured as the commander.

@v4hn -- I'm not sure that integrating it into moveit_commander is the right approach, given that the commander is a different approach with it's C++ underpinnings. Most of the code is from late 2013 / early 2014, so I think it does predate the planning scene service. That would be a good update.

My one request here would be: let's avoid breaking too much stuff. There are a number of people who are using moveit_python (not all of whom have released their code to the rosdistro, etc), because of it's simpleness.

Also, if you're going to change the interfaces significantly in a not so backwards compatible way, I would prefer creating an entirely new package (with a new package name, since ROS has a flat namespace) so that the choice/timing of migration is left to end users.

_(side note: I know there is a desire to have ALL the moveit code in one place, but having this in a separate repo is somewhat nice, because it's pure python and only depends on messages, so it has a near-zero build time compared to the rest of MoveIt)_

davetcoleman, 2017-10-18:
I am not a python expert but from user feedback and bug reports it would seem ``moveit_commander`` leaves a lot to be desired that could be supplemented with ``moveit_python``. 

My vote (as almost always) would be to move it into the main repo for increased visibility and maintenance. Just use ``catkin build --this --no-deps`` (``catkin bot``) if you want fast compile time for a single package.

Most importantly though I'd add it to the tutorials so that it can be discovered easily.

v4hn, 2017-10-18:
Ok, so altogether I propose

- add this to the main moveit repo

- retain the interfaces as much as possible

- adapt the code for the `apply` service

- add new interfaces where useful

alemme, 2017-10-18:
Okay. I adapt the code for the apply service (https://github.com/alemme/moveit_python/commit/0aa2de1baf6c5b681e9594cd761fb81c2af13a23) and tested it in simple cases like adding and removing boxes. However, I noticed that even if I use the service call the planning scene is not directly updated for the client. As a simple check I used the waitForSync function which was there before I made my changes.

You can reproduce this behavior by using [add_box.py](https://github.com/alemme/moveit_python/blob/0aa2de1baf6c5b681e9594cd761fb81c2af13a23/scripts/add_box.py) and [remove_objects.py](https://github.com/alemme/moveit_python/blob/0aa2de1baf6c5b681e9594cd761fb81c2af13a23/scripts/remove_objects.py). 

I am not sure if I did something wrong since I am not a python pro and all but my guess is that the service responds to early.

v4hn, 2017-10-18:
> However, I noticed that even if I use the service call the planning scene is not directly updated for the client

Of course. The service only guarantees that the *move_group*'s planning scene is updated.
I don't know whether we really want to wait for the next sync. Are there good arguments to do that?

Can you add a pull-request for the moveit repo?

alemme, 2017-10-18:
> Are there good arguments to do that?

In case you have a node which is adding CollisionObjects to the scene and triggers another node which should perform a trajectory planning task, it needs to be guaranteed that it knows about the new collision objects.

> Can you add a pull-request for the moveit repo?

sure will do tomorrow.

v4hn, 2017-10-18:
> In case you have a node which is adding CollisionObjects
> to the scene and triggers another node which should perform
> a trajectory planning task, it needs to be guaranteed
> that it knows about the new collision objects.

In the best-case scenario the other node plans the trajectory *via* `move_group`.
The is the case the service guarantees to work.

But in the case (I presume) you describe,
monitoring the `monitored_planning_scene` topic of `move_group`
will not guarantee a successful sync either.
You would have to re-publish and monitor the scene *your planning node* receives then.

This should definitely be possible, but I'd say it shouldn't be default behavior to
wait for republished scenes as verification.

> sure will do tomorrow.

Great! looking forward to have this packaged with moveit soon!

andreaskoepf, 2017-10-18:
> In the best-case scenario the other node plans the trajectory *via* `move_group`. The is the case the service guarantees to work.

Our use case might not be the most common. Just to give you a glimpse:  We have a custom script based planning lib running on top of MoveIt! .. in the node running this lib we instantiate beside other things a `PlanningScene` object for collision checks (e.g. `isStateColliding()`). With luajit this basically allows us to have a very fast but fully scripted way-planning system including collision checking (with minimal turn-around times from source change to test in less than 2s). But we of course are aware that there is no way around Python in the ROS ecosystem and we will want to offer strong support for Python Rosvita too. Therefore we are think about working towards a powerful scripting system for MoveIt in Python too. But since rospy is no wrapper around roscpp I am not really sure if it is possible to create something as powerful and fast as our current luajit-based impl. We might go for a hack as the one used in `MoveItCommander` which calls additionally to the pure python rospy init also `ros::init()` of `roscpp`  -then we could add a Python wrapper to the internal classes of moveit as we did for lua.

After what we analyzed today I guess it would be best to publish this as a separate package first and get some feedback from users.

I personaly will have to gather more experience with rospy (e.g. stuff like the msg numpyization process etcs.) before I can come up with a high-quality Python architecture.

v4hn, 2017-11-17:
We still need a pull-request that adds `moveit_python` to this repository. Anyone is welcome to contribute it