moveit/moveit #110: Update pr2 tutorials to work with Jade and Kinetic

davetcoleman, 2016-08-23:
Our current [tutorials](http://docs.ros.org/indigo/api/moveit_tutorials/html/) only work with Indigo. We would like to have them tested and fixed for Jade and possibly with Kinetic - though this would require building the PR2 dependencies from source. For now a note has been added to the tutorials page.

Random-Word, 2016-11-11:
I'd love to be able to compile these for Kinetic. Is there any documentation anywhere on how to do this? The pr2_simulator package's indigo-devel branch doesn't seem to be able to build under kinetic, and master is five years old.

davetcoleman, 2016-11-13:
I've updated the MoveGroup Interface package to build with Kinetic but its still under review: https://github.com/ros-planning/moveit_tutorials/pull/45

My strategy there was to remove the dependency on moveit_pr2 because maintenance on it has fallen behind. You're welcome to use my replacement package [pr2_moveit_config](https://github.com/davetcoleman/pr2_moveit_config) to fixup the other tutorials via pull requests. Thanks!

iretiayo, 2017-02-06:
@Random-Word @davetcoleman 
Are you able to successfully run pr2 on ros kinetic? I've tried installing pr2-related packages from source but I keep running into different issues.
Kindly advise.

v4hn, 2017-02-06:
@iretiayo What do you mean by "run pr2 on ros kinetic"?
I do not suppose you installed Ubuntu 16.04 on a PR2 and try to get it to run with MoveIt?

What are you doing and which errors do you encounter?

The hardest part I am aware of was to make pr2_arm_kinematics compile/work with MoveIt kinetic.
I just did the appropriate changes and pushed them to my fork [here](https://github.com/v4hn/pr2_kinematics/tree/kinetic-devel) for now.

With that I have the moveit `move_group_interface` tutorial up and running with MoveIt kinetic on ROS indigo (that's just my custom setup over here...)

iretiayo, 2017-02-06:
The pr2_simulator package (and most other PR2 pkgs) haven't been released for Kinetic. So I had to manually download the packages from the PR2 github [page](https://github.com/PR2). However, when I try to compile the codes, I get errors like below.
When I saw the title of this issue, I thought someone might have a working pr2-simulator on ros-kinetic and wanted to find out how they set it up.

catkin_make error:
```
/opt/ros/kinetic/include/hardware_interface/internal/interface_manager.h:194:21: error: no matching function for call to ‘pr2_mechanism_model::RobotState::RobotState()’
         iface_combo = new T;
```

v4hn, 2017-02-06:
> The pr2_simulator package (and most other PR2 pkgs) haven't been released for Kinetic. So I had to manually download the packages from the PR2 github [page](https://github.com/PR2).

Yes, this is the case. PR2 support has fallen behind a bit..
Quite a bit can still be compiled right away on kinetic, but you will hit some API changes..

However, this issue is about moveit_tutorials only. The tutorials rely on `pr2_moveit` and specifically `pr2_moveit_config`'s `demo.launch`, but not on the gazebo simulation.

nmalo5, 2017-03-11:
Hi!, 
 I´m a university student trying to make progress with MoveIt in ROS Kinetic and Ubuntu 16.04.
The main goal of my project is to do a Pick and Place movement in the simulator.
I found some tutorials about a PR2 arm and MoveIt (http://docs.ros.org/kinetic/api/moveit_tutorials/html/), but I found problems because some necessary packages in this tutorial makes reference to ROS Indigo instead of Kinetic.
Then, searching for similar issues, I ended up here. It seems that PR2 support isn´t available for ROS Kinetic yet... Is there any way to run the PR2 MoveIt tutorials in ROS Kinetic? Could I achieve this goal(Pick&Place in robotic arm) easily with other available tutorials?

davetcoleman, 2017-03-12:
A while back I fixed the [MoveGroupInterface tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html) to run on Kinetic with the PR2's URDFs. I achieved this by skipping all of PR2's custom plugins and instead using the default KDL IK solver, etc. This "slimmed down" version of pr2_moveit_config is [here](git clone https://github.com/davetcoleman/pr2_moveit_config.git)

This has nothing to do with actual hardware, however.

nsnitish, 2017-06-11:
Hi @davetcoleman ! I am trying to follow the [MoveGroupINterface tutorial on ROS-Kinetic](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html). 
While running roslaunch pr2_moveit_config demo.launch and roslaunch moveit_tutorials move_group_interface_tutorial.launch, I am getting the following error: 
/pr2_moveit_tutorial/catkin_ws/devel/lib/moveit_tutorials/move_group_interface_tutorial: error while loading shared libraries: libmoveit_planning_scene_interface.so.0.9.6: cannot open shared object file: No such file or directory. 

I am not able to see any related documentation. Can you please help?

davetcoleman, 2017-06-21:
That issue is off-topic for this discussion thread - next time please create a new issue. 

> libmoveit_planning_scene_interface.so.0.9.6: cannot open shared object file

That is a problem with your binary/debian installation - something is out of sync with your computer's installed packages.

janisa9, 2017-06-27:
Hi, same problem here, 

error while loading shared libraries: libmoveit_planning_scene_interface.so.0.9.6: cannot open shared object file: No such file or directory

Did little digging, and yeah there is  **libmoveit_planning_scene_interface.so.0.9.7** not **libmoveit_planning_scene_interface.so.0.9.6** in  **opt/ros/kinetic/lib** .   Is there any solutions to downgrade libraries or use this libmoveit_planning_scene_interface.so.0.9.7 instead of this libmoveit_planning_scene_interface.so.0.9.6 library ??

maxgitt, 2017-07-28:
Hi @davetcoleman I am completely new to moveit and am running through the "MoveIt! RViz Plugin Tutorial" for kinetic. What confuses me is the number of indigo references on this page.

Under pre-requisites the first thing to check is that I have the "pr2_moveit_config" which I can either find in the "MoveIt! Setup Assistant tutorial" or by typing in on my commandline:
<pre>sudo apt-get install ros-<b>indigo</b>-moveit-pr2</pre> 

I tried:
<pre>sudo apt-get install ros-<b>kinetic</b>-moveit-pr2</pre> but it doesn't seem to exist. So I tried the other option which was to go to the "MoveIt! Setup Assistant tutorial". This page is also listed as a kinetic page and after launching the moveit_setup_assistant, the first step is to load the xacro file found here: <pre>/opt/ros/<b>indigo</b>/share/pr2_description/robots/pr2.urdf.xacro</pre>

Because I am working with a kinetic distribution I of course don't have this file, but upon searching for the equivalent <pre>/opt/ros/<b>kinetic</b>/share/</pre> directory I could not find the `pr2_description` sub-directory. I found your comments explaining that I can find the slimmed down version of the `pr2_moveit_config` [here](https://github.com/davetcoleman/pr2_moveit_config) but don't find a xacro file within that repo either.

I am not sure what I've missed here and would appreciate your help loading the xacro file for kinetic.

davetcoleman, 2017-07-30:
First - yes, those tutorials sorely need to be updated for Kinetic. Just needs someone to put in the effort.

> Because I am working with a kinetic distribution I of course don't have this file, but upon searching for the equivalent

The xacro file is not in the ``moveit_pr2`` package but actually the [pr2_common](https://github.com/PR2/pr2_common) pkg that is available in Kinetic. 

> I can find the slimmed down version of the pr2_moveit_config here but don't find a xacro file within that repo either.

You need both the slimmed down ``pr2_moveit_config`` and ``pr2_common`` to run the tutorial in Kinetic.

maxgittelman, 2017-07-31:
I made the necessary changes for a more complete kinetic tutorial for the Setup Assistant. Which includes the installation of the pr2_common pkg. I would like to push the changes to the repo but am getting an access denied.

I tried creating a pull request but can't create my own branch so I'm unsure how to commit these changed. Please excuse my confusion I've never committed to an open project like this.

maxgittelman, 2017-07-31:
I created a pull request. The tutorial now includes the missing pr2_common pkg and some finer changes to make the kinetic tutorial more clear.

maxgittelman, 2017-08-01:
I also made a pull request for the RViz plugin tutorial. Hopefully this gets looked at and merged soon.

davetcoleman, 2017-08-02:
keep it coming!

davetcoleman, 2018-06-28:
Closing this issue because we have switched to the Franka Emika robot, away from the PR2, for all kinetic tutorials and newer:
https://github.com/ros-planning/moveit_tutorials/pull/166

closed 2018-06-28 by davetcoleman