moveit/moveit #111: Test Tutorials and add more visuals / screenshots

davetcoleman, 2016-08-23:
The consolidated the [moveit_tutorials](https://github.com/ros-planning/moveit_tutorials) have many pages without any images - currently only the Rviz Plugin tutorial and Setup Assistant really have visuals. It would be great if someone could run through the tutorials and take more screen shots of progress for tutorials that have no images.

uwesterh, 2016-08-23:
I've already found some broken links in the tutorial pages. So I pick up this issue and will provide some screen shots for the tutorials in addition to the link fixes.

uwesterh, 2016-08-23:
Most of the links are fixed now. For details, see my pull request at https://github.com/ros-planning/moveit_tutorials/pull/1

davetcoleman, 2016-08-23:
Great! Did you find any good use cases for additional images?

v4hn, 2016-08-26:
@uwesterh you showed me some pictures you embedded into the docs on Tuesday.
Why didn't you file a request to add them?

v4hn, 2016-09-24:
The issue as posted for WMD is now deprecated. Closing.
testing tutorials is *always* welcome even without an issue.

closed 2016-09-24 by v4hn