moveit/moveit #112: Expand the controller tutorial to include ros_control and fake_controllers

davetcoleman, 2016-08-23:
The current tutorial on using [controllers with moveit](http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/controller_configuration.html) does not mention how to integrate with ros_control or the fake controllers. We need to consolidate the READMEs from:

https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_plugins/moveit_ros_control_interface
https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_plugins/moveit_fake_controller_manager

And update the README.mds to instead point at the moveit_tutorials

leander-dsouza, 2022-02-18:
You can control the namespace of `moveit_ros_control_interface` through the `ros_control_namespace` parameter.

rickstaa, 2022-02-18:
@leander-dsouza Your right, that is another way to make MoveIt aware of the controller manager. Could also be added to the documentation.

leander-dsouza, 2022-02-18:
@rhaschke @rickstaa 
Why doesn't the `moveit_ros_control_interface` support the `GripperCommand` action?

01binary, 2024-02-08:
> This is correct. Note that the ros_control MoveIt controller manager is not intended to support the GripperCommand action.

@rhaschke would this be something you would like a PR for, or is GripperCommand controller meant to be a testing utility that should not be used on a real robot/in production?

rhaschke, 2024-02-08:
@01binary, the GripperCommand controller is only supported by MoveIt's SimpleControllerManager, not the ROSControlControllerManager. It is definitely intended for production use. You are welcome to extend the documentation.

01binary, 2024-02-08:
@rhaschke could you help me unravel the confusion around gripper action controller?

New MoveIt project created by MSA is setup as follows:

- `demo.launch` (Launch in RViz visualization only): uses *Fake Controller Manager*
- `demo_gazebo.launch` (Launch Gazebo simulation and RViz visualization): uses *ROS Control Controller Manager*
- `gazebo.launch` (Launch Gazebo simulation only): uses *ROS Control Controller Manager*
- `move_group.launch` (No visualization, no simulation): uses *Simple Controller Manager* by default

If I am a new MoveIt user setting up MoveIt for my own robot, my entry point into the documentation is [Integration with a New Robot](https://ros-planning.github.io/moveit_tutorials/#integration-with-a-new-robot) section.

I read about MSA, then URDF/SRDF, and then I land on [Low-Level Controllers](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html) page.

This has me setup a joint trajectory controller for the arm, and a gripper controller for the gripper.

Next it tells me to launch with Simple MoveIt Controller Manager. This example cannot actually be followed because it ends with "Now, you should be ready to have MoveIt talk to your robot." and there are no instructions on how to use a custom launch file it has you create. Let's assume you somehow figure it out and get it running.

Then the tutorial tells you:

> For most use cases, the included [MoveItSimpleControllerManager](https://github.com/ros-planning/moveit/blob/master/moveit_plugins/moveit_simple_controller_manager) is sufficient if your robot controllers already provide ROS actions for FollowJointTrajectory. If you use ros_control, the included [MoveItRosControlInterface](https://github.com/ros-planning/moveit/blob/master/moveit_plugins/moveit_ros_control_interface) is also ideal.

If my robot works with Joint Trajectory Controller from ROS Controls I stop there.
If I have a custom ROS Controller then this is telling me to switch to ROS Control Controller Manager.

So I switch to ROS Control Controller Manager, and my gripper stops working. It has a Controller Handle, but no Controller Handle Allocator. At this point I am confused - why can't I use the gripper? As a new user, I did everything correctly. So then I have to  write and export a Controller Handle Allocator for it as part of my own project, or go back to using MoveIt Simple Controller Manager.

Does this flow make sense to you? Maybe you can spot where the disconnect happens?

rhaschke, 2024-02-09:
If you need to use GripperCommand you must use the `SimpleControllerManager`.
Any ROS controller supporting the `FollowTrajectory` action API, works with the `SimpleControllerManager` as well.
If you have a ROS controller, which does not support that action, you are lost anyway, because both `SimpleControllerManager` and `ROSControlControllerManager` require that action.
So, my suggestion is: Use `SimpleControllerManager` and augment your ros_control controller with a `FollowTrajectory` action if needed.

01binary, 2024-02-09:
If you were to describe why Gripper Action Controller exists, would you say it's for "simple usage scenarios"? As in, it lets you setup a new robot in the fastest and easiest way possible (and you would be using Simple Controller Manager).

People who use ROS Control Controller Manager are likely to have their own setup for grippers (or already use Joint Trajectory Controller), so it's just not the sort of audience who would care to use Gripper Action Controller?

rhaschke, 2024-02-10:
The `GripperCommand` action is fundamentally different from the `FollowJointTrajectory` action: While the latter allows precise following of a joint trajectory, the GripperCommand action just commands the gripper to close or open, but doesn't specify a precise trajectory to do so (but rather a maximal force).
So there are conceptual reason, why we have two different action types.

01binary, 2024-02-10:
I see what you mean

```
// goal to be sent
control_msgs::GripperCommandGoal goal;
goal.command.position = 0.0;

// send last point
int tpoint = trajectory.joint_trajectory.points.size() - 1;
ROS_DEBUG_NAMED("GripperController", "Sending command from trajectory point %d", tpoint);
```