moveit/moveit #113: Create TracIK tutorials in moveit_tutorials

davetcoleman, 2016-08-23:
We'd like to streamline the usage of TracIK in MoveIt by giving it its own tutorials in MoveIt!-format. A page has [already been created](http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/trac_ik_tutorial.html) and we just need to test and bring in [this page](https://bitbucket.org/traclabs/trac_ik/src/HEAD/trac_ik_kinematics_plugin/)

dbolkensteyn, 2016-08-23:
I'm taking care of this one

v4hn, 2017-10-08:
I believe it would be a reasonable idea to generalize this planned tutorial a bit.

I would propose to remove the current TracIK stub in the tutorials section.
Instead I would prefer to add additional sections for TracIK, [bio_ik](https://github.com/TAMS-Group/bio_ik), and @mamoll's [cached ik plugin](https://github.com/ros-planning/moveit/pull/612) to http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/kinematics/src/doc/kinematics_configuration.html and rename this tutorial to `Choose Kinematics Solver`.

davetcoleman, 2017-10-09:
+1 to consolidating it to one page