moveit/moveit #116: Missing implemenation of CollisionPlugin

alemme, 2016-08-23:
This plugin interface is defined in  [collision_plugin.h](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/collision_detection/include/moveit/collision_detection/collision_plugin.h) but is not implemented.

The fcl implementaiton of the collision detection is used currently in the `kinematic_contraint.cpp`
<PRE>
kinematic_constraints::VisibilityConstraint::VisibilityConstraint(const robot_model::RobotModelConstPtr &model) :
  KinematicConstraint(model), collision_robot_(new collision_detection::CollisionRobotFCL(model))
{
  type_ = VISIBILITY_CONSTRAINT;
}
</PRE>

davetcoleman, 2016-08-23:
FCL is a working implementation of a CollisionPlugin - not sure what the problem is here.

@mikeferguson created the collision plugin concept, I believe he's relevant to this discussion

v4hn, 2016-08-23:
Not sure why @alemme did not reply (we discussed that earlier today), so I'll quickly summarize.

On Tue, Aug 23, 2016 at 08:33:32AM -0700, Dave Coleman wrote:
> FCL is a working implementation of a CollisionPlugin

It looks like this might not be the case. The FCL code does not implement the class `collision_detection::CollisionPlugin` which is used with pluginlib to register the collision plugins.

mikeferguson, 2016-08-23:
This is definitely something I did not finish. Some references:

 * Original ticket about supporting other collision checkers: https://github.com/ros-planning/moveit_ros/issues/532
 * Merged PR for adding the plugin class: https://github.com/ros-planning/moveit_core/pull/214
 * Merged PR for actually loading plugins: https://github.com/ros-planning/moveit_ros/pull/536
 * Future work to actually add distance fields for collision checking (one of the things we want/need for things like SBPL)
 * My SBPL work that never really got finished: https://github.com/ros-planning/moveit_planners/issues/57

The collision plugin stuff was added at a much later date than the original planning scene. Effectively, the plugin allows you to "swap" out the collision checker with a new one -- hence the FCL is there by default in each planning scene, but you can load a new one, and then by setting "exclusive", it will unload FCL. If you do your collision check through the PlanningScene (as most, but not all things in MoveIt do), then you use the consistent collision checker.

I did have an implementation of a collision checker loader somewhere -- but I can't currently find it.

alemme, 2016-08-24:
@mikeferguson I found [collision_plugin_loader.h](https://github.com/ros-planning/moveit/blob/4ac0c7432d335f57aab6836cbcaaac3fccf4b6f9/moveit_ros/planning/collision_plugin_loader/include/moveit/collision_plugin_loader/collision_plugin_loader.h)
This seems to be the only file where the collision_plugin.h is included, correct?

v4hn, 2016-08-26:
@mikeferguson do we want to *change* the current behavior and make the FCL implementation a plugin too?

The documentation aspect of this issue can be discussed in https://github.com/ros-planning/moveit.ros.org/pull/48 .

If we do not want to change the current implementation, I suggest to close this ticket again.

mikeferguson, 2016-08-26:
I think ideally we would make FCL a plugin, however, I think there are too many places where things use FCL directly, instead of through the planning scene, and so we would end up with a huge number of changes to implement this (and if we don't do it right, we might end up violating ODR).

v4hn, 2016-08-26:
On Fri, Aug 26, 2016 at 11:05:45AM -0700, Michael Ferguson wrote:
> I think ideally we would make FCL a plugin, however, I think there are too many places where things use FCL directly,
> instead of through the planning scene, and so we would end up with a huge number of changes to implement this

goofy, really. So MoveIt is bound to FCL for now.
Do you think it would require a restructuring of the `CollisionPlugin` interface to even allow MoveIt to use FCL exclusively through a plugin API?
The current plugin API relies on the plugin setting the collision plugin in *the planning scene*.

Which other locations did you have in mind? `grep -r FCL` lists:

- `moveit_core/kinematic_constraints/src/kinematic_constraint.cpp`

- some stuff in `moveit_experimental/collision_distance_field`

- and the planning scene

mikeferguson, 2016-08-26:
The collision plugin interface should require no refactoring -- it actually doesn't (re)define the collision checker interface -- all the plugin does is know how to create an instance of the appropriate collision checker and add it to the planning scene.

I'm not sure all of the locations where there are direct access -- maybe it's less than I thought and this really wouldn't be a challenge, but I remember someone pointing out to me that there are places where code directly instantiates a CollisionWorldFCL/CollisionRobotFCL

v4hn, 2016-08-26:
On Fri, Aug 26, 2016 at 11:49:48AM -0700, Michael Ferguson wrote:
> The collision plugin interface should require no refactoring -- it actually doesn't (re)define the collision checker interface -- all the plugin does is know how to create an instance of the appropriate collision checker and add it to the planning scene.

I know, but it adds it to the planning scene only, so everything that (might) want to use collision checking would have to have access to the scene.
This is why I was asking.

> I'm not sure all of the locations where there are direct access -- maybe it's less than I thought and this really wouldn't be a challenge, but I remember someone pointing out to me that there are places where code directly instantiates a CollisionWorldFCL/CollisionRobotFCL

That would be `collision_distance_field` and `kinematic_constraints` then.

davetcoleman, 2016-08-26:
Doesn't seem worth it to me to change FCL to be a proper plugin until we have other common alternative collision checking plugins