moveit/moveit #117: Replace ompl_interface with moveit_ompl_planning_interface

mamoll, 2016-08-23:
@ryanluna has written an amazing replacement for ompl_interface. The repo is available here: https://github.com/KavrakiLab/moveit_ompl_planning_interface. He has moved on to Google X and might not have a whole lot of time to help in getting this done. The replacement code allows users to easily add new plugins by just editing a YAML file. This is why you should care:
- It allows for the easy development of new planners without being forced to recompile MoveIt! or OMPL from scratch.
- It is also useful for creating customized planning contexts / parameter settings for a particular robot / application.
- It is stable and usable right now. Several members of our group have been using it for years.

See README.txt in the repo for details. I am not as familiar with the low-level details, but am willing to help as much as I can.

mamoll, 2016-08-24:
Perhaps this is not completely obvious, but the moveit_ompl_planning_interface (or OMPL) does not require that a planning algorithm uses sampling, so in principle if someone wanted to create a plugin that implements, say, Multi-Heuristic A\*, then this would be much easier with this new plugin system.

v4hn, 2016-09-26:
@mamoll should we, in your opinion move the code in with MoveIt, move the repository to the ros-planning organization or officially rely on KavrakiLab?

The module looks intriguing, I fully agree.

mamoll, 2016-09-26:
I am all for integrating this with MoveIt! It'd be nice if Moveit! could be this magical black box that solves all motion planning problems optimally, but in reality people will have to tinker with control knobs or customize planners. What Ryan has written will make this so much easier. I am not core MoveIt! developer, but with some help from Ryan I had no trouble defining new planning contexts for Robonaut2 and customize some planners.

(For future reference, see also discussion on issue #197.)

davetcoleman, 2017-08-15:
+1 to moving towards this new interface