moveit/moveit #118: expose OMPL's generic path cost infrastructure to MoveIt!

mamoll, 2016-08-23:
There are many planning algorithms that can optimize a path cost with respect to some arbitrary cost function (RRT\*, CHOMP, TrajOpt, to name a few). OMPL has a generic path and state cost infrastructure and many pre-defined common optimization objectives (path length, clearance, mechanical work, etc.). It'd be great if some of that functionality could be added to MoveIt! with a direct mapping to corresponding to OMPL classes when possible.

davetcoleman, 2016-08-23:
+1

note that @rbbg recently started work in this direction with this merged PR: https://github.com/ros-planning/moveit_planners/pull/75

docs: http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/ompl_interface_tutorial.html

mamoll, 2016-08-23:
Cool! I wasn't aware of that PR. That looks like a good first step. In the long run, it'd be nice if there was something generic that wasn't specific to OMPL. It'd be nice if (after the benchmarking stuff of #25 is merged), one could compare the convergence rate of, say, RRT\* and Multi-Heuristic A\*.

scottpaulin, 2016-08-24:
What ways should a Moveit path cost infrastructure differ to the one in OMPL? Does anyone know of some good libraries (doesn't have to be motion planning) other than OMPL that implement a generic cost infrastructure which we could look at for inspiration?

mamoll, 2016-08-24:
The relationship between costs and states/paths is the tricky part. The cost infrastructure needs to be flexible enough to be able to express a wide variety of cost functions and allow users to easily define their own cost function. This is exactly what we did in OMPL (well, actually, @giogadi did almost all of the work). This doesn't mean that all optimizing planners can suddenly plan optimal plans w.r.t. arbitrary cost functions, but it helps expose some of the implicit assumptions made by some algorithms. (Optimality is often only proven w.r.t. path length.)