moveit/moveit #121: indigo-devel ABI incompabilities

jspricke, 2016-08-23:
I've checked the ABI of 214c3b8 against the last releases (0.7.2) of moveit_core and moveit_ros and found the following problems:
compat_reports/libmoveit_collision_detection_fcl.so/1_to_2/compat_report.html https://github.com/ros-planning/moveit_core/pull/295
compat_reports/libmoveit_motion_planning_rviz_plugin_core.so/1_to_2/compat_report.html https://github.com/ros-planning/moveit_ros/pull/713
compat_reports/libmoveit_move_group_default_capabilities.so/1_to_2/compat_report.html https://github.com/ros-planning/moveit_ros/pull/716
compat_reports/libmoveit_trajectory_execution_manager.so/1_to_2/compat_report.html https://github.com/ros-planning/moveit/pull/72
compat_reports/libmoveit_common_planning_interface_objects.so/1_to_2/compat_report.html
compat_reports/libmoveit_constraint_samplers.so/1_to_2/compat_report.html
compat_reports/libmoveit_move_group_pick_place_capability.so/1_to_2/compat_report.html
Full report: [compat_reports.tar.gz](https://github.com/ros-planning/moveit/files/432778/compat_reports.tar.gz)

davetcoleman, 2016-08-23:
we're wrapping our shared_ptrs in a pre-existing MACRO so that we can switch to ``std`` in kinetic. i think these changes are worth the abi incompatibility - i think we should just email moveit-users with a warning when it is released

de-vri-es, 2016-08-23:
I can't see the use of the macro introducing ABI incompatibilities though. The types are still the same, it's just that they are declared with a macro instead of manually. The one point where we did change a type has already been reverted. It looks like the ABI breakage comes from different places.

jspricke, 2016-08-23:
Most of them are because of added variables in classes, which changes their size.

davetcoleman, 2016-08-23:
i bet there are cases where previously there wasn't both a shared_ptr and a const shared_ptr, which is what we have added

de-vri-es, 2016-08-23:
True, but those types didn't exist before, so they can't be part of the API or ABI.

v4hn, 2016-08-25:
Thanks a lot for looking into this @jspricke !
This saved me a lot of work!

@davetcoleman @de-vri-es
This has nothing to do with the shared pointer conversions.

I just looked through them:

The only severe ones that could be problematic are:

@Levi-Armstrong's https://github.com/ros-planning/moveit_core/pull/295/files#diff-090bd42b3571985d04cec3f6ebf33a36R95
broke ABI when someone uses `CollisionRobotFCL` from his own code. I don't consider this to be a public interface, so I don't think
it's critical. People are supposed to use the FCL agnostic interface.

@wkentaro's https://github.com/ros-planning/moveit/pull/72/files#diff-87e10acbb0a29d13b354b7ada0b8d249R8
@wkentaro could you please provide a patch to switch the two definitions, or was there a reason why you reordered them?
He added the `allowed_goal_duration_margin` dynamic parameter before `execution_velocity_scaling` and thus reordered the type
(ouch please don't do that). It wouldn't hurt to switch the two parameters. I'm not entirely sure if this is a real problem
here because I don't know enough details about DynamicReconfigure. In general, it might happen that somebody compiled against the old
headers (which is what ABI is about) and accesses `execution_velocity_scaling`, they will instead access `allowed_goal_duration_margin`
without knowing. The new dynamic reconfigure variable also breaks ABI (could induce a segfault), so to correctly fix this, we would have to revert.
I don't think this is too critical because this only effects people if they used their own `TrajectoryExecutionManager`, so I wouldn't revert.
But I would still switch the definitions (at least in indigo) to avoid data-misinterpretation.

So apart from the latter one, this actually seems to be ok and better than I imagined!

davetcoleman, 2016-08-26:
@jspricke could you elaborate on how you created this "compat_reports" - I don't know what compat is

v4hn, 2016-08-26:
On Fri, Aug 26, 2016 at 10:34:07AM -0700, Dave Coleman wrote:
> could you elaborate on how you created this "compat_reports" - I don't know what compat is

@davetcoleman This is the standard output of https://github.com/lvc/abi-compliance-checker/ .
Creating these reports consists of compiling the last released version of the libraries in debug mode and the current head too.
The tool(s) can then be used to compare the two versions of each library (and its headers)
and as you see it generates a list of incompatibilities for each library.

One can then look through the list and judge each problem for whether breaking the ABI there is ok or not.
I'll try to do this for each stable branch release in the future.

davetcoleman, 2016-08-26:
Great, and I added docs: https://github.com/ros-planning/moveit.ros.org/pull/51

v4hn, 2016-09-02:
I added https://github.com/ros-planning/moveit/pull/196 to address the dynamic reconfigure parameter breakage.

davetcoleman, 2016-09-09:
As #196 is closed, I believe this can be closed too now. Please reopen otherwise.

closed 2016-09-09 by davetcoleman