moveit/moveit #125: load default_capabilities by default

v4hn, 2016-08-23:
We have the problem that whenever we add a default capability to `move_group`, *every* user has to add the capability to their own moveit_config.

Instead we should (unless overridden) load all the default capabilities.

rhaschke, 2016-08-25:
How do you want to disable a particular default capability?

davetcoleman, 2016-08-25:
My thought: currently it's [opt-in](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/move_group.launch#L49):

```
    <!-- MoveGroup capabilities to load -->
    <param name="capabilities" value="move_group/MoveGroupCartesianPathService
				      move_group/MoveGroupExecuteTrajectoryAction
				      move_group/MoveGroupKinematicsService
				      move_group/MoveGroupMoveAction
				      move_group/MoveGroupPickPlaceAction
				      move_group/MoveGroupPlanService
				      move_group/MoveGroupQueryPlannersService
				      move_group/MoveGroupStateValidationService
				      move_group/MoveGroupGetPlanningSceneService
				      move_group/ApplyPlanningSceneService
				      move_group/ClearOctomapService
				      " />
```

we could also add a blacklist:

```
    <!-- MoveGroup capabilities to not load -->
    <param name="disable_capabilities" value="move_group/ClearOctomapService" />
```

Note the **disable_capabilities**

v4hn, 2016-08-25:
My initial thought was to keep a list of "default capabilities" in the code, or maybe there even is a pluginlib
interface for "give me all plugins from the `moveit_ros_move_group` package that are MoveItCapabilities,
and load these on startup. To override this behavior, one could specify a ROS parameter `no_default_capabilities`.

Now that I read Dave's proposal, I realize that people who make use of the `no_default_capabilities` flag will
evidentally not get to use new capabilities. So I think Dave's `disable_capabilities` mechanism is more user-friendly
and should be preferred.

rhaschke, 2016-08-25:
I actually like the fact, that you explicitly need to specify which capabilities to load. For example, one might want to replace one of the default capabilities with an own implementation. As long as the name matches, this will be perfectly fine. If a list of default capabilities is loaded silently, this might change the behaviour in unexpected ways as well.

> ... maybe there even is a pluginlib interface for "give me all plugins from the `moveit_ros_move_group` package that are MoveItCapabilities

That's available and used, for example, by rviz to list available (display) plugins.
However, I would strongly discourage to load all available plugins in the system. As pointed out above, there might be alternative implementations of the same capability which would cause conflicts when loaded together.

Overall, is it really a big issue to explicitly list all required capabilities and to ask the user to adapt that list? I think it doesn't happen so often that we change the default list. I guess, this is mainly related to `MoveGroupInterface` which expects specific capabilities to be present.

davetcoleman, 2016-08-26:
@v4hn's change would prevent us from having to send out emails like this everytime we want to change/add a capability:

https://groups.google.com/forum/#!searchin/moveit-users/move_group$20capability%7Csort:relevance/moveit-users/mUQqqFsGu9U/VSj0tswHfHAJ

We are currently going to have to send out a similar email for the new execute action. Often the maintainer for a moveit_config package has moved on, so even after this email is sent out many robot configs won't get updated until some new user get stuck on an error message at runtime. 

We can have a default list that always loads and, if you want to have custom capabilities, you can just name them differently and list the similar default one as disabled.

PR #359 (2016-11-16): add list of default capabilities
Functionality in MoveIt!'s move_group node is implemented in terms of
MoveGroupCapability plugins. But until now the choice of which plugins
are loaded was solely left to the user who in many cases is not familiar
with these plugins at all. This entails that whenever we introduce or change
a basic capability we have to annoy the user with warnings on startup that
bug him to change and/or update his move_group.launch file.

This commit changes the startup behavior of move_group to load a specified
list of default capabilities and provides a new ROS parameter `di [... cut]
commit dabe57f8 (2016-11-16): add list of default capabilities (#359)

Functionality in MoveIt!'s move_group node is implemented in terms of
MoveGroupCapability plugins. But until now the choice of which plugins
are loaded was solely left to the user who in many cases is not familiar
with these plugins at all. This entails that whenever we introduce or change
a basic capability we have to annoy the user with warnings on startup that
bug him to change and/or update his move_group.launch file.

This commit changes the startup behavior of move_group to load a specified
list of default capabilities and provides a new ROS param [... cut]
commit 19000104 (2016-11-17): add list of default capabilities (#359) (#366)

Functionality in MoveIt!'s move_group node is implemented in terms of
MoveGroupCapability plugins. But until now the choice of which plugins
are loaded was solely left to the user who in many cases is not familiar
with these plugins at all. This entails that whenever we introduce or change
a basic capability we have to annoy the user with warnings on startup that
bug him to change and/or update his move_group.launch file.

This commit changes the startup behavior of move_group to load a specified
list of default capabilities and provides a new RO [... cut]
closed 2016-11-16 by v4hn