moveit/moveit #136: enable proper collision checking for objects attached to links without geometry

mikeferguson, 2016-08-24:
Porting from https://github.com/ros-planning/moveit_core/issues/158, since this is still an issue, and one people frequently encounter:

Currently, links without geometry are left out of the updated links vector for a joint group. This also implies that objects attached to these links will not be part of collision checking for that group - this needs to be fixed.

Levi-Armstrong, 2016-08-25:
I recently been working a lot with the FCL collision robot code so I will see if I can help locate the issue. Looking at the code, the FCL collision robot uses the robot_model method [getLinkModelsWithCollisionGeometry](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/collision_detection_fcl/src/collision_robot_fcl.cpp#L42) which should get all links with collision geometry regardless of whether it is attached to a link without geometry.  Is this where you are seeing it not get all links with collision geometries?

If the above method is correctly acquiring all collision geometries for a given link. The second possibility is that the transform for the collision objects attached to a link without geometry are not getting updated. This gets done using the [robot_state.getCollisionBodyTransform](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/collision_detection_fcl/src/collision_robot_fcl.cpp#L92) which calls the method [robot_state.updateCollisionBodyTransforms](https://github.com/ros-planning/moveit/blob/d001746257005e7bde01f1bed581ab78a763d3f0/moveit_core/robot_state/src/robot_state.cpp#L441) prior to returning the transform. The method updateCollisionBodyTransforms updates all information for a given robot_state which should be independent of the group.  All of the link transformation appears to be updated [here](https://github.com/ros-planning/moveit/blob/d001746257005e7bde01f1bed581ab78a763d3f0/moveit_core/robot_state/src/robot_state.cpp#L476) which may be a good place to verify that every link is being updated.

v4hn, 2016-08-25:
+1 This remains a severe and safety critical issue in MoveIt and should finally be resolved!

Thanks for looking into this Levi!

gavanderhoorn, 2016-08-25:
Am I understanding you correctly @Levi-Armstrong that you feel this is not / should not be an issue anymore?

Levi-Armstrong, 2016-08-25:
@mikeferguson, I am not sure if this is the issue, but if you are trying to get links that are in the kinematic model for the group but outside the group this is done using a separate method of the JointModelGroup. If you call the JointModelGroup::getLinkModels() you will only get links that are part of the joint group which does not include links outside the group. Here is an example (URDF description is below): Group A defined by chain Link1->Link5 which does not include Link6. So, if you were to call the method [JointModelGroup::getLinkModels()](https://github.com/ros-planning/moveit/blob/4ac0c7432d335f57aab6836cbcaaac3fccf4b6f9/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h#L149-L150) you would get links (1, 2, 3, 4, and 5) but it would not include 6. If you want to get all links that get updated when this group move, use the method [JointModelGroup::getUpdatedLinkModels()](https://github.com/ros-planning/moveit/blob/4ac0c7432d335f57aab6836cbcaaac3fccf4b6f9/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h#L237-L240) which would return links (1, 2, 3, 4, 5, and 6). Hope this helps.

URDF
- Link1
    - Link2
        - Link3
            - Link4
                - Link5
            - Link6

Levi-Armstrong, 2016-08-25:
@gavanderhoorn: I don't thinks it is an issue anymore. I think it my be the case describe in my previous comment. I recently ran into this issue myself.

v4hn, 2016-08-25:
@Levi-Armstrong you don't think this is an issue anymore but you recently ran into the issue yourself?
You lost me...

We recently had an incident in our lab that bend a pringles can out of shape after attaching it to a link without geometry, so I believe this is still an issue.

gavanderhoorn, 2016-08-25:
@v4hn wrote:
> @Levi-Armstrong you don't think this is an issue anymore but you recently ran into the issue yourself?

I think the first *this* refers to the issue reported by @mikeferguson, while the second *this* refers to the issue that @Levi-Armstrong describes in https://github.com/ros-planning/moveit/issues/136#issuecomment-242444391.

mikeferguson, 2016-08-25:
I cannot confirm or deny that this is still an issue -- towards the end of World MoveIt Day, I started porting issues out of the old repos so they don't get lost, and I started with tickets that could be critical bugs and that I knew for a fact were definitely an issue at *some point* in time, so this ended up being one of the first tickets to be ported, because I know that I ran into it ~2 years ago, and that there were no linked commits/PRs that actually made it into the mainline repo.

v4hn, 2016-08-25:
@gavanderhoorn in that case, the issue still exists in two *different* parts:

- if the EEF is a link without attached geometry, collisions for objects attached to it are not checked (I suppose still a problem according to my latest accident)

- objects attached to links that are not directly part of the chain are not checked. (the case described by @Levi-Armstrong )

Levi-Armstrong, 2016-08-26:
@v4hn By default the collision robot does not care about groups and the only way a group can affect the collision checking is if in the collision request the programmer enables the group. Are you able to test it where you enable it and disable the group for collision checking, which is done by setting [this](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h#L186-L187) in the request. If you find that it works for the one when the group is not enabled then there must be an issue in this portion of the [code](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/robot_model/src/joint_model_group.cpp#L228-L244).

Levi-Armstrong, 2016-08-26:
>@Levi-Armstrong you don't think this is an issue anymore but you recently ran into the issue yourself?
You lost me...

I don't think there is an issue with collision checking when having links with geometry attached to links without. I currently have several cells which are setup where the end effector is attached to tool0 which has no geometry and it has no problem detecting collisions for the end effector.  Now that being said there could be an issue with how moveit collects the [updated list of links](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/robot_model/src/joint_model_group.cpp#L228-L244) depending on how the group is defined.

The issue I ran into was I thought the method JointModelGroup::getLinkModels() should return all links that get updated, but it does not. You must use this method JointModelGroup::getUpdatedLinkModels(). Not knowing much about what the original developer was doing when this issue was created, I thought this may be the issue they were running into.

Levi-Armstrong, 2016-08-26:
I just did another test case to see what happens if you define the move_group such that it does not include the last link: chain link1-tool0. I tested it in rviz and MoveIt still detects collision between the end effector and other objects.  I don't think there is an issue anymore. Also given that almost every robot within  ros has a defined tool0 without geometry, I would have expected more recent comments if this was still an issue.

Now MoveIt does not prevent collision a 100% because it is only doing discrete checks at every state in the robot trajectory. It is very possible to have a collision occur if there are thin objects in the collision environment. The second, possibility is when using collada files for collision models, there is an issue somewhere in how collada file are processed when used as collision models. I recently used ran into this were MoveIt was detecting collision when it should not be and not detecting collisions when it should have been.

URDF
- Link1 (has geometry)
    - Link2 (has geometry)
        - Link3 (has geometry)
            - Link4 (has geometry)
                - Link5 (has geometry)
                    - Link6 (has geometry)
                        - tool0 (does not have geometry)
                            - eff (has geometry)

Jntzko, 2016-08-30:
> I don't think there is an issue anymore

@v4hn and I created a setup were you can see that there is no collision checking for collision objects attached to links without collision geometry (at least with this setup).

The first picture shows a trajectory where the cylinder passes through the collision object when it is attached to `s_model_tool0`, the last link in our move group. This link has no collision geometry.
The group is declared as `<chain base_link="ur5_base_link" tip_link="s_model_tool0" />` in the srdf.

[image]

The second picture shows a trajectory that avoids the collision. Here, the cylinder is attached to `ur5_ee_link`, a link with collision geometry. Btw, the geometry there has obviously been added to *avoid this very bug*.
[image]

To run this setup you need the following repos in indigo (not sure if things work in kinetic yet):
- https://github.com/TAMS-Group/tams_ur5
- https://github.com/TAMS-Group/tams_ur5_setup
- https://github.com/Jntzko/collision_test

The urdf/moveit config depend on
- https://github.com/ros-industrial/robotiq
- https://github.com/ros-industrial/universal_robot
- https://bitbucket.org/traclabs/trac_ik.git

Run
```
roslaunch tams_ur5_setup_moveit_config demo.launch
```
and
```
rosrun collision_test collision_test
```
to see the two movements.

v4hn, 2016-08-30:
@Levi-Armstrong The setup @Jntzko just posted isolates the bug we experience and it definitely is a safety issue that should be fixed.
Could you have a look at it please? You seem to be have read through that part of the code quite a bit, so maybe you can find the bug with a working test-case?

Levi-Armstrong, 2016-08-30:
No problem, I will take a look. Thanks for the test case.

Levi-Armstrong, 2016-08-31:
@v4hn and @Jntzko, thanks for the test case it helped a lot. I did not realize the issue was when a object was dynamically attached to the robot.  I was able to locate the issue, it will occur if a group is enabled during the collision check which moveit enable by default in the collision request. I was able to resolve the issue so you should be seeing a PR soon.

v4hn, 2016-09-02:
Closed by #191

closed 2016-09-02 by v4hn